nextcloud/nextcloudpi #227: nextcloudpi certificate not valid after upgrade

voldemort13, 2017-09-28:
(this is a dublicate from nextcloud own support forum becouse i belive this belongs better here)
(total noob)
NEXTCLOUDPI
Upgrade nextcloud 12.0.2 to 12.0.3 flooded (about 30 of them) with certificate error like:

    core
        EXCEPTION
            OC\IntegrityCheck\Exceptions\InvalidSignatureException
            Certificate is not valid.
    twofactor_backupcodes
        EXCEPTION
            OC\IntegrityCheck\Exceptions\InvalidSignatureException
            Certificate is not valid.
    files_sharing
        EXCEPTION
            OC\IntegrityCheck\Exceptions\InvalidSignatureException
            Certificate is not valid.
    admin_audit
        EXCEPTION
            OC\IntegrityCheck\Exceptions\InvalidSignatureException
            Certificate is not valid.

Nextcloud version (12.3):
Operating system and version (raspian):
Apache or nginx version (dunno):
PHP version (dunno):
Is this the first time you’ve seen this error?:
yes
Can you reliably replicate it? (If so, please outline steps):
updgrade nextcloudpi with the webinterface to the newest stable version
The issue you are facing:
reinstalled from scratch same error 2 times

i can not log in with the domain(?) url only with my local ip. with domain it gives me 503 service unavailable

what do i do?

voldemort13, 2017-09-28:
[Certificate is not valid.txt](https://github.com/nextcloud/nextcloudpi/files/1341835/Certificate.is.not.valid.txt)
Here is the whole log output.
more like 100 or more  Certificate is not valid errors

Ricardosgeral, 2017-09-28:
i also can't access nextcloud from url after 0.30.1. I can access it with local ip

what's happening?

nachoparker, 2017-09-29:
This is duplicated in the [forums](https://help.nextcloud.com/t/nextcloudpi-upgrade-nextcloud-12-0-2-to-12-0-3-flooded-with-certificate-error/21643/3), so we can continue over there

closed 2017-09-29 by nachoparker