nextcloud/nextcloudpi #229: Apache not starting after generating certificates

yolani, 2017-09-29:
Hi there,

First of all thank you for this wonderful software bundle which will replace my synology based owncloud. 

However, i ran into a Problem. I did a fresh install using the 9-12-17 Image. The first thing i did was creating the Let‘s Encrypt certificates with the Web gui. There were no errors in the Output and the task finished with the usual „Please donate“. 

After that i did a reboot via ssh and the apache Server did not start up again. I checked the status, here is the output (of course XXX is Not my domain):

pi@raspberrypi:~ $ sudo systemctl status apache2
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-09-29 15:58:03 UTC; 15min ago
  Process: 440 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

Sep 29 15:58:01 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Sep 29 15:58:03 raspberrypi apachectl[440]: AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:
Sep 29 15:58:03 raspberrypi apachectl[440]: SSLCertificateFile: file '/etc/letsencrypt/live/XXX/fullchain.pem' does not exist or is empty
Sep 29 15:58:03 raspberrypi apachectl[440]: Action 'start' failed.
Sep 29 15:58:03 raspberrypi apachectl[440]: The Apache error log may have more information.
Sep 29 15:58:03 raspberrypi systemd[1]: apache2.service: Control process exited, code=exited status=1
Sep 29 15:58:03 raspberrypi systemd[1]: Failed to start The Apache HTTP Server.
Sep 29 15:58:03 raspberrypi systemd[1]: apache2.service: Unit entered failed state.
Sep 29 15:58:03 raspberrypi systemd[1]: apache2.service: Failed with result 'exit-code'.
pi@raspberrypi:~ $

However, the file is there and it contains a certificate. It‘s a symlink, but that should not be a Problem 

Any ideas or suggestions?

Greetings,

Christian

nachoparker, 2017-09-29:
What version are you in?

yolani, 2017-09-29:
The Web ui showed v0.26.30. and the content of the symilnk seems okay:

pi@raspberrypi:/etc/letsencrypt $ sudo ls -lah live/XXX
total 12K
drwxr-xr-x 2 root root 4.0K Sep 29 15:51 .
drwx------ 3 root root 4.0K Sep 29 15:51 ..
lrwxrwxrwx 1 root root   39 Sep 29 15:51 cert.pem -> ../../archive/XXX/cert1.pem
lrwxrwxrwx 1 root root   40 Sep 29 15:51 chain.pem -> ../../archive/XXX/chain1.pem
lrwxrwxrwx 1 root root   44 Sep 29 15:51 fullchain.pem -> ../../archive/XXX/fullchain1.pem
lrwxrwxrwx 1 root root   42 Sep 29 15:51 privkey.pem -> ../../archive/XXX/privkey1.pem
-rw-r--r-- 1 root root  543 Sep 29 15:51 README
pi@raspberrypi:/etc/letsencrypt $ sudo cat archive/XXX/fullchain1.pem
-----BEGIN CERTIFICATE-----
MIIFATCCA+mgAwIBAgISAydxgAspcWHmWC6Wuw55CDWgMA0GCSqGSIb3DQEBCwUA
MEoxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MSMwIQYDVQQD
ExpMZXQncyBFbmNyeXB0IEF1dGhvcml0eSBYMzAeFw0xNzA5MjkxNDUxMzBaFw0x
NzEyMjgxNDUxMzBaMBoxGDAWBgNVBAMTD2xhbmkuZHNjbG91ZC5tZTCCASIwDQYJ
KoZIhvcNAQEBBQADggEPADCCAQoCggEBAM9rN7x3yGxv77e6xMWtqSf7rQfiz4u0
......

nachoparker, 2017-09-29:
Wow that's so weird, so in ncp.conf does it have the exact path to the existing pem file?

yolani, 2017-09-29:
Oh my god i got it....

I am using the DynDns service from synology, since i own a NAS from synology. My domain is <Name>.DSCloud.me and this is what i entered in the gui when creating the certificates. 

So the XXX from above contained upper case letters in the config, but the folder in /etc/letsencrypt/ was created only in lower case. 

I just renamed the folder and everything is working, however this seems to be a little bug in the Web gui task when the config is altered After creating the certificates.

nachoparker, 2017-09-29:
so letsencrypt created the folder in lowercase? ok, then I'll have to modify where ncp.conf is written and convert the path to lowercase.

LOL I could'nt see that because you were doing XXX in place of your real domain (understandably).

Thanks for reporting

yolani, 2017-09-29:
Yes exactely, nice one... ;-) It‘s like „Why did nobody have this problem until now“...

Savest thing is probably to make everything lower case.

commit c2abfaa0 (2017-09-30): letsencrypt: fix uppercase domains cert path (Closes #229)
closed 2017-09-30 by nachoparker