nextcloud/nextcloudpi #233: Docker configuration after install not correct

pmk1c, 2017-10-01:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
Docker (latest, Image ID: 742464e57794, created 3 months ago)

### Expected behavior
After installing and running NextcloudPi Docker I see the login in the webinterface and can login with
admin / ownyourbits

### Actual behaviour
I see the Nextcloud Configuration Interface. Trying to create a new configuration by providing the default data here:

admin / myownpassword
DB Name: nextcloud
DB Host: localhost
DB User: ncadmin
DB Password: ownyourbits

Results in this error:
```
Error while trying to create admin user: An exception occurred while executing 'CREATE TABLE `oc_appconfig` (`appid` VARCHAR(32) DEFAULT '' NOT NULL, `configkey` VARCHAR(64) DEFAULT '' NOT NULL, `configvalue` LONGTEXT DEFAULT NULL, INDEX appconfig_config_key_index (`configkey`), INDEX appconfig_appid_key (`appid`), PRIMARY KEY(`appid`, `configkey`)) DEFAULT CHARACTER SET utf8 COLLATE utf8_bin ENGINE = InnoDB':

SQLSTATE[42S01]: Base table or view already exists: 1050 Table 'oc_appconfig' already exists
```

### Steps to reproduce, from a freshly installed image

### Include logs
Nothing in the nextcloud.log

nachoparker, 2017-10-01:
did you check the logs? The first time takes a little bit to configure everything on the volume.

Check

```
docker logs nextcloudpi
```

pmk1c, 2017-10-03:
@nachoparker Thanks for the help. It seems, I interrupted the installation process the first time the container was setting up and ended up with a corrupted volume. Deleting the volume and restarting the container gave me a working volume after a while.

Thanks for your help.

closed 2017-10-03 by pmk1c

nachoparker, 2017-10-03:
good, thanks for your feedback