nextcloud/nextcloudpi #236: can't log in via NCP Web

user-jan, 2017-10-02:
Hi Nacho,
I recently updated to the latest version and when I try to call up the NCP WEB GUI it presents me with the following dialogue:

[image]

Now, when I try to input my current password it won't accept it...

Best,
Jan

What version of NextCloudPi are you using? ( eg: v0.17.2 )

v0.31.6 

What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

NextcloudPi_09_03_17 (the command says "untagged" but I keep the downloaded image in my pc).

Expected behavior

Connect to Nextcloud via https://192.168.1.2:4443

Actual behaviour

Can't log in

nachoparker, 2017-10-02:
Hi Jan,

You have to use user `pi` and the password for that user. This is `raspbian` by default, but hopefully you have changed it already ;)

closed 2017-10-02 by nachoparker

user-jan, 2017-10-02:
Thanks a lot Nacho for your untiring assistance :-)