nextcloud/nextcloudpi #237: NextCloudPi problem with LetsEncrypt

FlexMcMurphy, 2017-10-02:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
0.31.6
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
12.0.3 which judging from the filename appears to have been built on 27th Sept 2017

You can discover this by executing
The LetsEncrypt routine from the NextCloudPi config screen on a Raspberry Pi 3.

### Expected behavior
LetsEncrypt should set up working HTTPS

### Actual behaviour
I get "connection refused" from letsencrypt when it tries to "Fetch"? a connection to the dynamic dns host I set up and now I can't get in to the NextCloudPi web admin screen either. This was working previously.

### Steps to reproduce, from a freshly installed image
Run the LetsEncrypt routine from the NextCloudPi config screen on a Raspberry Pi 3.

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging
Web interface not working - connection refused.

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
{"reqId":"w7MyRHI6XXShtFo4vYjz","level":4,"time":"2017-09-27T23:00:07+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":"12.0.3.3"}
{"reqId":"WdClnX8AAQEAAAJzi-wAAAoV","level":3,"time":"2017-10-01T08:21:56+00:00","remoteAddr":"192.168.1.1","user":"admin","app":"internet_connection_check","method":"GET","url":"\/index.php\/settings\/ajax\/checksetup","message":"Exception: {\"Exception\":\"GuzzleHttp\\\\Exception\\\\RequestException\",\"Message\":\"cURL error 18: transfer closed with 26004 bytes remaining to read\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(103): GuzzleHttp\\\\Exception\\\\RequestException::wrapException(Object(GuzzleHttp\\\\Message\\\\Request), Object(GuzzleHttp\\\\Ring\\\\Exception\\\\RingException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#2 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(132): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/react\\\/promise\\\/src\\\/FulfilledPromise.php(25): GuzzleHttp\\\\RequestFsm->GuzzleHttp\\\\{closure}(Array)\\n#8 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/ringphp\\\/src\\\/Future\\\/CompletedFutureValue.php(55): React\\\\Promise\\\\FulfilledPromise->then(Object(Closure), NULL, NULL)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Message\\\/FutureResponse.php(43): GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureValue->then(Object(Closure), NULL, NULL)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/RequestFsm.php(134): GuzzleHttp\\\\Message\\\\FutureResponse::proxy(Object(GuzzleHttp\\\\Ring\\\\Future\\\\CompletedFutureArray), Object(Closure))\\n#11 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(165): GuzzleHttp\\\\RequestFsm->__invoke(Object(GuzzleHttp\\\\Transaction))\\n#12 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Client.php(125): GuzzleHttp\\\\Client->send(Object(GuzzleHttp\\\\Message\\\\Request))\\n#13 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Http\\\/Client\\\/Client.php(138): GuzzleHttp\\\\Client->get('http:\\\/\\\/www.next...', Array)\\n#14 \\\/var\\\/www\\\/nextcloud\\\/settings\\\/Controller\\\/CheckSetupController.php(125): OC\\\\Http\\\\Client\\\\Client->get('http:\\\/\\\/www.next...')\\n#15 \\\/var\\\/www\\\/nextcloud\\\/settings\\\/Controller\\\/CheckSetupController.php(108): OC\\\\Settings\\\\Controller\\\\CheckSetupController->isSiteReachable('www.nextcloud.c...')\\n#16 \\\/var\\\/www\\\/nextcloud\\\/settings\\\/Controller\\\/CheckSetupController.php(414): OC\\\\Settings\\\\Controller\\\\CheckSetupController->isInternetConnectionWorking()\\n#17 [internal function]: OC\\\\Settings\\\\Controller\\\\CheckSetupController->check()\\n#18 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Http\\\/Dispatcher.php(160): call_user_func_array(Array, Array)\\n#19 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Http\\\/Dispatcher.php(90): OC\\\\AppFramework\\\\Http\\\\Dispatcher->executeController(Object(OC\\\\Settings\\\\Controller\\\\CheckSetupController), 'check')\\n#20 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/App.php(114): OC\\\\AppFramework\\\\Http\\\\Dispatcher->dispatch(Object(OC\\\\Settings\\\\Controller\\\\CheckSetupController), 'check')\\n#21 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Routing\\\/RouteActionHandler.php(47): OC\\\\AppFramework\\\\App::main('OC\\\\\\\\Settings\\\\\\\\Con...', 'check', Object(OC\\\\AppFramework\\\\DependencyInjection\\\\DIContainer), Array)\\n#22 [internal function]: OC\\\\AppFramework\\\\Routing\\\\RouteActionHandler->__invoke(Array)\\n#23 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Route\\\/Router.php(299): call_user_func(Object(OC\\\\AppFramework\\\\Routing\\\\RouteActionHandler), Array)\\n#24 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/base.php(1004): OC\\\\Route\\\\Router->match('\\\/settings\\\/ajax\\\/...')\\n#25 \\\/var\\\/www\\\/nextcloud\\\/index.php(48): OC::handleRequest()\\n#26 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/guzzlehttp\\\/guzzle\\\/src\\\/Exception\\\/RequestException.php\",\"Line\":51}","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/61.0.3163.100 Safari\/537.36","version":"12.0.3.3"}
{"reqId":"WdDKu38AAQEAAAJzjr8AABYB","level":2,"time":"2017-10-01T11:00:12+00:00","remoteAddr":"192.168.1.1","user":"admin","app":"core","method":"POST","url":"\/index.php\/login\/confirm","message":"Login failed: 'admin ' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/61.0.3163.100 Safari\/537.36","version":"12.0.3.3"}
{"reqId":"WdDKxX8AAQEAAAJzjsAAABcM","level":2,"time":"2017-10-01T11:00:22+00:00","remoteAddr":"192.168.1.1","user":"admin","app":"core","method":"POST","url":"\/index.php\/login\/confirm","message":"Login failed: 'admin ' (Remote IP: '192.168.1.1')","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/61.0.3163.100 Safari\/537.36","version":"12.0.3.3"}

Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>
==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Sat Sep 30 22:20:41.117259 2017] [proxy_fcgi:error] [pid 612:tid 2966647856] [client 192.168.1.1:51075] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Sat Sep 30 22:20:41.324798 2017] [proxy_fcgi:error] [pid 612:tid 2966647856] [client 192.168.1.1:51075] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Sat Sep 30 22:20:41.518277 2017] [proxy_fcgi:error] [pid 612:tid 2966647856] [client 192.168.1.1:51075] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[... cut ...]
2017-10-01 22:40:56 1988665344 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 5210329
2017-10-01 22:40:56 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-10-01 22:40:56 1988665344 [Note] Plugin 'FEEDBACK' is disabled.
2017-10-01 22:40:56 1988665344 [Note] Server socket created on IP: '127.0.0.1'.
2017-10-01 22:40:57 1988665344 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

</details>

FlexMcMurphy, 2017-10-02:
Just to clarify some more... I have set up a no-ip dynamic dns domain name and checked via their website that it is updating with my routers external ip address. I am also using the port forwarding tool in the command line UI but when I then use the letsencrypt tool it seems to fail on connecting to that no-ip domain name. It might be the way my port forwarding is set up. I was trying to do this in the router but now I am only using the command line UI tool to do the port forwarding.

Also, I still cannot connect to the web UI... I always get: connection refused.

Any help or advice gratefully appreciated!

Flex

nachoparker, 2017-10-03:
`nc-forward-ports` is not guaranteed to work. It requires for you to enable UPnP in your router. In any case it will tell you in the output how it went.

In any case, it is recommended to do the forwarding manually in your router an disable UPnP in your router.

closed 2017-10-06 by nachoparker