nextcloud/nextcloudpi #241: After latest update, nextcloud not working (possibly stuck in maintenance mode)

musashi242, 2017-10-03:
Running nextcloudpi 026.30 and just upgraded nextcloud, after which nothing seems to work, and I suspected things were stuck in maintenance mode.  trying to turn off maintenance mode with occ, I received the following error:

root@raspberrypi:/var/www/nextcloud# sudo -u www-data php occ maintenance:mode --off
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:61
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(148): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(150): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(588): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1345): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(309): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Authenticati...')
#19 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#20 /var/www/nextcloud/lib/private/Server.php(313): OC\ServerContainer->query('OC\\Authenticati...')
#21 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Authenticati...')
#23 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#24 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OC\\Authenticati...')
#25 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#26 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Authenticati...')
#27 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#28 /var/www/nextcloud/lib/private/Server.php(329): OC\ServerContainer->query('OC\\Authenticati...')
#29 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#30 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OCP\\IUserSessio...')
#31 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IUserSessio...')
#32 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OCP\\IUserSessio...')
#33 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#34 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('UserSession')
#35 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('UserSession')
#36 /var/www/nextcloud/lib/private/Server.php(1220): OC\ServerContainer->query('UserSession')
#37 /var/www/nextcloud/lib/private/Server.php(939): OC\Server->getSession()
#38 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#39 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Security\\CSR...')
#40 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Security\\CSR...')
#41 /var/www/nextcloud/lib/private/Server.php(935): OC\ServerContainer->query('OC\\Security\\CSR...')
#42 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#43 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('CsrfTokenManage...')
#44 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('CsrfTokenManage...')
#45 /var/www/nextcloud/lib/private/Server.php(1693): OC\ServerContainer->query('CsrfTokenManage...')
#46 /var/www/nextcloud/lib/private/Server.php(769): OC\Server->getCsrfTokenManager()
#47 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#48 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OCP\\IRequest')
#49 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IRequest')
#50 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OCP\\IRequest')
#51 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#52 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('Request')
#53 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('Request')
#54 /var/www/nextcloud/lib/private/Server.php(1087): OC\ServerContainer->query('Request')
#55 /var/www/nextcloud/lib/private/Server.php(416): OC\Server->getRequest()
#56 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#57 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OCP\\L10N\\IFacto...')
#58 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\L10N\\IFacto...')
#59 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OCP\\L10N\\IFacto...')
#60 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#61 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('L10NFactory')
[... cut ...]
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
```
</details>
<details>
<summary>Apache logs</summary>

```
Paste the output of `systemctl status apache2`
Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

nachoparker, 2017-10-03:
Please, provide the logs requested in the template in order to get proper help

musashi242, 2017-10-03:
Sure thing:

Nextcloud:

sudo tail -n 100 /var/www/nextcloud/data/nextcloud.log
{"reqId":"UolgBcw9ioiUk8H6EAGb","level":4,"time":"2017-09-12T08:48:16+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}

Apache:

pi@raspberrypi:~ $ systemctl status apache2
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: 
   Active: active (running) since Tue 2017-10-03 14:19:36 UTC; 2h 5min ago
  Process: 581 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCES
 Main PID: 799 (apache2)
   CGroup: /system.slice/apache2.service
           ├─799 /usr/sbin/apache2 -k start
           ├─879 /usr/sbin/apache2 -k start
           └─880 /usr/sbin/apache2 -k start

Oct 03 14:19:34 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Oct 03 14:19:36 raspberrypi systemd[1]: Started The Apache HTTP Server.

***********

pi@raspberrypi:~ $ tail -n 100 /var/log/apache2/*.log
==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Tue Oct 03 16:26:52.358966 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:52.559223 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:52.759536 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:52.959792 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:53.160054 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:53.360305 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:53.560593 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:53.760839 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:53.961136 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:54.161387 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:54.361631 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:54.561889 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:54.762142 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:54.962418 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:55.162715 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:55.362968 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:55.563223 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[Tue Oct 03 16:26:55.763506 2017] [proxy_fcgi:error] [pid 880:tid 1868575792] [client 192.168.1.193:54880] AH01071: Got error 'PHP message: PHP Warning:  filesize(): stat failed for /run/ncp.log in /var/www/ncp-web/ncp-output.php on line 52\n'
[... cut ...]
2017-10-03 14:03:15 1638835008 [Note] Event Scheduler: Purging the queue. 0 events
2017-10-03 14:03:16 1426060096 [Note] InnoDB: FTS optimize thread exiting.
2017-10-03 14:03:16 1638835008 [Note] InnoDB: Starting shutdown...
2017-10-03 14:03:17 1638835008 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-10-03 14:03:18 1638835008 [Note] InnoDB: Shutdown completed; log sequence number 295411326
2017-10-03 14:03:18 1638835008 [Note] /usr/sbin/mysqld: Shutdown complete

nachoparker, 2017-10-04:
your database is not running. Did you turn it off?

musashi242, 2017-10-04:
Thank you kindly for the help.  I did not turn off the database.  When I restarted it with 'sudo systemctl start mysqld,' it returned the following message:

pi@raspberrypi:~ $ sudo systemctl start mysqld
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.

systemctl status mariadb.service returns the following:

pi@raspberrypi:~ $ systemctl status mariadb.service
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2017-10-04 13:46:18 UTC; 6min ago
  Process: 7048 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 7045 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 6946 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environmen
t _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 6941 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 6939 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 7048 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Oct 04 13:46:06 raspberrypi systemd[1]: Starting MariaDB database server...
Oct 04 13:46:18 raspberrypi mysqld[7048]: 2017-10-04 13:46:18 1988497408 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 7048 ...
Oct 04 13:46:18 raspberrypi mysqld[7048]: 2017-10-04 13:46:18 1988497408 [Warning] Can't create test file /media/myCloudDrive/ncdatabase/raspberrypi.lower-test
Oct 04 13:46:18 raspberrypi mysqld[7048]: [113B blob data]
Oct 04 13:46:18 raspberrypi mysqld[7048]: 2017-10-04 13:46:18 1988497408 [ERROR] Aborting
Oct 04 13:46:18 raspberrypi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Oct 04 13:46:18 raspberrypi systemd[1]: Failed to start MariaDB database server.
Oct 04 13:46:18 raspberrypi systemd[1]: mariadb.service: Unit entered failed state.
Oct 04 13:46:18 raspberrypi systemd[1]: mariadb.service: Failed with result 'exit-code'.

journalctl -xe returns the following:

pi@raspberrypi:~ $ journalctl -xe
Oct 04 13:46:06 raspberrypi sudo[6933]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Oct 04 13:46:06 raspberrypi systemd[1]: Starting MariaDB database server...
-- Subject: Unit mariadb.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit mariadb.service has begun starting up.
Oct 04 13:46:18 raspberrypi mysqld[7048]: 2017-10-04 13:46:18 1988497408 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 7048 ...
Oct 04 13:46:18 raspberrypi mysqld[7048]: 2017-10-04 13:46:18 1988497408 [Warning] Can't create test file /media/myCloudDrive/ncdatabase/raspberrypi.lower-test
Oct 04 13:46:18 raspberrypi mysqld[7048]: [113B blob data]
Oct 04 13:46:18 raspberrypi mysqld[7048]: 2017-10-04 13:46:18 1988497408 [ERROR] Aborting
Oct 04 13:46:18 raspberrypi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Oct 04 13:46:18 raspberrypi systemd[1]: Failed to start MariaDB database server.
-- Subject: Unit mariadb.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit mariadb.service has failed.
-- 
-- The result is failed.
Oct 04 13:46:18 raspberrypi systemd[1]: mariadb.service: Unit entered failed state.
Oct 04 13:46:18 raspberrypi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Oct 04 13:46:18 raspberrypi sudo[6933]: pam_unix(sudo:session): session closed for user root

*    *    *

Thanks again for your help!  It is much appreciated.

nachoparker, 2017-10-04:
seems a permissions problem

```
ls -ld /media/myCloudDrive/ncdatabase
```

musashi242, 2017-10-04:
Hmmm... I thought the same thing.  Though could also be that I believe I set the database to 'ncdata.'   I tried to create the dir ncdatabase and give all write access but get the same error:

pi@raspberrypi:/media/myCloudDrive $ ls -al
total 40
drwxrwxrwx  7 root     root      4096 Oct  4 14:09 .
drwxrwxrwx  3 root     root      4096 Oct  3 14:19 ..
drwxr-x--- 10 www-data www-data  4096 Oct  3 13:57 blgdata
drwx------  2 root     root     16384 Oct  2 20:39 lost+found
drwxr-x---  8 www-data www-data  4096 Oct  2 21:31 ncdata
drwxrwxrwx  2 root     root      4096 Oct  4 14:22 ncdatabase
drwxr-xr-x  5 mysql    mysql     4096 Oct  3 14:03 ncdatabase-backup
pi@raspberrypi:/media/myCloudDrive $ cd ncdata
-bash: cd: ncdata: Permission denied
pi@raspberrypi:/media/myCloudDrive $ su
Password: 
root@raspberrypi:/media/myCloudDrive# ls
blgdata  lost+found  ncdata  ncdatabase  ncdatabase-backup
root@raspberrypi:/media/myCloudDrive# cd ncdata
root@raspberrypi:/media/myCloudDrive/ncdata# ls
admin  appdata_ocmx0enqff4e  appdata_ocnwok3l4thl  files_external  nextcloud.log  tmp
root@raspberrypi:/media/myCloudDrive/ncdata# 

* * *
Should I try to point MariaDB to ncdata somehow?  Thanks again for all your help.

nachoparker, 2017-10-05:
no, you should point it to `ncdatabase`. The permissions you have for `ncdatabase-backup` are correct.

I think you are confusing ncdata (for files) and ncdatabase (for the database)

nachoparker, 2017-10-06:
ask if still stuck

closed 2017-10-06 by nachoparker

musashi242, 2017-10-06:
Thanks so much for the help!