nextcloud/nextcloudpi #242: The nc-backup shows an error about managing access.log

Larsene, 2017-10-04:
### PROBLEMS

After running nc-backup to generate a backup of the database, there's an error about the access.log file : 

[image]

### What version of NextCloudPi are you using?  v0.36.1 

### What is the base image that you installed on the SD card? NextCloudPi_09-12-17

### Steps to reproduce, from a freshly installed image

After some use, try to launch the nc-backup script from nextcloudpi-config

nachoparker, 2017-10-05:
Hi,

The warning about 'file changed as we read it' shouldn't make the thing fail, it is only a warning.

That was fixed for nondata backups but thanks to your report I also fixed it for data backups.

Please, update to `v0.31.11` and try again

Thanks for your feedback

Larsene, 2017-10-06:
Perfect, it now works whithout any (message) errors.

thanks !

closed 2017-10-06 by Larsene