nextcloud/nextcloudpi #243: unable to enable some apps (News, Mail, JS XMPP,...)

Darkmagister, 2017-10-05:
I've tried to enable some apps from the store, like Mail and News (that i'm already using on a VPS) but i can not enable it and i'm getting errors in the logs that i don't understand
some others apps like Contacs, Calendar, Checksum and Metadata are working ok

### What version of NextCloudPi are you using?
0.31.11
### What is the base image that you installed on the SD card? 
NextCloudPi_10-03-17

### Expected behavior
go to Apps, click on enable of Mail or News and get enabled
### Actual behaviour
unable to enable app

### Include logs
<details>
<summary>Nextcloud logs</summary>

{"reqId":"WdY7LH8AAQEAADyOZ6EAAEUG","level":3,"time":"2017-10-05T14:01:42+00:00","remoteAddr":"XX.XX.XX.XX","user":"admin","app":"index","method":"POST","url":"\/index.php\/settings\/ajax\/enableapp.php","message":"Exception: {\"Exception\":\"Error\",\"Message\":\"Class 'OC\\\\DB\\\\MDB2SchemaReader' not found\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/db.php(173): OC\\\\DB\\\\MDB2SchemaManager->createDbFromStructure('\\\/var\\\/www\\\/nextcl...')\\n#1 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Installer.php(138): OC_DB::createDbFromStructure('\\\/var\\\/www\\\/nextcl...')\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(375): OC\\\\Installer->installApp('ojsxc')\\n#3 \\\/var\\\/www\\\/nextcloud\\\/settings\\\/ajax\\\/enableapp.php(46): OC_App->enable('ojsxc', NULL)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Route\\\/Route.php(155) : runtime-created function(1): require_once('\\\/var\\\/www\\\/nextcl...')\\n#5 [internal function]: __lambda_func()\\n#6 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Route\\\/Router.php(299): call_user_func('\\\\x00lambda_2305', Array)\\n#7 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/base.php(1004): OC\\\\Route\\\\Router->match('\\\/settings\\\/ajax\\\/...')\\n#8 \\\/var\\\/www\\\/nextcloud\\\/index.php(48): OC::handleRequest()\\n#9 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/DB\\\/MDB2SchemaManager.php\",\"Line\":69}","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:57.0) Gecko\/20100101 Firefox\/57.0","version":"12.0.3.3"}

nachoparker, 2017-10-06:
Couldn't reproduce. Note that for instance the News app requires a working internet connection for NCP, so I suspect that might be your problem.

closed 2017-10-06 by nachoparker

Darkmagister, 2017-10-15:
Ok but the pi is connected to internet I was enabling those app from work (while per ncp is at home) and opened port 80 and 443 in the router what do I need more? How can I get some more info to get some help?

nachoparker, 2017-10-19:
That nextcloud.log line contains some information, but seems internal to the app and I cannot reproduce it. I suggest you reopen this issue in [nextcloud server](https://github.com/nextcloud/server), or in the specific app github