nextcloud/nextcloudpi #245: Request entity too large

Ricardosgeral, 2017-10-06:
Hi

It's only me that can't upload big files (in this case a 177Mb zip) using the web interface. 
It says "file entity too large".

I think that something is not configured right in my settings, or is something general.

thanks

nachoparker, 2017-10-06:
I just uploaded a 1.5 GB

closed 2017-10-06 by nachoparker

Ricardosgeral, 2017-10-06:
Nacho did you upload from the browser?

nachoparker, 2017-10-06:
yes,

do you have free space? did you move to an USB drive?

Ricardosgeral, 2017-10-08:
Yes I have 2Tb free.
The data is in an USB HDD

thibmaek, 2017-10-17:
Same here, it's only an issue when I try accessing it from a public DNS record.
It works fine with the .local TLD, not with my .be TLD

File is 5MB.

nachoparker, 2017-10-19:
interesting, so I always test this on my local network. I can't see how it would be different with uploading from outside, I'll try next week from work.

nachoparker, 2017-10-24:
I just tested uploading a 200MB file externally (from work), and it went fine

Ricardosgeral, 2017-10-24:
After a fresh instalation of nextcloudpi so can I. I may have messed the
configuration previoulsly!