nextcloud/nextcloudpi #246: certbot error during update

Tercus, 2017-10-06:
### PROBLEM

I wanted to update my nextcloudpi and during the process got the message below. I tried running `sudo ncp-update` again and it didn't show the error anymore.

### Version of NextCloudPi
v0.26.32 -> v0.31.11

### Terminal message

```
[...]
Upgrading certbot-auto 0.16.0 to 0.19.0...
Replacing certbot-auto...
Creating virtual environment...
Installing Python packages...
Had a problem while installing Python packages.

pip prints the following errors: 
=====================================================
Traceback (most recent call last):
  File "/opt/eff.org/certbot/venv/bin/pip", line 9, in <module>
    load_entry_point('pip==9.0.1', 'console_scripts', 'pip')()
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/__init__.py", line 219, in main
    cmd_name, cmd_args = parseopts(args)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/__init__.py", line 159, in parseopts
    parser = create_main_parser()
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/__init__.py", line 137, in create_main_parser
    parser = ConfigOptionParser(**parser_kw)
  File "/opt/eff.org/certbot/venv/local/lib/python2.7/site-packages/pip/baseparser.py", line 148, in __init__
    self.config.read(self.files)
  File "/usr/lib/python2.7/ConfigParser.py", line 305, in read
    self._read(fp, filename)
  File "/usr/lib/python2.7/ConfigParser.py", line 512, in _read
    raise MissingSectionHeaderError(fpname, lineno, line)
ConfigParser.MissingSectionHeaderError: File contains no section headers.
file: /etc/pip.conf, line: 1
'extra-index-url=https://www.piwheels.hostedpi.com/simple/zope.components
'
=====================================================

Certbot has problem setting up the virtual environment.

We were not be able to guess the right solution from your pip 
output.

Consult https://certbot.eff.org/docs/install.html#problems-with-python-virtual-environment
for possible solutions.
You may also find some support resources at https://certbot.eff.org/support/ .
NextCloudPi updated to version v0.31.12
```

nachoparker, 2017-10-07:
hi,

please, try to update to `v0.31.13` and run certbot again

```
sudo /etc/letsencrypt/letsencrypt-auto --help
```

Tercus, 2017-10-07:
I *think* that solved the problem. `ncp-update` ran through with no problem (werid, `ncp-check-update` told me my version was the latest one).

I also ran the second command, and it ran through and only at the end showed me the different parameters. Is there any way I can confirm that certbot really works now?

nachoparker, 2017-10-08:
yes, the problem was with installation of certbot, so now it is properly installed. In order to configure it use letsencrypt from `nextcloudpi-config` or ncp-web.

closed 2017-10-08 by nachoparker

bennuttall, 2017-11-07:
```
file: /etc/pip.conf, line: 1
'extra-index-url=https://www.piwheels.hostedpi.com/simple/zope.components'
```

@Tercus this indicates your `/etc/pip.conf` is misconfigured. It looks like it's been patched to remove or comment out a line (people were suggesting this recently while piwheels had a bug in it), but it's been edited into a broken state. The piwheels bug is fixed now so there's no need to edit `/etc/pip.conf`. You should set it back to:

```
[global]
extra-index-url=https://www.piwheels.hostedpi.com/simple
```

and try again.