nextcloud/nextcloudpi #247: Can't connect to NextCloudPi via Samba

FlexMcMurphy, 2017-10-06:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.31.12
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
09-27-17

### You can discover this by executing
After initial installation and setup of NextCloudPi I cannot connect to the Samba share: 

[image]

### Expected behavior
I should be able to see the folders displayed in a Windows 7 explorer window in the path...
path = /var/www/nextcloud/data/admin/files

### Actual behaviour
I can never get past the above login window no matter what username password combination I try. I get this error message:
[image]

### Steps to reproduce, from a freshly installed image
I have setup SAMBA using both the command line and tried again via the web admin tool. I checked and SAMBA is running. After I confirmed that I could NOT connect via Samba I pasted into smb.conf my own share definition at the end that does let me in to the root of the Raspbian-Stretch OS with full access to everything. Obviously this is not safe practice... I would like to know why SAMBA does not authenticate for me from your image... do I need to change something in smb.conf??

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

{"reqId":"w7MyRHI6XXShtFo4vYjz","level":4,"time":"2017-09-27T23:00:07+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":"12.0.3.3"}
{"reqId":"WdfpPn8AAQEAABTweq0AAABE","level":3,"time":"2017-10-06T20:36:14+00:00","remoteAddr":"51.255.211.107","user":"--","app":"PHP","method":"GET","url":"\/en\/wp-login.php","message":"Exception: The requested uri(\/en\/wp-login.php) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/61.0.3163.100 Safari\/537.36","version":"12.0.3.3"}
{"reqId":"WdfvpX8AAQEAAA9f390AAE4G","level":3,"time":"2017-10-06T21:03:50+00:00","remoteAddr":"192.168.1.254","user":"admin","app":"index","method":"PUT","url":"\/index.php\/settings\/users\/admin\/settings","message":"Exception: {\"Exception\":\"OCP\\\\Files\\\\NotFoundException\",\"Message\":\"\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/Node.php(217): OC\\\\Files\\\\Node\\\\Node->getFileInfo()\\n#1 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/Node.php(117): OC\\\\Files\\\\Node\\\\Node->getPermissions()\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/Folder.php(336): OC\\\\Files\\\\Node\\\\Node->checkPermissions(8)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/SimpleFS\\\/SimpleFolder.php(65): OC\\\\Files\\\\Node\\\\Folder->delete()\\n#4 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/CardDAV\\\/PhotoCache.php(244): OC\\\\Files\\\\SimpleFS\\\\SimpleFolder->delete()\\n#5 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/AppInfo\\\/Application.php(121): OCA\\\\DAV\\\\CardDAV\\\\PhotoCache->delete('2', 'Database:admin....')\\n#6 [internal function]: OCA\\\\DAV\\\\AppInfo\\\\Application->OCA\\\\DAV\\\\AppInfo\\\\{closure}(Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent), '\\\\\\\\OCA\\\\\\\\DAV\\\\\\\\CardDA...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher))\\n#7 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/event-dispatcher\\\/EventDispatcher.php(174): call_user_func(Object(Closure), Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent), '\\\\\\\\OCA\\\\\\\\DAV\\\\\\\\CardDA...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher))\\n#8 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/event-dispatcher\\\/EventDispatcher.php(43): Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher->doDispatch(Array, '\\\\\\\\OCA\\\\\\\\DAV\\\\\\\\CardDA...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent))\\n#9 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/CardDAV\\\/CardDavBackend.php(672): Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher->dispatch('\\\\\\\\OCA\\\\\\\\DAV\\\\\\\\CardDA...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent))\\n#10 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/CardDAV\\\/SyncService.php(284): OCA\\\\DAV\\\\CardDAV\\\\CardDavBackend->updateCard('2', 'Database:admin....', 'BEGIN:VCARD\\\\r\\\\nVE...')\\n#11 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/AppInfo\\\/Application.php(131): OCA\\\\DAV\\\\CardDAV\\\\SyncService->updateUser(Object(OC\\\\User\\\\User))\\n#12 [internal function]: OCA\\\\DAV\\\\AppInfo\\\\Application->OCA\\\\DAV\\\\AppInfo\\\\{closure}(Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent), 'OC\\\\\\\\AccountManag...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher))\\n#13 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/event-dispatcher\\\/EventDispatcher.php(174): call_user_func(Object(Closure), Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent), 'OC\\\\\\\\AccountManag...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher))\\n#14 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/symfony\\\/event-dispatcher\\\/EventDispatcher.php(43): Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher->doDispatch(Array, 'OC\\\\\\\\AccountManag...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent))\\n#15 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Accounts\\\/AccountManager.php(111): Symfony\\\\Component\\\\EventDispatcher\\\\EventDispatcher->dispatch('OC\\\\\\\\AccountManag...', Object(Symfony\\\\Component\\\\EventDispatcher\\\\GenericEvent))\\n#16 \\\/var\\\/www\\\/nextcloud\\\/settings\\\/Controller\\\/UsersController.php(817): OC\\\\Accounts\\\\AccountManager->updateUser(Object(OC\\\\User\\\\User), Array)\\n#17 \\\/var\\\/www\\\/nextcloud\\\/settings\\\/Controller\\\/UsersController.php(750): OC\\\\Settings\\\\Controller\\\\UsersController->saveUserSettings(Object(OC\\\\User\\\\User), Array)\\n#18 [internal function]: OC\\\\Settings\\\\Controller\\\\UsersController->setUserSettings('contacts', 'admSeanin', 'contacts', '', 'private', '', 'contacts', '', 'private', '', 'private', '', 'private')\\n#19 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Http\\\/Dispatcher.php(160): call_user_func_array(Array, Array)\\n#20 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Http\\\/Dispatcher.php(90): OC\\\\AppFramework\\\\Http\\\\Dispatcher->executeController(Object(OC\\\\Settings\\\\Controller\\\\UsersController), 'setUserSettings')\\n#21 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/App.php(114): OC\\\\AppFramework\\\\Http\\\\Dispatcher->dispatch(Object(OC\\\\Settings\\\\Controller\\\\UsersController), 'setUserSettings')\\n#22 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Routing\\\/RouteActionHandler.php(47): OC\\\\AppFramework\\\\App::main('OC\\\\\\\\Settings\\\\\\\\Con...', 'setUserSettings', Object(OC\\\\AppFramework\\\\DependencyInjection\\\\DIContainer), Array)\\n#23 [internal function]: OC\\\\AppFramework\\\\Routing\\\\RouteActionHandler->__invoke(Array)\\n#24 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Route\\\/Router.php(299): call_user_func(Object(OC\\\\AppFramework\\\\Routing\\\\RouteActionHandler), Array)\\n#25 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/base.php(1004): OC\\\\Route\\\\Router->match('\\\/settings\\\/users...')\\n#26 \\\/var\\\/www\\\/nextcloud\\\/index.php(48): OC::handleRequest()\\n#27 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/Node.php\",\"Line\":97}","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/61.0.3163.100 Safari\/537.36","version":"12.0.3.3"}
{"reqId":"WdfwAn8AAQEAAA9egSAAAA8N","level":2,"time":"2017-10-06T21:05:07+00:00","remoteAddr":"192.168.1.254","user":"admin","app":"core","method":"POST","url":"\/index.php\/settings\/personal\/changepassword","message":"Login failed: 'admin' (Remote IP: '192.168.1.254')","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/61.0.3163.100 Safari\/537.36","version":"12.0.3.3"}
[... cut ...]
Oct 06 20:37:16 nextcloudpi systemd[1]: smbd.service: Supervising process 758 which is not our child. We'll most likely n
Oct 06 20:37:17 nextcloudpi systemd[1]: Started Samba SMB Daemon.
Oct 06 21:19:57 nextcloudpi smbd[6867]: pam_unix(samba:session): session closed for user nobody
Oct 06 21:32:57 nextcloudpi smbd[7631]: pam_unix(samba:session): session closed for user nobody
~
```
</details>
<details>
<summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

FlexMcMurphy, 2017-10-07:
To give some more information about my problem...

After initial installation of NextCloudPi... Here's how I set up SAMBA...
• Type into putty: nextcloudpi-config and activate Samba (Yes). Make the username as Pi and set up the password also.
• The share: NEXTCLOUDPI shows up in my Windows 7 Explorer but I can't get in to the share because it prompts me for a username/password and it didn't recognise the username (Pi) and pwd I set up in nextcloudpi-config… I tried prefixing the Pi username with WORKGROUP instead of my laptops computer name that was there by default.... WORKGROUP\Pi... but still no joy. Instead of WORKGROUP I also tried the Pis IP address and "NextcloudPi" and still nothing!

I even checked via putty with smbpasswd pi which lets you create and change SAMBA users and their passwords. However the user Pi is there. I changed it's password but it was still rejected at the windows log-in to the SAMBA share so I gave up and just did this...
• In [GLOBAL] section of /etc/samba/smb.conf make "map to guest = Bad Password" instead of "Bad User" and change the  [NEXTCLOUDPI] share definition from…

[NextCloudPi]
path = /var/www/nextcloud/data/admin/files
writeable = yes
; browseable = yes
valid users = pi
force group = www-data
create mask = 0770
directory mask = 0771
force create mode = 0660
force directory mode = 0770

To this instead…

[NextCloudPi]
path = /var/www/nextcloud/data/admin/files
writeable = yes
browseable = yes
;valid users = pi
read only = no
public = yes
guest ok = yes
force group = www-data
create mask = 0770
directory mask = 0771
force create mode = 0660
force directory mode = 0770

Then type in putty: service smbd restart

Now I can at least browse the Samba share just by clicking on the share name on Windows Explorer.. it doesn't even ask me to type in a username/password... but I've made changes to your "NextCloudPi" share definition I suspect are not safe and it doesn't give me the correct access permissions to the actual files.

Any idea what I am doing wrong? It's kind of really a SAMBA question... actually on other projects of mine I've never been able to figure out SAMBA authentication and permissions so I just make a share that is wide open!

Thank you for your help,

Flex

FlexMcMurphy, 2017-10-07:
Feels like I'm doing all the work here! Ha.. just kidding.

So I made some changes to your /etc/samba/smb.conf file and now after restarting Samba I get one share on the Windows 7 explorer window called - NEXTCLOUDPI. When I click on this it displays two folders from the NextCloudPi Image and I don't get any username/password prompt.

[NextCloudPi] -- Shows the files in /var/www/nextcloud/data/admin/files
I can browse, open, edit, delete them from my Windows explorer window. Much easier than typing arcane commands via Putty.

[root] -- Shows the root directory of the Raspbian Stretch Light OS
I can browse, open, edit, delete any files and folders I wish. Risky I know, but again, much easier getting around the OS.

So this is the way I always end up using Samba. The authentication never works for me and I don't really know what I'm doing so I just googled like crazy and found a way to make access to the shares wide open and apparently totally uncontrolled.

I would much rather get the authentication working properly. I would be grateful if you have any helpful thoughts and feedback. 

Here is my smb.conf file in case anyone is interested...
```
# Global parameters
[global]
        log file = /var/log/samba/log.%m
        max log size = 1000
        syslog = 0
        panic action = /usr/share/samba/panic-action %d
        usershare allow guests = Yes
        guest account = root
        map to guest = Bad Password
        obey pam restrictions = Yes
        pam password change = Yes
        passwd chat = *Enter\snew\s*\spassword:* %n\n *Retype\snew\s*\spassword:* %n\n *password\supdated\ssuccessfully* .
        passwd program = /usr/bin/passwd %u
        security = USER
        server role = standalone server
        unix password sync = Yes
        dns proxy = No
        idmap config * : backend = tdb

[homes]
        comment = Home Directories
        browseable = No
        create mask = 0700
        directory mask = 0700
        valid users = %S

[printers]
        comment = All Printers
        path = /var/spool/samba
        browseable = No
        printable = Yes
        create mask = 0700

[print$]
        comment = Printer Drivers
        path = /var/lib/samba/printers

[NextCloudPi]
        path = /var/www/nextcloud/data/admin/files
        create mask = 0777
        directory mask = 0777
        force create mode = 0777
        force directory mode = 0777
        force group = www-data
        guest ok = Yes
        read only = No

[root]
        comment = System root directory.
        path = /
        create mask = 0770
        directory mask = 0771
        force create mode = 0660
        force directory mode = 0770
        force group = www-data
        guest ok = Yes
        read only = No
```

nachoparker, 2017-10-08:
hi

Thanks for all that info and feedback. I think it would be nice to add a 'discoverable' option to NCP config, or maybe even make it discoverable by default.

Unfortunately I don't have a windows machine to test this on windows so somebody else will have to provide the 'perfect configuration' so I can tweak NCP.

nachoparker, 2017-10-08:
btw, probably this should be in the configuration

```
browseable = yes
```

nachoparker, 2017-10-08:
Let's wait to see if somebody else runs into the same problem or come up with a working configuration for windows.

In Linux, I just mount it with

I mount it using

```
sudo mount -t cifs //192.168.0.129/NextCloudPi /mnt -o username=pi,password=raspberry
```

In Arch I had to install gvfs-smb in order for nemo to be able to handle samba.

Can you try this (adjust your IP)

```
nemo smb://192.168.0.129/NextCloudPi
```

I also wrote [this](https://ownyourbits.com/2017/04/16/share-your-files-in-your-lan-with-samba/)

FlexMcMurphy, 2017-10-09:
Hello,

Thanks for linking your other page with details on SAMBA...

[image]

I checked with smbpasswd and the [pi] user does exist and I even changed the pwd to be sure I am using the correct one from the windows popup.

I was wondering if I'm getting the windows username/password login pop up that then doesn't authenticate because of a mismatch between file access permissions specified in smb.conf and in the Raspbian OS. So here is a screenshot with output of the ls - l command.

[image]

Those permissions should be ok I think to let me in from windows 7? I think Windows only recognises the OWNER bit and the [files] folder is [rwx]. But the share definition on the [files] folder is...

```
[NextCloudPi]
    path = /var/www/nextcloud/data/admin/files
    guest ok = yes
    writeable = yes
    browseable = yes
    ;valid users = pi
    force group = www-data
    create mask = 0777
    directory mask = 0777
    force create mode = 0777
    force directory mode = 0777
```
0777 means rwxrwxrwx but maybe the last four lines in the SAMBA share definition should be 7(rwx)7(rwx)5(r-x) ?? Well I tried changing those last four lines to 0775 and un-commenting the [ valid users] line. Now when I click on the share I get the pop-up authentication window again and username populates as NEXTCLOUDPI\pi and I type in the pasword but still get:

[image]

My set up is basically a Windows 7 laptop and a Pi running your NextCloudPi image. I do have Ubuntu also that I can run from the laptop on Oracle Virtual Box (Virtual Machine). At the command line under Ubuntu I ran your command:

`sudo mount -t cifs //192.168.0.129/NextCloudPi /mnt -o username=pi,password=raspberry`

... with my details and indeed the [files] folder did show up in the /mnt directory with no problem.

This suggests that Linux machines can connect fine to the NextCloudPi share but Windows machines probably need to make changes to smb.conf and for all I know... elsewhere too!

I couldn't understand what you were talking about with gvfs-smb and nemo ?? Are these other ways of accessing Samba shares from a command line interface? There was no gvfs-smb to install via apt-get. Also I don't see myself ever wanting to connect to Samba shares via command line!

Thank you,

Flex

nachoparker, 2017-10-09:
Hi,

the _gvfs-smb_ is only for Arch Linux, so forget about that. It is needed in Arch to make samba accessible in Nemo, which is a Linux file explorer, so no command line ;).

let's see if somebody has been able to do this in windows and can help with the configuration.

FlexMcMurphy, 2017-10-11:
No takers?

Just to give a short re-cap on my problem. The smb.conf file with NextCloudPi sets up a Samba share so the documents you upload to NextCloudPi can be browsed directly from a file explorer window in a Unix type OS. It also enables password authentication of a specific user: Pi.

However when I try to access this share in Windows 7 (Home Premium) Explorer I get the log-in dialogue box that doesn't authenticate me with the username and password I know is valid in Samba and Raspbian OS that NextCloudPi is installed on.

Some change needs to be made to smb.conf so the authentication works from Windows. I have researched some more and in previous versions of Samba users have had this issue with a Windows 7 client that was solved with changes to the registry. I tried some of these changes to no avail.

Any advice gratefully appreciated,

Flex

FlexMcMurphy, 2017-10-15:
I spent some more time trying to get Samba authentication working from my Windows 7 Home Premium computer to Samba shares on Raspbian OS and figured it out with my own smb.conf file with much help from this thread... [https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=195176&p=1223054#p1223054](url)

So I tried again with the smb.conf file that comes with the NextCloudPi image (no changes made) and now my Windows 7 Home Premium computer CAN access the Samba shares on NextCloudPi image and authenticate correctly.

I'm confused about what was going wrong for me. It is possible that my laptop was not using NTLMv2 authentication which Samba now supports.

`max protocol = SMB3`

... goes in the [GLOBAL] section and is the default value so doesn't need to be explicitly stated in smb.conf. So your setup will require Clients to use NTLMv2 to authenticate. It's possible my laptop was setup to only use NTLM.

But I'm not sure if this was really my issue. If anyone has this same problem, you don't need to make any changes to the default NextCloudPi smb.conf file but if you are a Windows 7 user check that the registry DWORD: LmCompatibilityLevel in HKLM\SYSTEM\CurrentControlSet\Control\Lsa is set to 3

[https://technet.microsoft.com/en-us/library/cc960646.aspx](url)

So I think this can be closed now.

Thank you,

Flex

closed 2017-10-15 by FlexMcMurphy

nachoparker, 2017-10-18:
Good investigation, thanks!