nextcloud/nextcloudpi #249: Request: Add NextcloudPi to the official Nextcloud Devices Website.

sunjam, 2017-10-10:
[Entered as issue #625 on the Nextcloud Website Github here.](https://github.com/nextcloud/nextcloud.com/issues/625)  Please add any thoughts or suggestions.   
    
Adding the issue here as well at the behest of Nachoparker in order to get it for visibility.

nachoparker, 2017-10-10:
Hi,

Thanks so much for the iniciative! It would be also nice to have a subforum, or at least a mention in the title of the [appliaces forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

sunjam, 2017-10-10:
No problem.  Imho, you are at the absolute top of quality Nextcloud contributers this last year, and deserve acknowledgement for your excellent volunteer work.

sunjam, 2017-10-19:
GOOD NEWS: Jos at Nextcloud says this is approved.  They would like us to edit the website via Git and add NextcloudPi.  [See https://github.com/nextcloud/nextcloud.com for the repo!
](https://github.com/nextcloud/nextcloud.com)

Looks like we might be able to add a couple things...    
1. Each device has a photo ([we could just use this generic photo of a naked Raspberry Pi](https://upload.wikimedia.org/wikipedia/commons/2/2d/Raspberry-Pi-2-Bare-FL.jpg)) and a simple description.
2. Each device has a "Learn More" page [ala the Nextcloud Box here at Nextcloud.com/box](https://nextcloud.com/box/).  It'd be super legit to have a full page at **Nextcloud.com/pi** + a link to [Own Your Bits website](https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/) from that page.

_Here is my example description:_
NextcloudPi is an official community release, running on Raspian as the OS for those who wish to Roll-Their-Own!  Simply copy the latest image to an SD Media card, then plug it in to your Pi to get started.

You'll have to supply your own Raspberry Pi 2 or Raspberry Pi 3 device + a USB Keyboard and compatible Monitor to get NextcloudPi running.  Built in command line and web interface tools make it simple to add additional USB Hard drives, and more easily maintain your installation!

NextcloudPi is compatible with both Raspberry Pi 2 and Raspberry Pi 3 devices (not included!) and peripherals. Support for other boards is also possible via the Docker image.

nachoparker, 2017-10-20:
wow this is huge!!

Thanks so much everyone!!

I like the description, let's do this!

JimmyKater, 2017-10-20:
i'd like to suggest the following...
change
> You'll have to supply your own Raspberry Pi 2 or Raspberry Pi 3 device + a USB Keyboard and compatible Monitor to get NextcloudPi running.

to 
You'll have to supply your own Raspberry Pi 2 or Raspberry Pi 3 device + a USB Keyboard and compatible Monitor at least once to get NextcloudPi running (for first run. after set-up headless running possible)

or something like this. what do you think?

nachoparker, 2017-10-20:
it's fine. In reallity, you don't need a monitor or keyboard... just plug it in to power and ethernet, then you can access the web interface on `nextcloudpi.local:4443` and nextcloud on `nextcloudpi.local`.

You can do most of the stuff from there. Maybe we'll include the option to open SSH from ncp-web in the future 

#130

nachoparker, 2017-10-25:
will you create the PR to the web @sunjam ?

thanks again for everything

Iolaum, 2017-11-13:
@JimmyKater Nachoparker 's right. You just need to [enable ssh after flashind the sd card image](https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-a-Raspberry-Pi#enable-ssh-optional) and then ssh in.

JimmyKater, 2017-11-13:
@Iolaum thanks for this hint... 

i was already aware of it, though ;)

sunjam, 2017-11-14:
@nachoparker I'll work on this today.  I've never submitted a pull request, but I will certainly try.
[image]

nachoparker, 2017-11-15:
@sunjam thank you, but first talk to @Pant because I think he was trying to do it himself, maybe you can help each other, or maybe he doesn't have time at the moment and you can do it

@Pant ?

Pant, 2017-11-16:
Hello @sunjam,
I tried setting up a development environment with wordpress but I wasn't able to make it work, because I have the same system for a long time and I've been experimenting a lot with it, thus it is messed up.

Said that I suggest you set up a docker container with wordpress (and an other container with mysql) as described in the [docker docs](https://docs.docker.com/compose/wordpress/) and modiffing it a bit, so it can see the `theme` directory, as it should based on the Readme.md of nextcloud.com [repository](https://github.com/nextcloud/nextcloud.com) 

Don't forget to fork the [nextcloud.com repo](https://github.com/nextcloud/nextcloud.com) and make commits to it, and then make a Pull Request as described in the [Development Process](https://github.com/nextcloud/nextcloud.com#development-process) section.

nachoparker, 2017-11-16:
Thank you @sunjam for doing this for us

@Pant I think it's better if we keep this open until the work is finished ;)

Pant, 2017-11-16:
@nachoparker Sure. Lets keep it open and discuss everything related to this process.

TomTurnschuh, 2018-05-18:
@nachoparker
Thanks for the remark.

> ... several of the latest ncp-apps are not documented yet in the Configuration Reference.

That's what I've recently discovered. However, I don't have direct access to my device at the moment, hence I don't even know which ncp-apps are available. Still, I'm planning to make update suggestions sometime in the future.

nachoparker, 2018-05-19:
Sweet, thanks!

nachoparker, 2018-06-19:
@tflidd two different people have tried this and it seems pretty hard to setup the wordpress environment. Maybe you can help with a quick edit? If I am not mistaken you send PRs to nextcloud.com.

that'd be great!!

nachoparker, 2018-06-26:
cleanup, this doesn't seem like is ever gonna be done

closed 2018-06-26 by nachoparker

sunjam, 2018-08-27:
NextcloudPi is now listed on https://nextcloud.com/devices!  Congratulations!

nachoparker, 2018-08-27:
thanks everyone!