nextcloud/nextcloudpi #250: Request: Replicate Google Takeout backups in NextcloudPi

sunjam, 2017-10-10:
[Would it be possible to have backups be handled in a similar fashion to Google Takeout](https://myaccount.google.com/?pli=1), allowing for selectable back up items + a final, singular tar or zip archive containing all of the data?

Example:    
- Show sub-menu of items being backed up or exported (Per App Configurations, Files)
- Show Additional list of items not supported (Installed Apps not yet supported by NextcloudPi backup)
- Time stamps

Why?
For easier migration of data and configurations to additional Nextcloud instances.

nachoparker, 2017-10-11:
added to the wishlist

nachoparker, 2017-12-04:
seems a bit out of scope right now, unless somebody wants to take on it

closed 2017-12-04 by nachoparker