nextcloud/nextcloudpi #251: Provide a default NCP theme

nachoparker, 2017-10-11:
We would have to decide wether to go for hardcore theming or using the theme app. We want it to be automated in the build process, so I don't think that the `occ` commands for theming are enough (haven't looked in depth).

I was thinking about this motto: "Keep your data close", also changing the background (maybe a blued image of a Raspberry Pi), but all that is open for discussion.

I tested for 5 minutes the theming app and it worked except for the logo (it wasn't SVG, don't know if its a requirement) and the background image (it was ignoring it, but didn't try much).

Some links

https://docs.nextcloud.com/server/12/developer_manual/core/theming.html
https://docs.nextcloud.com/server/12/admin_manual/configuration_server/theming.html

Shadowstreik, 2017-10-13:
I use a transparent PNG file for my logo.

I threw together a quick background-like RPi image and have it in Google Drive if anyone would like to see.

https://drive.google.com/open?id=0B725jO9bgCk0aGhvQklWbzRtdms

I did some theming graphics about 12 years ago for an online chat game I used to play (http://spacefleetonline.com). I also dipped into some 3D graphics (I completely forget how to do that, these days). But, I play with graphics when I need to.

Any thoughts are welcome.

nachoparker, 2017-10-19:
thanks!

man, you read my mind! I started doing the exact same thing with this image from wikipedia

[image](https://upload.wikimedia.org/wikipedia/commons/thumb/d/d4/Raspberry-Pi-2-Bare-BR.jpg/302px-Raspberry-Pi-2-Bare-BR.jpg)

Just making it blueish.

So, what did you use for theming Nextcloud? the theme app, or did you go hardcore into changing CSS files and stuff?

Shadowstreik, 2017-10-19:
Hi!

I looked at mine and felt it would be appropriate to have an image that fills the page as much as possible. I didn't want there to be too much, if any, blank area around the board tech. I did an image search using DuckDuckGo(no Google, Bing, Yahoo! etc. for me), found the image and saved it off. Then I opened it using a graphics program and turned it to black and white. After that, colorized it with a blueish color and done. It took about 5 to 10 minutes, I believe.

I saw in the docs that specific color code values are given, but I didn't pay attention to that detail when colorizing the image I was manipulating. I was only making an example image. That, of course, doesn't mean I couldn't make changes to get it just right. Or, it could be used 'as-is'. I don't mind, naturally.

Here's the example of my live instance...

https://drive.google.com/open?id=0B725jO9bgCk0SmZ1XzBaYU1zS2c

What do you think? I really like the logo, myself. It has 3 logos to make up one logo. The left is our family's coat of arms. The middle is a logo for my wife and my business (made by yours truly using MS Paint and Gimp 7 years ago). The right logo represents our fandom of the Ronald D. Moore production of Battlestar Galactica, it's the Colonial Phoenix.

That spacefleetonline.com logo was done in 2005 using MSPaint and Photoshop. The NCP shot uses an image I made black&white in color. I just made sure it was large enough to completely fill the page since I knew it would resize it properly. From getting the image, I used the in-app uploads in the theming area of the admin page inside of Nextcloud.

But, the image I made can certainly be used for NCP. Or, I could produce another, perhaps. I can make what you like. I could, possibly, make a logo if I have a base image to work from.I like that favicon that's used for the NCP web. If I could get a large copy, I can make one that would fit in that logo spot, nicely.

Any thoughts are appreciated.

- Shadowstreik

nachoparker, 2017-10-20:
Awesome,

Yeah, I noticed from the first day that you had a cool theme ;)

I really like the image that you made, the RPi logo is big and nice. The only thing I would be careful would be to check if that image has any rights to it.

It would be nice to have a SVG version of the NCP logo. At this point is only png.

The favicon was simply derived from the logo

[image]

So, it would be great if we could somehow automate the process of using the theme app... through `occ` or otherwise. Still haven't investigated.

Thanks!

Shadowstreik, 2017-10-20:
I drew these up using that large image. Knowing the time zone difference (approx. 7 hours), I wanted to get these out. They are SVG, as desired. Feel free to use. :smiley: 

No Text
https://drive.google.com/open?id=0B725jO9bgCk0Z2NjaEs3LUM3cFk

With Text
https://drive.google.com/open?id=0B725jO9bgCk0ZTVEeFhvSHlHZzQ

My one image from BSG- I don't have much to worry about rights, necessarily. That image is from the BSG wiki and a homage of our fandom(a bit of geek pride). My use of the NCP instance is only for use as a media archive for my family members. No public access, what-so-ever. Folks from my family never log in, as it is, since it's only effective as an archive. It's just there should anyone need any pictures or genealogy data (hence, the family coat of arms in the logo).

Please, tell me what you think of the logo images I drew up. It would be appreciated. If you would like to use them, please do so.

- Shadowstreik

nachoparker, 2017-10-20:
I really like your raspberry pi blue background image, that's the one I was asking about rights ;D

Thanks for the SVG!! awesome help. Unfortunately I was refering to using _vector_ graphics.

Probably it can be combined from getting a SVG raspberry logo and a SVG nextcloud logo in Inkscape

Shadowstreik, 2017-10-20:
I'll look into that. I did the SVG save using Inkscape from the PNG version. I may be able to do an import of sorts and convert it that way.

I found the image use rules at the RPi site...
https://www.raspberrypi.org/trademark-rules/

It states this, specifically about the use of the image...

> 2. When can I use the Raspberry Pi Logo?
> 
> You may only use the Raspberry Pi Logo in connection with the sale or distribution of genuine Raspberry Pi products or services. You must ensure that such use abides by the requirements and restrictions set out in our visual identity guidelines.
> 
> If you use the Raspberry Pi Logo in this way on a website, the logo must link to our website at http://www.raspberrypi.org

It also states when it can't be used...

> 3. When can’t I use the Raspberry Pi Marks?
> 
> You may not:
> 
> (i) use the Raspberry Pi Marks as part of your company name, company logo or product name;

The page has a lot of good information. I don't see how this couldn't be used without specific licensing, but it may require it given the name for NCP.

If having a similar logo image would work as a workaround, I have one from an individual that's really nice and I use as my internal SSH logo for my RPi units. Permission for use from the designer would, certainly, be needed.

I'm playing with the SVG in Inkscape to see if I can get it to draw out in vector graphics, specifically.

Shadowstreik, 2017-10-20:
I just looked at the Nextcloud usage rules at https://nextcloud.com/trademarks/

> Logo Usage Requirements
> Do not alter the logo in any way or overlap it with additional logos or images.
> 
> The logo is light blue (#0082c9) with white text/image. It should not be used differently, especially not inverted. The name "Nextcloud" can be optionally left out though, we provide logos without the text.
> 
> The typeface of the logo (or similar ones) should not be used anywhere else. Instead use Open Sans, in light weight. Use semibold sparingly and only for selected emphasis.

From how I understand it, NCP can't use the Nextcloud logo. On the safe side, using the striaght-up RPi logo should probably not be used, either.

Perhaps using the modified RPi logo (with permission) could be done and create something similar that is Nextcloud-like appear on either side, so it's understood, yet different. Any thoughts on that?

Shadowstreik, 2017-10-20:
I just had a thought.

Being an official Nextcloud project, wouldn't that give this project full usage rights, as needed, of the Nextcloud logo? If you think so, that would settle half of the logo issue. Although, I have an alternative in mind, just in case.

nachoparker, 2017-10-23:
Hi again, I have been away

I will contact NC and RPi in order to ask for permission for the logo. Good point

I am still worried about the background image (blue rpi board) that you created. Does that derive from a copyrighted image at all?

I am familiar with inkscape. If you can download a SVG vector version of both logos it should be easy to overlap them like I did with the PNG version.

Thanks again!

Shadowstreik, 2017-10-23:
I'm uncertain if it does. I'll look to see if there are any royalty-free or open-use images available and tweak that and use the color values the theme color convention directs. I also found a good converter. That, alone, takes a lot of work out of the mix.

nachoparker, 2017-10-23:
thanks,

that's why I chose the one from wikipedia, because the licenses are on the creative commons side of things.

That pic you found though is amazing, you can see the chip and the logo really well. Hopefully we can find the author and it is usable :D

Shadowstreik, 2017-10-23:
I've located the Raspberry Pi Visual Guidelines document at https://static.raspberrypi.org/files/Raspberry_Pi_Visual_Guidelines.pdf

Page 11 addresses photos in which it reads...

> Photographs of Raspberry Pi products and events can be made available upon
> request (see page 13). We ask that they are displayed at a clear size and are not
> altered in any way especially in terms of colour, orientation or proportion.
> A photo credit must always be present alongside any photographs used.
> Please credit www.raspberrypi.org.

Given that, am image such as the one I found, which is royalty-free from Shutterstock. The caveat is that the colors have to remain as they are and not be modified unless it was to be asked for permission to the Raspberry Pi Foundation at info@raspberrypi.org. A special use permission request should probably be sent from yourself as you're the project lead. It may also be beneficial to ask for any available closeup images in a large resolution that might be preferrable, perhaps.

nachoparker, 2017-10-23:
Great, thanks... I could do that certainly. I'll look around first if there's another one that is creative commons or free to use

Shadowstreik, 2017-10-23:
Ok, excellent. I just got word back from SiRo126 on Reddit about the other raspberry logo.

https://drive.google.com/open?id=0B725jO9bgCk0QXd4dW9sTmo1cXM

He says is can be used if you like. I have one with a black border I did to let it "pop" and stand out. I also have an idea for a NCP logo that uses this modded raspberry logo and doesn't step on Nextcloud. I have to put it together, but can certainly make it if you would like to see what I'm thinking.

nachoparker, 2017-10-23:
Awesomeee, we could definitely use that. 

I wrote NC this morning asking for permission, let's see first what they say, we might not have to change the logo depending on the answer.

Shadowstreik, 2017-10-23:
That's good to hear. I'll let him know. He also asked if I could keep him up to date on anything. I would believe it would be on the goings on for his image.

It'll be good to hear what NC says on the NC logo.

Shadowstreik, 2017-10-27:
Here is a converted logo image. It took some time to process, so I hope it's satisfactory.

https://drive.google.com/open?id=0B725jO9bgCk0QzQ5NVFfMy1XNnM

Shadowstreik, 2017-10-27:
I have a question out to the foundation regarding the background image with the blue hue.

nachoparker, 2017-10-29:
thanks so much for your time.

When we have permission to use the background image we'll put everything together ;)

The only thing now is to investigate if that can be automated for the build process somehow.

Shadowstreik, 2017-10-29:
> -------- Original Message --------
> Local Time: October 29, 2017 9:10 AM
> UTC Time: October 29, 2017 1:10 PM

You're very welcome. I like to dabble in graphics, on occasion. I hope everything goes well with it!

:D

Sent with [ProtonMail](https://protonmail.com) Secure Email.

nachoparker, 2017-11-03:
hi again @Shadowstreik 

do you still have the original or black and white version of the rpi board (the one that you made blueish)

also, did you hear anything about the rights of that image?

Shadowstreik, 2017-11-04:
Hi! I have the orignial color image. Making a black and white is just a few clicks of it.

I also heard back from the foundation. The image is clear to be used.

nachoparker, 2017-11-05:
that's great to hear

could you share the original one and the black and white one please? the blue needs to be darker so you can see the text in the login page, but I can do that ;)

about the logo... well, the NC designer was really kind to propose a new logo for us!

[image]

Shadowstreik, 2017-11-05:
Here is the entire library I made. It includes the original and black/white with some blue variants.

[](https://drive.google.com/open?id=0B725jO9bgCk0aGhvQklWbzRtdms)

I do like that logo. I see how it really puts Nextcloud in the forefront with the Pi leaves as the accent. I saw that and to make NCP be somewhat of an equal, II would have the leaves in place of the smaller circles I now think of as 'ears' whereas the large circle is the face. I can imagine those ears being the Raspberry ears. X) It _is_ a nice logo suggestion from the designer.

nachoparker, 2017-11-10:
Thanks everyone, this is the final result that will be included in the next release.

I  was able to modify the raspi background with a tune of blue that was matching the logo but it was hard to read the text: too much detail. In the end we decided to use this cleaner background ( thanks paolo ).

[image]

closed 2017-11-10 by nachoparker