nextcloud/nextcloudpi #252: Failed to start MariaDB database server

karnschi, 2017-10-11:
There was a problem when suddenly (I didn't change anything) the NC didn't work normally anymore. That was when I still had v0.26.11 so therefore I made a backup and updated the cloud. Still it didn't work so I searched here if maybe someone ended up with the same problem. I turned off the feature where the logs go into the ram but it wasn't helping.
My mounted HDD (datadir) and my permanent usb stick (database) are both mounted as they should be.

I guess that there is some kind of problem with my MariaDB as you can see below

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.31.14
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
untagged (sry but i cant remember when I downloaded it - only could guess it was late august or early september)

### Expected behavior
Connecting to Nextcloud via no-ip address or sync with Nextcloud Client on PC
### Actual behaviour
Client:
Error downloading https://MYDDNS/status.php - server replied: Internal server error
Browser:
site is just left blank
### Steps to reproduce, from a freshly installed image
I recently updated
### Include logs
<details>
<summary>Nextcloud logs</summary>
{"reqId":"znouJ7azVChaOZlMRlbo","level":2,"time":"2017-09-06T06:15:11+00:00","remoteAddr":"192.168.0.100","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"192.168.0.100\" tried to access using \"MYDDNS\" as host.","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
</details>
<details>
<summary>Apache logs</summary>
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-10-10 11:03:32 UTC; 1 day 1h ago
  Process: 15494 ExecReload=/usr/sbin/apachectl graceful (code=exited, status=0/SUCCESS)
  Process: 428 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
 Main PID: 672 (apache2)
   CGroup: /system.slice/apache2.service
           ├─  672 /usr/sbin/apache2 -k start
           ├─15603 /usr/sbin/apache2 -k start
           └─15604 /usr/sbin/apache2 -k start

Oct 10 11:03:29 nextcloud systemd[1]: Starting The Apache HTTP Server...
Oct 10 11:03:31 nextcloud apachectl[428]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Oct 10 11:03:32 nextcloud systemd[1]: Started The Apache HTTP Server.
Oct 11 06:25:02 nextcloud systemd[1]: Reloading The Apache HTTP Server.
Oct 11 06:25:02 nextcloud apachectl[15494]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message
Oct 11 06:25:02 nextcloud systemd[1]: Reloaded The Apache HTTP Server.

Log:
==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Wed Oct 11 06:25:02.474830 2017] [ssl:warn] [pid 672:tid 1992310784] AH01909: 127.0.1.1:4443:0 server certificate does NOT include an ID which matches the server name
[Wed Oct 11 06:25:03.000840 2017] [mpm_event:notice] [pid 672:tid 1992310784] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Wed Oct 11 06:25:03.000928 2017] [core:notice] [pid 672:tid 1992310784] AH00094: Command line: '/usr/sbin/apache2'

==> /var/log/apache2/other_vhosts_access.log <==
127.0.1.1:80 209.126.136.4 - - [11/Oct/2017:06:27:11 +0000] "GET / HTTP/1.1" 302 556 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36"
127.0.1.1:80 155.94.88.58 - - [11/Oct/2017:09:54:08 +0000] "GET / HTTP/1.0" 302 560 "-" "sysscan/1.0 (https://github.com/robertdavidgraham/sysscan)"</details>
<details>
<summary>mariaDB logs</summary>
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: signal) since Wed 2017-10-11 12:51:27 UTC; 3s ago
  Process: 31119 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=killed, signal=ABRT)
  Process: 31110 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 31011 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$VAR || exit 1 (code=exited, status=0/SUCCESS)
  Process: 31007 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 31004 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 31119 (code=killed, signal=ABRT)
   Status: "InnoDB: Fatal: Ending processing because of a corrupt database page."
   CGroup: /system.slice/mariadb.service

Oct 11 12:51:27 nextcloud systemd[1]: mariadb.service: Main process exited, code=killed, status=6/ABRT
Oct 11 12:51:27 nextcloud systemd[1]: Failed to start MariaDB database server.
Oct 11 12:51:27 nextcloud systemd[1]: mariadb.service: Unit entered failed state.
Oct 11 12:51:27 nextcloud systemd[1]: mariadb.service: Failed with result 'signal'.

Log:
InnoDB: End of page dump
2017-10-11 14:25:28 7682c000 InnoDB: uncompressed page, stored checksum in field1 3752031756, calculated checksums for field1: crc32 2377472849, innodb 3752031756, none 3735928559, stored checksum in field2 3874833060, calculated checksums for field2: crc32 2377472849, innodb 910562196, none 3735928559, page LSN 0 1024165755, low 4 bytes of LSN at page end 1024162802, page number (if stored to page already) 809, space id (if created with >= MySQL-4.1.1 and stored already) 0
InnoDB: page type 2 meaning UNDO LOG
InnoDB: Page may be an update undo log page
2017-10-11 14:25:28 1988280320 [Note] InnoDB: It is also possible that your operating system has corrupted its own file cache and rebooting your computer removes the error. If the corrupt page is an index page. You can also try to fix the corruption by dumping, dropping, and reimporting the corrupt table. You can use CHECK TABLE to scan your table for corruption. Please refer to http://dev.mysql.com/doc/refman/5.6/en/forcing-innodb-recovery.html for information about forcing recovery.
2017-10-11 14:25:28 1988280320 [ERROR] InnoDB: Ending processing because of a corrupt database page.
2017-10-11 14:25:28 7682c000  InnoDB: Assertion failure in thread 1988280320 in file ha_innodb.cc line 21990
InnoDB: We intentionally generate a memory trap.
InnoDB: Submit a detailed bug report to http://bugs.mysql.com.
InnoDB: If you get repeated assertion failures or crashes, even
InnoDB: immediately after the mysqld startup, there may be
InnoDB: corruption in the InnoDB tablespace. Please refer to
InnoDB: http://dev.mysql.com/doc/refman/5.6/en/forcing-innodb-recovery.html
InnoDB: about forcing recovery.
171011 14:25:28 [ERROR] mysqld got signal 6 ;
This could be because you hit a bug. It is also possible that this binary
or one of the libraries it was linked against is corrupt, improperly built,
or misconfigured. This error can also be caused by malfunctioning hardware.

To report this bug, see https://mariadb.com/kb/en/reporting-bugs

We will try our best to scrape up some info that will hopefully help
diagnose the problem, but since we have already crashed, 
something is definitely wrong and this may fail.

Server version: 10.1.23-MariaDB-9+deb9u1
key_buffer_size=16777216
read_buffer_size=131072
max_used_connections=0
max_threads=153
thread_count=0
It is possible that mysqld could use up to 
key_buffer_size + (read_buffer_size + sort_buffer_size)*max_threads = 351335 K  bytes of memory
Hope that's ok; if not, decrease some variables in the equation.

Thread pointer: 0x0
Attempting backtrace. You can use the following information to find out
where mysqld died. If you see no messages after this, something went
terribly wrong...
stack_bottom = 0x0 thread_stack 0x30000
The manual page at http://dev.mysql.com/doc/mysql/en/crashing.html contains
information that should help you find out what is causing the crash.
</details>

ibegleris, 2017-10-17:
I see similar issues when updating

karnschi, 2017-10-18:
have you found any solution?

ibegleris, 2017-10-18:
Short of installing a new nextcloudpi system at version 0.31.14? No.

nachoparker, 2017-10-18:
Hi,

Never seen this before, but it seems like your database is corrupt. Maybe you had a power cut or a faulty SD card/power adaptor?

Have a look at this.

https://dev.mysql.com/doc/refman/5.6/en/forcing-innodb-recovery.html

closed 2017-10-18 by nachoparker