nextcloud/nextcloudpi #253: Rebuild of Docker RPI version of nextcloudpi hangs

jeromeof, 2017-10-12:
Hi,
  Just using your instructions to try and rebuild the Docker image of NextcloudPI  hangs after the 'Setting up ssl-certs' command. 

This was with the latest build of nextcloudpi on a RPI v3, by using the command line indicated on Git:

**git clone https://github.com/nextcloud/nextcloudpi.git
make -C nextcloudpi**

Logs =>
To activate the new configuration, you need to run:
  service apache2 reload
Considering dependency proxy for proxy_fcgi:
Enabling module proxy.
Enabling module proxy_fcgi.
Module setenvif already enabled
To activate the new configuration, you need to run:
  service apache2 restart
Enabling conf php7.0-fpm.
To activate the new configuration, you need to run:
  service apache2 reload
Enabling module rewrite.
To activate the new configuration, you need to run:
  service apache2 restart
Enabling module headers.
To activate the new configuration, you need to run:
  service apache2 restart
Module dir already enabled
Module mime already enabled
Considering dependency setenvif for ssl:
Module setenvif already enabled
Considering dependency mime for ssl:
Module mime already enabled
Considering dependency socache_shmcb for ssl:
Enabling module socache_shmcb.
Enabling module ssl.
See /usr/share/doc/apache2/README.Debian.gz on how to configure SSL and create self-signed certificates.
To activate the new configuration, you need to run:
  service apache2 restart
Module status disabled.
Module reqtimeout disabled.
Module env disabled.
Module autoindex disabled.
Module access_compat disabled.
Module auth_basic disabled.
Module authn_file disabled.
Module authn_core disabled.
Module alias disabled.
Module access_compat already disabled
To activate the new configuration, you need to run:
  service apache2 restart
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  openssl
Suggested packages:
  ca-certificates openssl-blacklist
The following NEW packages will be installed:
  openssl ssl-cert
0 upgraded, 2 newly installed, 0 to remove and 11 not upgraded.
Need to get 712 kB of archives.
After this operation, 1156 kB of additional disk space will be used.
Get:1 http://ftp.heanet.ie/mirrors/raspbian/raspbian stretch/main armhf openssl armhf 1.1.0f-3 [692 kB]
Get:2 http://ftp.heanet.ie/mirrors/raspbian/raspbian stretch/main armhf ssl-cert all 1.0.39 [20.8 kB]
Preconfiguring packages ...
Fetched 712 kB in 1s (684 kB/s)
Selecting previously unselected package openssl.
(Reading database ... 13233 files and directories currently installed.)
Preparing to unpack .../openssl_1.1.0f-3_armhf.deb ...
Unpacking openssl (1.1.0f-3) ...
Selecting previously unselected package ssl-cert.
Preparing to unpack .../ssl-cert_1.0.39_all.deb ...
Unpacking ssl-cert (1.0.39) ...
Setting up openssl (1.1.0f-3) ...
Setting up ssl-cert (1.0.39) ...

nachoparker, 2017-10-18:
Hi,

This is a duplicate of #179 

I am close to being done with the rework of the docker image. I will upload the update within weeks. 

The old build system was based on Jessie, now it has been migrated to stretch and the new stuff has been included, such as ncp-web

nachoparker, 2017-10-18:
Duplicate of #179

closed 2017-10-18 by nachoparker