nextcloud/nextcloudpi #254: Enable wifi at first boot

Bridouz, 2017-10-13:
Hi,

I want to install nextcloudpi at home and I want to know if as on Raspbian it's possible to enable wifi by creating a `wpa_supplicant.conf` in the boot partition ?

Thanks.

nachoparker, 2017-10-18:
Yes it is.

Sorry have been away for several days.

nachoparker, 2017-10-18:
I suggest you use the forums in the future for questions like this ;)

closed 2017-10-18 by nachoparker

karnschi, 2017-10-20:
Here I got a link with a how-to in it.
https://www.raspberrypi.org/forums/viewtopic.php?t=191252
I haven't tried it myself but I am going to later that day

nachoparker, 2017-10-20:
the problem with this is that you need to already know the details of your SSID, password and so on.

there's nothing I can do about that at build time.

Maybe it could be added to the wiki though, and modified before flashing the image to SD?

karnschi, 2017-10-20:
you just have to put a file into the boot partition of your SD card

for enabling the SSH you just copy an empty file with the name "ssh" into the boot section and ssh will be enabled automatically

for the wifi connection you have to generate a file called "wpa_supplicant.conf" with the following lines inside:

```
country=US
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
    ssid="your_real_wifi_ssid"
    scan_ssid=1
    psk="your_real_password"
    key_mgmt=WPA-PSK
}
```

I am setting up my NCP right now and tried it and it worked at first try!

nachoparker, 2017-10-20:
nice!

The thing is that I cannot automate that and personalize images for every user, so the best we can do is to include this in the wiki, in case somebody wants to install it that way