nextcloud/nextcloudpi #255: Let's Encrypt issue

fab1ton, 2017-10-14:
My ports are both open, TCP 80 and 443 which I checked on portchecker.co.

The error I am getting is the following:

Domain: XXXX
   Type:   connection
   Detail: Connection refused

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address. Additionally, please check that
   your computer has a publicly routable IP address and that no
   firewalls are preventing the server from communicating with the
   client. If you're using the webroot plugin, you should also verify
   that you are serving files from the webroot path you provided.

Not sure how I can fix that. Like I said my port are both open and I am able to connect to my cloud outside of my network.

Thanks for your help!

j-ed, 2017-10-14:
Usually I would say that this is a pure RaspberryPi web server configuration problem and not a Nextcloud problem. You should make sure that `http://<yourserver>/.well-known/acme-challenge/` can be accessed from the Internet by the Let's Encrypt servers to be able to verify the generate challenge.

Shadowstreik, 2017-10-14:
It could, also, be useful to check if your ISP happens to block those ports. Some do. Though, I believe a lot don't. Mine doesn't, luckily.

fab1ton, 2017-10-15:
I was actually able to fix the problem. All good. Thanks guys!

closed 2017-10-15 by fab1ton

j-ed, 2017-10-15:
@fab1ton To allow other users to learn from this issue it would help if you could describe how you solved it instead of only writing `it's ok now` ;-)

fab1ton, 2017-10-15:
Oh sorry. I am having a DD-WRT which runs OpenVPN. My ports weren’t opened by just forwarding them. I either had to add them manually into the firewall or I excluded my Nextcloud IP from the OpenVPN, which I did. After that I was able to set up Let’s Encrypt.