nextcloud/nextcloudpi #256: Permissions problem when setting nextcloudpi directories on USB external drive

Iolaum, 2017-10-14:
### PROBLEMS

I am trying to use NextCloudPi on a nextcloudbox with a raspberry pi 2 model b. I have problems setting the nextcloud data directory and nextcloud database directory.

### What version of NextCloudPi are you using?
v0.31.14 

I remember that the image comes with an earlier version but I was prompted to update upon running `sudo nextcloudpi-config` for the first time.

### What is the base image that you installed on the SD card?
When extracted: NextCloudPi_10-02-17.img
With command: NextCloudPi_10-03-17

### Expected behavior

The configuration process would accept as a target a directory located on the external hard disk drive.

### Actual behaviour

```
ERROR: the user mysql does not have access permissions over $PARENT_OF_DIR_SPECIFIED
```

This happens on both the web interface and when connecting through ssh.
### Steps to reproduce, from a freshly installed image

- The external hard disk drive has been formatted with Gnome Disks (Ubunu 16.04.3 version) as EXT4 with name wd1.
- The sd card is flashed with nextcloud pi image.
- I log in through ssh, change default password.
- I set locales and languages.
- I setup the process to automatically mount USB drives, then reboot. The external hard disk drive is mounted at `/dev/wd1`.
- I then try to set up nextcloudpi data and database directories. I get errors that the appropriate user (mysql or www-data) doesn't have access permissions on the **parent** folder of the folder I designated.

For example at first I tried to set the nextcloudpi data directory at `/dev/wd1/datadir` and I got:
```
Launching nc-datadir
ERROR: the user www-data does not have access permissions over /media/wd1
```

The same happenedd when I tried to set the nextcloudpi database directory at `/dev/wd1/dbdir`
```
Launching nc-database
ERROR: the user mysql does not have access permissions over /media/wd1/
```

To try to get around that I tried to create a directory to assign to the proper user and then assign a directory inside it as the database directory. Hence I tried to assign the database directory at `/media/wd1/dbsedir/ncdatabase` while having manually made  mysql the owner of `/dev/wd1/dbsedir`. It still didn't work:

```
pi@nextcloudpi:/media/wd1 $ sudo nextcloudpi-config
# graphics stuff to select /media/wd1/dbsedir/ncdatabase
Launching nc-database
ERROR: the user mysql does not have access permissions over /media/wd1/dbsedir
Done. Press any key...
s
pi@nextcloudpi:/media/wd1 $ ls -la
total 44
drwx------ 8 pi    pi     4096 Oct 14 17:43 ./
drwxr-xr-x 3 root  root   4096 Oct 14 18:30 ../
drwxr-xr-x 2 pi    pi     4096 Oct 14 16:24 bkupdir/
drwxr-xr-x 2 pi    pi     4096 Oct 14 17:43 datadir/
drwxr-xr-x 2 mysql mysql  4096 Oct 14 17:43 dbsedir/
drwx------ 2 root  root  16384 Oct 14 03:31 lost+found/
drwxr-xr-x 2 pi    pi     4096 Oct 14 16:27 swap/
drwxr-xr-x 2 pi    pi     4096 Oct 14 16:54 test/
pi@nextcloudpi:/media/wd1 $ pwd
/media/wd1
```

You can see through `ls -la` that `mysql` is the owner of the parent directory

[... 295 lines cut ...]
2017-10-14 18:50:25 1639437120 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-10-14 18:50:28 1639437120 [Note] InnoDB: Shutdown completed; log sequence number 3031868
2017-10-14 18:50:28 1639437120 [Note] /usr/sbin/mysqld: Shutdown complete

2017-10-14 18:50:52 1989017600 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-10-14 18:50:52 1989017600 [Note] InnoDB: The InnoDB memory heap is disabled
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-10-14 18:50:52 1989017600 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Using Linux native AIO
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Using generic crc32 instructions
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Completed initialization of buffer pool
2017-10-14 18:50:52 1989017600 [Note] InnoDB: Highest supported file format is Barracuda.
2017-10-14 18:50:53 1989017600 [Note] InnoDB: 128 rollback segment(s) are active.
2017-10-14 18:50:53 1989017600 [Note] InnoDB: Waiting for purge to start
2017-10-14 18:50:53 1989017600 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3031868
2017-10-14 18:50:54 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-10-14 18:50:54 1989017600 [Note] Plugin 'FEEDBACK' is disabled.
2017-10-14 18:50:54 1989017600 [Note] Server socket created on IP: '127.0.0.1'.
2017-10-14 18:50:54 1989017600 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

Iolaum, 2017-10-15:
I think I found the common denominator of the problem. Let's take the case of changing the database directory to one on the hard drive.

The code errors at [this line](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-database.sh#L52):

```
sudo -u mysql test -x "$BASEDIR"
```

I was able to create a small script to test it. From the raspberrypi's pi user's home folder:

```
pi@nextcloudpi:~ $ pwd
/home/pi
pi@nextcloudpi:~ $ ll
total 32
drwxr-xr-x 3 pi   pi   4096 Oct 15 16:45 ./
drwxr-xr-x 4 root root 4096 Oct  3 19:14 ../
-rw------- 1 pi   pi   1146 Oct 14 19:52 .bash_history
-rw-r--r-- 1 pi   pi    220 Sep  7 15:59 .bash_logout
-rw-r--r-- 1 pi   pi   3524 Oct 14 16:04 .bashrc
-rwxr-xr-x 1 pi   pi    116 Oct 15 16:45 dbdirperm.sh*
drwxr-xr-x 2 pi   pi   4096 Oct 14 16:03 .nano/
-rw-r--r-- 1 pi   pi    675 Sep  7 15:59 .profile
pi@nextcloudpi:~ $ cat dbdirperm.sh 
#!/usr/bin/env bash

if test -x "$1"; then
  echo "Doing good!"
else
  echo "Houston we have a problem"
  exit 1
fi
pi@nextcloudpi:~ $ sudo -u mysql ./dbdirperm.sh .nano/
Doing good!
pi@nextcloudpi:~ $ sudo -u mysql ./dbdirperm.sh /media/wd1/dbsedir/
Houston we have a problem
pi@nextcloudpi:~ $ sudo -u mysql ./dbdirperm.sh /media/wd1/
Houston we have a problem
pi@nextcloudpi:~ $ ll /media/wd1/
total 44
drwx------ 8 pi    pi     4096 Oct 14 17:43 ./
drwxr-xr-x 3 root  root   4096 Oct 14 19:53 ../
drwxr-xr-x 2 pi    pi     4096 Oct 14 16:24 bkupdir/
drwxr-xr-x 2 pi    pi     4096 Oct 14 17:43 datadir/
drwxr-xr-x 2 mysql mysql  4096 Oct 14 17:43 dbsedir/
drwx------ 2 root  root  16384 Oct 14 03:31 lost+found/
drwxr-xr-x 2 pi    pi     4096 Oct 14 16:27 swap/
drwxr-xr-x 2 pi    pi     4096 Oct 14 16:54 test/
```

The first commands are so you understand what I did. Note `alias ll='ls -laF'`, ubuntu has it by default and I am taking it with me whenever I use a shell. The output of the last `ls -laF` command shows the problem. It's there in the first post as well but I didn't see it :(

The problem are the permissions of the hard drive!

```
pi@nextcloudpi:~ $ stat -c "%a %n" /media/wd1/
700 /media/wd1/
```

Only the pi user has access! 

I 'm guessing the next logical step now is to see what [nc-automount.sh](https://github.com/nextcloud/nextcloudpi/blob/2217045f6c16cb989c44c4ca7228acb1afaec9a2/etc/nextcloudpi-config.d/nc-automount.sh) does.

Any help appreciated.

nachoparker, 2017-10-19:
Hi,

It's interesting that only you are seeing this. The only thing I can support is if you use `nc-format-USB` to format to ext4, but in principle it should not matter that you formated in your PC.

what is the output of

```
lsblk -f
```

Why would you get different permissions? nc-automount only really uses udiskie

Iolaum, 2017-10-19:
Hello,

Here's the required output

```
pi@nextcloudpi:~ $ lsblk -f
NAME        FSTYPE LABEL UUID                                 MOUNTPOINT
sda                                                           
└─sda1      ext4   wd1   49ee0967-ddc9-4279-9259-b0f61d3400d1 /media/wd1
mmcblk0                                                       
├─mmcblk0p1 vfat   boot  B60A-B262                            /boot
└─mmcblk0p2 ext4         9a7608bd-5bff-4dfc-ac1d-63a956744162 /
```

I tried to fix this by changing the service command for udiskie (by [this line](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-automount.sh#L45) not on the code but at `/usr/lib/systemd/system/nc-automount.service`) I didn't really know what I was doing (which is why I didn't take notes and/or post anything about it here) and my incorrect arguments led to the drive been mounted by root instead of pi user most of the time. I also tried changing [this line](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-automount.sh#L35) after it got output to `/etc/udev/rules.d/99-udisks2.rules` with no good result. My intend was to force permissions during mount to 755 (for example with an option like umask=0022) but I didn't get it to work, if it is even possible.

In the end I cop out of it and run a simple:
```
$ pwd
/media/wd1
$ chmod 755 .
```

I was afraid this wouldn't hold after a restart but it did. Not sure why. Everything went smooth after this. After I 've done all this I noticed there is a [faq question about this](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#what-userpermissions-should-i-have-to-the-external-usb-drive-mount-point-the-ncdata-and-ncdatabase-directory). Even though I don't have 751 but 755 things seem to be working fine (maybe though 75**5** is a security risk?, not sure).

nachoparker, 2017-10-24:
good that you had it working.

still, it is strange that you had those permissions from udiskie... or did you create manually that folder in the first place?

We'll keep an eye for this, in case other users are seeing the same (which doesn't seem likely)

> maybe though 755 is a security risk?, not sure

Not huge, but the less permissions the better ;)

closed 2017-10-24 by nachoparker

wayneoutthere, 2018-11-06:
i am not sure if this is related but i had massive problems trying to mount an encrypted drive with the https://ownyourbits.com/downloads/NextCloudPi_RPi_10-23-18/
After i did an ncp-update the entire apache2 died and could not be saved.  Did many re-installs of image no luck. After moving to the newest version of NCP suddenly now i'm having a good running system but... I'm having permission issues like this above when trying to move my data from NCP to the encrypted USB drive:

[ nc-datadir ]
Cannot load Zend OPcache - it was already loaded
ERROR: the user www-data does not have access permissions over /media/USBdrive

[ nc-database ]
ERROR: the user mysql does not have access permissions over /media/USBdrive

Maybe i'm just not shutting down a service or something... but, my main point here is that with previous version of NCP there was some major problem with this area but now seems better....

nachoparker, 2018-11-07:
are you using automount? the error message is quite self-explanatory ;D

wayneoutthere, 2018-11-07:
i was trying to use automount but no go.
On the new version for some reason i was able to make much more progress.

btw, thanks for your awesome work man.  we in the UBports community are backing NCP.

nachoparker, 2018-11-07:
automount uses udiskie. IIRC you can do encrypted drives with udiskie, but that is not covered by NCP.

You need to provide access to the `www-data` user to `cd` and write to the datadir.

> btw, thanks for your awesome work man. we in the UBports community are backing NCP.

:D

wayneoutthere, 2018-11-07:
so to do encrypted drives in ncp, i need to do it manually, correct?  It's possible but I have to walk through and make sure everything is manually mounted, files are manually moved, and permissions are manually set, correct?

nachoparker, 2018-11-07:
I guess so, nobody has added support for it, so you will have to do things manually where NCP fails.

It would be a great addition a little guide, wiki article, blog article or even a PR if you can code bash so we can support it natively in NCP. wink wink

wayneoutthere, 2018-11-07:
i'm 99% finished the guide if you want to troubleshoot the last few steps with me! :) I've got UBports community backing me but there are just permission issues.  I was planning to just post it on my personal blog but if you have a better spot for it let me know

nachoparker, 2018-11-07:
that's so awesome!!!!

well, it's up to you. I don't know if you have seen the wiki here, many people contribute there with practical instructions and support.

if you want something more bloggish, you are welcome to guest-publish in the NCP blog ownyourbits.com. I you prefer to publish it on your blog let me know, I will probably want to reference it from ownyourbits

I barely have the time to maintain this, I wish I had more time to play, but I can try to help ;)

what is your blog?

wayneoutthere, 2018-11-07:
ok cool. issue is that i have a few final steps that are not quite working with permissions and that's where i'm stuck, i know i'm close.  can you PM me on matrix or telegram? @wayneoutthere:matrix.org or @wayneoutthere respectively. then i'll get it rolling and then wiki-post it for you or send you the markdown

nachoparker, 2018-11-08:
sure, you can find me on the telegram group https://t.me/NextCloudPi