nextcloud/nextcloudpi #258: Repeat of "Log in window" via NCP-WebGui

user-jan, 2017-10-15:
Hi Nacho,

Whenever I try to log in to the NCP-Web-Gui (log-in window) with the "correct" log-in name and password it repeats the log-in window so that I have to type the same info again until letting me make changes such as NCP-update...

[image]

[image]

I'll be out now for lunch :-)

Best,
Jan

What version of NextCloudPi are you using? ( eg: v0.17.2 )

v0.31.14

What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

Expected behavior

Connect to Nextcloud via https://19x.16x.x.x:4443 through log-in window

Actual behaviour

Can't log in the 1st time, only the second or third time

testbird, 2017-10-17:
Looks like this is already fixed for the next version:
https://github.com/nextcloud/nextcloudpi/commit/e3ef9730bf54d68cb04a009db6b2786c190eabc9

user-jan, 2017-10-18:
Hi testbird, Hi Nacho,
On my MacBook I can now log in to the webGui after the 1st time. 
But when I try to log in to the webGui on my iPad mini I have to fill in the log-in window 4 times.
How strange.
Best Regards,
Jan

nachoparker, 2017-10-18:
Hi guys,

The issue was badly patched in the past, but I will post a permanent (and better) solution soon.

nachoparker, 2017-10-20:
Hi, I improved the design of permissions for ncp-web, and I think that the issue is gone for good now.

Please, tell me if this is still happening after `v0.31.15`

closed 2017-10-20 by nachoparker

user-jan, 2017-10-20:
Brilliant, thank you Nacho 😊