nextcloud/nextcloudpi #259: Stuck in Maintenance Mode

cottoa00, 2017-10-17:
I'm stuck in Maintenace mode...

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.31.14

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

cat /usr/local/etc/ncp-baseimage
replies "untagged"
older version... had to update Apache config via your script... but it worked to get Web-admin page running

### Actual behaviour
Was trying to get the video player close icon "X" to appear... blogs suggested was fixed and latest version should fix...updated apps and then tried to update nextcloud itself... BUT from within nextcloud ... not ncp-update (had already done that). Was running through sequence and got to backup and choked... now I'm stuck in maintenance mode. Did reboot... hard shutdown, etc... NADA.

How can I turn maintenance mode off at the command line?

NOTE: logs below reflect repeated  attempts by me to get into the web-ui to see if could fix things from there. apparently I did not record my changed credentials;-(

### Include logs
<details>
<summary>Nextcloud logs</summary>

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of  /var/www/nextcloud/data/nextcloud.log

sudo tail -n 100 /var/www/nextcloud/data/nextcloud.log
{"reqId":"WeSiA8CoAZcAAFNY8pgAAAAT","level":2,"time":"2017-10-16T12:11:47+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSie8CoAZcAAFNY8pkAAAAU","level":2,"time":"2017-10-16T12:13:47+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSi9MCoAZcAAFNY8poAAAAV","level":2,"time":"2017-10-16T12:15:48+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSjbMCoAZcAAFNY8psAAAAW","level":2,"time":"2017-10-16T12:17:48+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSj5cCoAZcAAFNY8pwAAAAX","level":2,"time":"2017-10-16T12:19:49+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSkXcCoAZcAAFNY8p0AAAAY","level":2,"time":"2017-10-16T12:21:49+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSk1sCoAZcAAFNY8p4AAAAA","level":2,"time":"2017-10-16T12:23:50+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSlT8CoAZcAAFNY8p8AAAAB","level":2,"time":"2017-10-16T12:25:51+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSlx8CoAZcAAFNY8qAAAAAC","level":2,"time":"2017-10-16T12:27:51+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSmQMCoAZcAAFNY8qEAAAAD","level":2,"time":"2017-10-16T12:29:52+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSmuMCoAZcAAFNY8qIAAAAE","level":2,"time":"2017-10-16T12:31:52+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSnMcCoAZcAAFNY8qMAAAAF","level":2,"time":"2017-10-16T12:33:53+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSnqcCoAZcAAFNY8qQAAAAG","level":2,"time":"2017-10-16T12:35:53+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSoIsCoAZcAAFNY8qUAAAAH","level":2,"time":"2017-10-16T12:37:54+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSomsCoAZcAAFNY8qYAAAAI","level":2,"time":"2017-10-16T12:39:54+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSpE8CoAZcAAFNZQJ8AAABP","level":2,"time":"2017-10-16T12:41:55+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSpi8CoAZcAAFNZQKAAAABQ","level":2,"time":"2017-10-16T12:43:56+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSqBMCoAZcAAFNZQKEAAABR","level":2,"time":"2017-10-16T12:45:56+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSqfMCoAZcAAFNZQKIAAABS","level":2,"time":"2017-10-16T12:47:57+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1; WOW64; Trident\/7.0; rv:11.0) like Gecko","version":"12.0.0.29"}
{"reqId":"WeSq9cCoAZcAAFNZQKMAAABT","level":2,"time":"2017-10-16T12:49:57+00:00","remoteAddr":"79.153.95.2","user":"--","app":"core","method":"POST","url":"\/","message":"Trusted domain error. \"79.153.95.2\" tried to access using \"100.14.82.22\" as host.","userAgent":"Mozilla\/5.0 (Windows  [... cut ...]
192.168.1.151:4443 192.168.1.47 - pi [17/Oct/2017:13:07:06 -0400] "GET /ncp-logo.png HTTP/2.0" 401 464 "-" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:54.0) Gecko/20100101 Firefox/54.0"
192.168.1.151:4443 192.168.1.47 - - [17/Oct/2017:13:07:19 -0400] "GET /ncp.js HTTP/2.0" 401 464 "-" "Mozilla/5.0 (X11; Ubuntu; Linux x86_64; rv:54.0) Gecko/20100101 Firefox/54.0"
</details>
<details>

summary>mariaDB logs</summary>

```
Paste the output of `systemctl status mysqld`
Paste the output of `tail -n 100 /var/log/mysql/*.log`
```
</details>

Shadowstreik, 2017-10-18:
> sudo -u www-data php /var/www/nextcloud/occ maintenance:mode --off

I keep this in a text file for safekeeping. I haven't needed it, luckily.

nachoparker, 2017-10-18:
thanks @Shadowstreik

closed 2017-10-18 by nachoparker

cottoa00, 2017-10-18:
thanks @Shadowstreik... I'm back in business ;-)