nextcloud/nextcloudpi #260: File system not expanded, backup doesn't check available local space, inconsistent state

giddyhup, 2017-10-18:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

v0.31.1 

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

NextCloudPi_09-12-17

### Expected behavior

File system expansion during setup. Backup checks local disk space (if writing locally) or does not write locally at all.

### Actual behaviour

I don't know if during setup file system expansion should have happened automatically or if there was a question about it. In my case it wasn't expanded, I might have missed the question.

I started a backup to USB drive from the web console and apparently the local file system ran over. I now expanded the file system but the nextcloud appears in the original setup state as if the databases weren't there. The NextCloudPi web console isn't up. How can I recover?

### Steps to reproduce, from a freshly installed image

Use non-expanded image. Try backup to USB drive.

### Include logs
<details>
<summary>Nextcloud logs</summary>
[only contain web access data]
</details>
<details>
<summary>Apache logs</summary>
[only contain web access data]
</details>
<details>
<summary>mariaDB logs</summary>

Paste the output of `systemctl status mysqld`

 mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2017-10-18 06:00:47 UTC; 33min ago
  Process: 934 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 931 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 684 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 524 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_START_POSITION=$V
  Process: 516 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 490 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 901 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─901 /usr/sbin/mysqld

Oct 18 06:00:33 mycloud systemd[1]: Starting MariaDB database server...
Oct 18 06:00:45 mycloud mysqld[901]: 2017-10-18  6:00:45 1988497408 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 901 ...
Oct 18 06:00:47 mycloud systemd[1]: Started MariaDB database server.

Paste the output of `tail -n 100 /var/log/mysql/*.log`

empty :(
</details>

nachoparker, 2017-10-19:
If I am not mistaken, the filesystem expands on the first boot.

It would be good to include a free space check.

What do you mean _as if the databases weren't there_? The backup does not modify anything, just copies stuff around.

> The NextCloudPi web console isn't up

is php up?

```
systemctl status php7.0-fpm
```

nachoparker, 2017-10-19:
this should be a straightforward enhancement of the code, if somebody wants to help with this, here is the code for `nc-backup`

https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/nc-backup.sh

nachoparker, 2017-10-31:
I fixed both things

 - expand filesystem at first boot v0.31.28
 - check for available space before backup v0.31.27

closed 2017-10-31 by nachoparker

giddyhup, 2017-11-04:
Thanks for fixing this, @nachoparker. Please accept my apologies that I didn't react to questions. I was on the orad.

nachoparker, 2017-11-04:
you are welcome, thanks for your feedback