nextcloud/nextcloudpi #261: UPnP activated but forwarding ports fails

JVAQUEROM, 2017-10-18:
Hi,

I changed my internet company (Orange now) and the new router has UPnP active by default (UPnP IGD..). However, nc-forwarding-ports fails with this message:

```
Launching nc-forward-ports
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.1.1:62833/de9d00f7/IGDV1/rootDesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.1.1:62833/de9d00f7/rootDesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.1.1:62833/de9d00f7/ctl/IPConn
Local LAN ip address : 192.168.1.222
UPNP_DeletePortMapping() returned : 0
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.1.1:62833/de9d00f7/IGDV1/rootDesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.1.1:62833/de9d00f7/rootDesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.1.1:62833/de9d00f7/ctl/IPConn
Local LAN ip address : 192.168.1.222
UPNP_DeletePortMapping() returned : 0
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.1.1:62833/de9d00f7/IGDV1/rootDesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.1.1:62833/de9d00f7/rootDesc.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

Found valid IGD : http://192.168.1.1:62833/de9d00f7/ctl/IPConn
Local LAN ip address : 192.168.1.222
ExternalIPAddress = 92.186.168.69
AddPortMapping(443, 443, 192.168.1.222) failed with code 606 (Action not authorized)
GetSpecificPortMappingEntry() failed with code 714 (NoSuchEntryInArray)
```

nachoparker, 2017-10-18:
Seems like your router needs some extra authorization for UPnP to work, or just a broken implementation.

Not much we can do about that

See https://github.com/lontivero/Open.NAT/issues/30

closed 2017-10-18 by nachoparker

JVAQUEROM, 2017-10-20:
I finally figure out how to do it manually. The router webpage has a menu under Advanced configuration, NAT/PAT, and there you can activate different services/apps, so one selects "secure web server (https)" and then the raspberrypi in "device". All good!

nachoparker, 2017-10-23:
great, glad that it works.

NCP can try UPnP but the router has to support it and have it activated. The prefered method is still to configure it manually like you did.