nextcloud/nextcloudpi #266: 32GB SD Card only has 2GB usable - how to resize?

chapman57, 2017-10-23:


PunkRa1n, 2017-10-23:
sudo raspi-config -> 7 Advanced Options -> A1 Expand Filesystem 
Reboot - Done

chapman57, 2017-10-23:
Huh - never thought of raspi-config. Is that still there in the nextcloudpi image? Need that option in the nextcloudpi-config utility

Thanks,
Don

PunkRa1n, 2017-10-23:
Yes, it is. The lastest image (10-02-17) is based on "Raspbian GNU/Linux 9 (stretch)"
You can jump from raspi-config directly to nextcloudpi-config (fist option "0 NextCloudPi Configuration")

If you need more information like Distribution Name, Distribution Release Version, Release Notes, Distribution Codename and Distribution Description...
type:
cat /etc/*-release

nachoparker, 2017-10-23:
thanks @PunkRa1n 

That is correct. I thought that the filesystem was automatically expanded though. 

At least it was last time I tried

nachoparker, 2017-10-23:
For technical (or any) questions, please try to use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) in the future ;)

chapman57, 2017-10-23:
Thanks guys. Worked like a champ. I got the SD card expanded and an external 250GB SSD drive configured and everything moved over.

I am very impressed with what you have here and I can finally pull everything off of dropbox and I have a TON more space!

I wish I could contribute! I do write code, C#, VB.Net and Android - Java  (beginner at Android) but I don't know linux.

nachoparker, 2017-10-23:
great, that's the idea! make it easy for everyone to host their own stuff.

Well, some contributors with web development skills would be really useful!

Also helping others, testing, reporting are [good ways to help](https://ownyourbits.com/2017/07/07/how-can-you-help-own-your-bits/). You can also have a look at github issues with the tag 'help wanted' to see if they fit you ;)

closed 2017-10-23 by nachoparker

chapman57, 2017-10-23:
I'll take a look. :)