nextcloud/nextcloudpi #267: Can't create new users

PunkRa1n, 2017-10-23:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
NextCloudPi v0.31.15
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_10-03-17
### Expected behavior
Create a new user
### Actual behaviour
If I hit create, a frame appears "Authentication required"
I type my admin password to confirm and nothing happens 
### Steps to reproduce, from a freshly installed image
Create a new user
### Include logs
<details>
<summary>Nextcloud logs</summary>

```
{"reqId":"We4jwH8AAQEAAANGgdAAAFIT","level":2,"time":"2017-10-23T17:15:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4j3n8AAQEAAANFoYEAABMN","level":2,"time":"2017-10-23T17:16:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4j-H8AAQEAAANGgdEAAEsB","level":2,"time":"2017-10-23T17:16:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kA38AAQEAAANGgdIAAEQU","level":2,"time":"2017-10-23T17:16:51+00:00","remoteAddr":"192.168.111.205","user":"--","app":"no app in context","method":"POST","url":"\/index.php\/heartbeat","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kGn8AAQEAAANFoYIAABQF","level":2,"time":"2017-10-23T17:17:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kOH8AAQEAAANGgdMAAEgR","level":2,"time":"2017-10-23T17:17:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kVn8AAQEAAANFoYMAABUJ","level":2,"time":"2017-10-23T17:18:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kdH8AAQEAAANGgdQAAEcI","level":2,"time":"2017-10-23T17:18:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kkn8AAQEAAANGgdUAAFMX","level":2,"time":"2017-10-23T17:19:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4ksH8AAQEAAANFoYQAABYP","level":2,"time":"2017-10-23T17:19:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4kz38AAQEAAANGgdYAAEYO","level":2,"time":"2017-10-23T17:20:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4k7H8AAQEAAANGgdcAAFQA","level":2,"time":"2017-10-23T17:20:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4lCn8AAQEAAANGgdgAAEUF","level":2,"time":"2017-10-23T17:21:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4lKX8AAQEAAANGgdkAAFUK","level":2,"time":"2017-10-23T17:21:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4lR38AAQEAAANFoYUAABcQ","level":2,"time":"2017-10-23T17:22:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4lZX8AAQEAAANFoYYAABgI","level":2,"time":"2017-10-23T17:22:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4lgn8AAQEAAANGgdoAAFYP","level":2,"time":"2017-10-23T17:23:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4loH8AAQEAAANFoYcAAAAH","level":2,"time":"2017-10-23T17:23:45+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notifications","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:45.0) Gecko\/20100101 Firefox\/45.0","version":"12.0.3.3"}
{"reqId":"We4lvn8AAQEAAANFoYgAAAIM","level":2,"time":"2017-10-23T17:24:15+00:00","remoteAddr":"192.168.111.205","user":"admin","app":"no app in context","method":"GET","url":"\/ocs\/v2.php\/apps\/notifications\/api\/v2\/notification [... cut ...]
2017-10-23 17:28:10 1988677632 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 11748903
2017-10-23 17:28:11 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-10-23 17:28:11 1988677632 [Note] Plugin 'FEEDBACK' is disabled.
2017-10-23 17:28:11 1988677632 [Note] Server socket created on IP: '127.0.0.1'.
2017-10-23 17:28:11 1988677632 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

nachoparker, 2017-10-25:
Duplicate of #11

nachoparker, 2017-10-25:
Look at #11, probably you have the wrong date

closed 2017-10-25 by nachoparker

PunkRa1n, 2017-10-25:
after reconfigure via raspi-config it works great again.
I know, one of the first things when you setup a Pi or a machine, CHECK YOUR DATE...

Thank you very much...