nextcloud/nextcloudpi #269: wiki: backup and restore a NCP instance process

nachoparker, 2017-10-25:
It is common for people to want  to migrate to a new Pi or upgrade the base system. 

This would be a nice addition to the wiki

ovpc, 2017-12-21:
[Backup](https://docs.nextcloud.com/server/12/admin_manual/maintenance/backup.html)
and
[Restore](https://docs.nextcloud.com/server/12/admin_manual/maintenance/restore.html)
are already documented. 
You want a NCP specific step-by-step ?

nachoparker, 2017-12-21:
well, I was more thinking about the steps using `nc-backup`, `nc-restore`, `nc-import` and `nc-export`

ovpc, 2018-01-03:
Started [wiki](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-nextcloudpi-config) and listing screenshots used here for practical purpose only.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

nachoparker, 2018-01-03:
great thanks! tell us when you are done, and so we can review it

ovpc, 2018-01-08:
Adding some more screenshot for nc-backup with ncdata
[image]
[image]
[image]

+ screenshots for nc-restore process

[image]
[image]
[image]

nachoparker, 2018-02-05:
thanks for the documentation @ovpc !

closed 2018-02-05 by nachoparker

voldemort13, 2018-04-26:
How do I know where the backup file is going? I have an usb stick connected to raspberry and the backup works fine but i cant find the generated file anywhere

Oceania2301, 2018-12-06:
Where can I find information about to restore data using nc- snapshots ?

ovpc, 2018-12-06:
[Here](https://ownyourbits.com/category/btrfs/) is what Nacho has added to his blog concerning btrfs and snapshots. There is no wiki page for that yet......

nachoparker, 2018-12-07:
> Where can I find information about to restore data using nc- snapshots ?

In order to restore the data from a snapshot, you have to replace the `ncdata` subvolume, something like

```
sudo btrfs subvolume delete /media/USBdrive/ncdata   # remove datadir
sudo btrfs subvolume snapshot /media/USBdrive/ncp-snapshots/daily-xxxxx /media/USBdrive/ncdata                      # restore old copy
nc-scan                                       # update NC database with the new/old files
```

It would be great to add that to the wiki

ovpc, 2018-12-07:
Made start at this [here](https://github.com/nextcloud/nextcloudpi/wiki/How-to-snapshot-with-btrfs) will add steps to enable snapshot auto later.

kpleines, 2018-12-28:
I've run NextcloudPi inside Docker with an exportet /data volume.
Backup seems to run smoothe but ends with the following error message:

tar: /data/app/data/..: Cannot open: No such file or directory

Can I fix this problem myselfe? Thx
[image]

nachoparker, 2018-12-28:
this might have been already fixed. Please update to the latest version (`v0.67.11`) and try again.

StevenDecatur, 2019-02-02:
This might be obvious to people who know what they are doing, but what is the purpose of backing up the configuration file with nc-export-ncp when you restore the instance with the file made by nc-backup?  Do you need the configuration backup file made by nc-export-ncp when restoring?

Also, if you restore from a dataless backup, do you just copy the data back to the data directory when it finishes? For example, if you used rsync to save the data elsewhere?

ovpc, 2019-02-02:
nc-export-ncp exports the setting of NCP only (app enabled/disabled/settings), not NC's config, maybe wiki does need clarification, feel free to edit/add information to improve usability, thank you

You don't need it, to restore from a backup made previously. 
After restore from data less backups: you can set /path/to/ncdata in config and run nc-scan

kpleines, 2019-02-03:
Update to the latest Version works. Thx

SoongJr, 2021-07-26:
Thanks for the guide!
I'm trying to replace my old ncp image with a docker instance on my RaspberryPi 2, so I made a backup of the old one as per the guide, flashesd a new card with default Raspberry OS, started up the docker container and restored the data.

I've had some issues I could work around myself, so now restore has succeeded, but my SSL certificate does not seem to have been restored. Is this not part of the backup? I thought controlling the whole environment was the reason for having NCP int he first place, so all the DynDNS and SSL and firewall stuff should be part of a backup, no?
Am I just using the wrong kind of backup and should have done nc-export-ncp or something or do I have to manually transfer my ssl certificates? What else do I have to do manually that one could expect ncp-backup to be doing for me? 😉

ovpc, 2021-07-26:
@SoongJr Please use the [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33) to ask questions or if need assistance.

SoongJr, 2021-07-26:
> @SoongJr Please use the [forum](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm/33) to ask questions or if need assistance.

OK thanks, will do. The guide itself said to "please comment or ask for help HERE", so that's what I did ;)

johnmcd88, 2022-12-11:
Question - when you choose to include data in your ncp-backup, do external folders from network shares that you've added to NCP get backed up?

Thanks

ailieff, 2023-01-06:
> Question - when you choose to include data in your ncp-backup, do external folders from network shares that you've added to NCP get backed up?
> 
> Thanks

That is something that I was wondering too? Or maybe just the reference to the files/mounts and you have to restore the configuration, too.