nextcloud/nextcloudpi #270: Request: DynDNS support for accounts at spdns.de

TimmThaler, 2017-10-27:
There are already a lot of dyndns services that can be configured within the setup.
Nevertheless, I really would appreciate, if you could add a script for accounts at *spdns.de*

I already found a python script by *gbausch* here on github and perhaps you could integrate that dyndns service, too.

Here come the link to the python script: https://github.com/gbausch/spdnsUpdater

Thank you very much.

nachoparker, 2017-10-29:
Hi,

Well, I don't know how popular that service is but it is fine by me to include it.

The only thing is that somebody that actually uses that service needs to maintain it, I can't keep working things that I don't use ;)

PRs are welcome, see some [docs on adding NCP options](https://github.com/nextcloud/nextcloudpi/wiki/Developer-guide), and ask anything ;)

TimmThaler, 2017-10-30:
Hey,
I don't know how popular that service is, either, but for german users it might be very interesting, because this provider follows the german laws and restrictions for privacy etc.

The script I mentioned in the opening post has been updated over a year ago. So actually there were no changes in the last year.
Perhaps it would be possible to regularly check for changes in the scipt's repository and then update or copy from there, as I think, the original creator uses his own script and wants it up to date and working.

When I find some time, I will read the guide for the NCP options 😄

nachoparker, 2017-10-30:
cool, it's easy to add something to NCP, you can check the other DDNS services as a starting point. 

Ask anything :D

TimmThaler, 2017-10-30:
Alright, I'll do my best 😆

TimmThaler, 2017-10-30:
Okay,
I tried to create a shell script from the original python script.
It works on my computer very well.

I also tried to write a script for the setup by taking the [duckDNS.sh](https://github.com/nextcloud/nextcloudpi/blob/master/etc/nextcloudpi-config.d/duckDNS.sh) as a basis for that.

Perhaps somebody could test my setup file and tell if it works. I would appreciate that!

Load my spdnsUpdaterNCP.sh script here:
https://github.com/TimmThaler/spdnsUpdater

nachoparker, 2017-10-31:
Nice!

Looks good... two little things

The NCP extra needs to be one single file, so you can either download the python script in the install() step or, even better, include it in the install step as a HERE document

Have you run it from nextcloudpi-config or ncp-web?

TimmThaler, 2017-10-31:
Hey,
okay, I changed that.
Originally I thought, I could download the script every time new when the updater is set up new to always have an up-to-date version.

Now I write the sh script during the install() step to a file and call it in the configure() step:
[https://github.com/TimmThaler/spdnsUpdater/blob/master/spdnsUpdaterNCP.sh](https://github.com/TimmThaler/spdnsUpdater/blob/master/spdnsUpdaterNCP.sh)

Unfortunately I do not own a NextcloudPi instance at the moment, that's why I asked for testing of the setup script. I know that the actual dyndns update script works properly.

nachoparker, 2017-10-31:
sweeet, looks good to me ;)

Let's see if someone has spdns and is willing to try it 

thank you!

TimmThaler, 2017-10-31:
Cool 😎 
It would be great, if someone could just test the setup/config script. I'm pretty sure that the updater script will work 😆  Even if one doesn't have an account at spdns the script could be tested with some random variables. The response will just report an error, but that's fine.

You're welcome!

nachoparker, 2017-10-31:
I test stuff with QEMU so I don't need to use a RPi. Also, I am working on a x86 docker for development

In the ownyourbits page you have the code to run it in QEMU

TimmThaler, 2017-10-31:
Excellent. I'm gonna test this.

TimmThaler, 2017-10-31:
Alright. It works!
I just moved the script creation from the install step to the configuration step. When I just copied my NCP script to the config folder and updated via nc-update my update script wasn't created.
On the other hand now the update script only exists when the 'active' option is set to yes.

From my point of view *everything works* ! 😄

nachoparker, 2017-11-01:
Great, thanks!!

Do you want me to grab it from your repo or do you want to send a PR? If you send PR your name will be in the contribution commit. Please, PR to the devel branch.

Please, also move the creation to the install() step again, it will be created only once when we upload them to the repo and ncp-update is executed.

TimmThaler, 2017-11-01:
No problem :)

So I just moved the creation back to the install() step again and removed the lines that delete the script created during the install() step, so only the cron file will be created new or will be deleted.

Now I hope, I created the PR correctly.

nachoparker, 2017-11-02:
Cool, thanks

The good think about the PR is that the contribution will be under your name, as it should be.

If you have some time, I recommend that you learn a bit how to do a PR, it's not that hard but maybe a bit confusing the first time.

You have to hit the PR button (remember, on the devel branch, not master), and then you get a _fork_ or a copy of the code. 

Then, upload your file to `usr/local/etc/nextcloudpi-config.d` in your copy, and _commit_ the changes (with a comment 'added spdyn.de'). 

Then, you submit the PR and I will be able to see your changes before accepting them.

If it's too confusing it's ok, tell me and I'll include your script myself. Ask anything, we can try however many times we want ;)

I'll close your PR now, you can try again

If it helps, see section 'Creating a Pull Request' [here](https://yangsu.github.io/pull-request-tutorial/)

TimmThaler, 2017-11-02:
😆 Okay, so I tried again:

1. I created a fork of the repo.
2. Uploaded / created the file in nextcloudpi/etc/nextcloudpi-config.d/
3. Made a new branch in my fork repo
4. Sent a PR with comment "added spdns.de DynDNS service"

Was that correctly done? 😆

nachoparker, 2017-11-02:
almost! you created a PR of your own fork.

You have to 

1. click PR in this repo (my repo). This automatically creates a fork
2 in this PR/fork, upload the file like you did
3. send the PR

You can see that the PR appears in your copy of the repo, but not in mine.

We're close!

TimmThaler, 2017-11-02:
Oh, ok...
I tried it again 🙈

nachoparker, 2017-11-02:
Sweet! that's good now

We can continue the conversation on the PR itself

thanks a lot

closed 2017-11-02 by nachoparker