nextcloud/nextcloudpi #271: Install and boot NCP from an external USB drive with Berryboot

nachoparker, 2017-10-29:
This issue is to separate the conversation that started here https://github.com/nextcloud/nextcloudpi/issues/209

@ovpc has implemented the port to berryboot/squashfs and is in conversation with them to include the image.

Also, he is going to write instructions in the wiki

Thanks for all that ;)

ovpc, 2017-10-29:
Just screenshots for wiki in progress ;-)
[image]
[image]

ovpc, 2017-10-30:
An issue :( with berryboot image: cant run : sudo raspi-config, it throws error that /boot is not mounted and only option is exit ...nextcloudpi-config runs tho. Will post at maxnet with screenshot, they got theirs working on NC11.img, /boot is on sdcard managed by berry. That doent seem to work with raspi-config out of the box......

nachoparker, 2017-10-31:
looks great, thank you very much!

ovpc, 2017-11-01:
Uncommenting the line for /boot in /etc/fstab and pointing it LABEL=boot resolved the issue
[image]
BTW Had not noticed yet there is a link to nextcloudpi-config inside raspi-config, nice job !

Please grab berryboot NCP torrent from 
[HERE](https://drive.google.com/open?id=0B1wGkGPW-CxLeXFnUWNlb09oMWc)
and help us seed, thank you

nachoparker, 2017-11-01:
good job! thanks

closed 2017-11-01 by nachoparker