nextcloud/nextcloudpi #272: New wizard UI/UX

Espina2, 2017-10-29:
I started to design an new interface for the wizard. 

You can see everything in the link above. Please comment and write what you think about it.

https://www.figma.com/proto/gaktdvehsLpLbdXMxQPoQ7nd/NEXTCLOUD-PI?node-id=1%3A2&scaling=contain

Daniel-Krebs, 2017-11-01:
Looks pretty elegant & modern, I like it, Especially the "Let's Go" Button. In the current Wizard you can proceed by clicking on the "USB Configuration" Tab or the nextcloudpi logo, neither of it ist intuitive in my opinion. Are you going to rework other parts of the UI as well?

Espina2, 2017-11-01:
@Daniel-Krebs First I need to get this done, just after I can evaluate if I time to rework the other UI. But I guess it will do it, I just don't know when I have the time to do it. If you wanna help in something let me know, it will be very much appreciated.

Daniel-Krebs, 2017-11-02:
@Espina2 Sure, while I have basic understanding of HTML and CSS I'm not sure if I can be of any help for actual coding but I'm glad to help in conceptualisation and testing / feedback. I'm just getting into NCP but I think the whole webinterface needs some redesign. The alphabetical sorting doesn't work well with 20+ items, most of them just starting with "nc-*" and feels rather confusing IMHO.

nachoparker, 2017-11-02:
Love it @Espina2 , good job!

rencire, 2017-12-29:
Hi, I am open to helping out as well.  @Espina2, do you have more mocks for the rest of the setup flow?

nachoparker, 2017-12-29:
great @rencire, thanks!

Espina2, 2018-01-03:
Hi @rencire, I think just one or two more mockups are needed, because it always the same things. But if you think you will need more just say me which ones you need that I make the mock straight away. 

I have some things already done here, 
http://git.paulomoura.com.pt/root/nextcloudpi

Feel free to start wherever you want. Thanks in advance

nachoparker, 2018-01-03:
yes, @rencire feel free to ask. I created the backend and the little API... you should be fine using what we already have as a reference, but feel free to ask.

We can discuss here, also we have a telegram group if you are interested

https://help.nextcloud.com/t/nextcloudpi-telegram-group/25001

VagishVela, 2018-06-11:
Something else that would be nice is using the wording "USB/HDD" for Odroid HC2, as that one can do a 3.5 inch hard drive.

nachoparker, 2018-10-27:
let's move this to https://github.com/nextcloud/nextcloudpi/issues/622

closed 2018-10-27 by nachoparker