nextcloud/nextcloudpi #276: File locking problems with big data transfers from sync client, and the need for Redis

nachoparker, 2017-11-02:
Are other users seeing this?

https://help.nextcloud.com/t/nextcloudpi-stops-responding-when-using-nextcloud-win7-x64-client/

We might need to install Redis after all.

Thoughts?

Shadowstreik, 2017-11-03:
I'd have to say yes, but it is unconfirmed. My encounters were over the Summer and I didn't bother to check the logs. My NCP instance is very inactive since it's an archive for family media.

But, some indo that may, possibly, be of assistance:
- large groups of smal files don't appear to cause any issues.
- large groups of large video files (MKV, MP4, etc) do cause a 'lockup' and takes more than 1-2 days to complete.

My workaround was to simply not sync up video files(VHS, 8mm cassette conversions of home movies). I keep those aside and keep them available, offline, for family should they ask for any. Though I was able to sync videos of old 8mm films that were converted years ago without any issue.

nachoparker, 2017-11-10:
After quite a bit of testing, and the help from edson for testing, we have decided to include redis caching for the next release to account for this heavy usage cases.

Resource consumption is a bit higher but upload performance for many small files, specially with the sync client is way better.

closed 2017-11-10 by nachoparker