nextcloud/nextcloudpi #278: LetsEncrypt Renewal Notices and auto-renewal.

Shadowstreik, 2017-11-04:
I have received two LetsEncrypt renewal notifications via email. Here is a piece of the latest...

> Your certificate (or certificates) for the names listed below will expire in
> 9 days (on 13 Nov 17 18:27 +0000). Please make sure to renew
> your certificate before then, or visitors to your website will encounter errors.
> 

When should I expect NCP to handle the renewal?

nachoparker, 2017-11-05:
Did you use ncp-web to register more than one domain?

Shadowstreik, 2017-11-05:
Naw, just the one domain.

Shadowstreik, 2017-11-05:
I need to mention that I only use port 443 as 80 is being used on another PI for Craftbian's developer.

Pant, 2017-11-05:
Hi @Shadowstreik,

> I need to mention that I only use port 443 as 80 is being used on another PI for Craftbian's developer.

This is a problem. Letsencrypt uses ports 443 and 80 to create a certificate and verify it.
A workaround to this would be to use a ddns service, and create a forward to the ports the nextcloudpi uses.

nachoparker, 2017-11-05:
Yes, that's what it is then. You will have to find a way to open that, or do it manually with another method (like DNS challenge), but that's more advanced

closed 2017-11-05 by nachoparker

Shadowstreik, 2017-11-05:
> -------- Original Message --------
> Local Time: November 5, 2017 12:34 PM
> UTC Time: November 5, 2017 5:34 PM

What I'll do is forward 80 to be the same as 443 so the auto-renewal can occur. I'll just change it back, afterwards.