nextcloud/nextcloudpi #283: Installation on already existing Ubuntu

cryptodigger, 2017-11-07:
I was wondering if I can also install NextcloudPi on my already existing Ubuntu 16.04 distro?

Pant, 2017-11-07:
You can try: `curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash`. 
It requires an arm Debian distro. It is only tested roughly on Raspbian and Debian, so we do not know if it is going to work on Ubuntu. 
If you do try it, please share some feedback.

nachoparker, 2017-11-07:
thanks @Pant ,

Yes, this is based on Debian 9

If anybody wants to test this against Ubuntu, you would have to comment out [this distro check](https://github.com/nextcloud/nextcloudpi/blob/master/install.sh#L24)

I'll reopen and retag in case anyone wants to experiment with this

cryptodigger, 2017-11-07:
Thanks for the replies. In case I move over to Debian 9. What would be the command to download NextcloudPi into my existing Debian distro?

nachoparker, 2017-11-07:
I'm waiting for the next release to announce this, but it would be

```
# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
```
You should not trust me, and read the script first ;)

derdershat, 2017-11-11:
hm I tried your install.sh on a fresh Raspbian Stretch Full but  sudo nextcloudpi-config gives me command not found 

https://192.168.78.24:4443/ gives me ERR_CONNECTION_REFUSED

btw you have a typo in install.sh "sudo nextcoludpi-config"

nachoparker, 2017-11-12:
@derdershat 

Hi,

Like I said, this has not been announced, so it is not final. I have been making changes today and yesterday.

I tested this five minutes ago and it worked with a small issue: I had to reboot the pi for Nextcloud to be available.

I'll solve that next.

In any case, please report that on a different issue if you run into problems again because it is unrelated to Ubuntu, and if you want to try it again feedback will be more than welcome ;)

edit: and please include logs if something fails!

derdershat, 2017-11-12:
@nachoparker 

I used the curl command again today and yippy yeah it works! 

This is a really big step for all inexperienced users out there who wants to use Nextcloud (on top of their already installed systems)

thank you very much for all your work

nachoparker, 2017-11-12:
@derdershat 

great!

thanks for the feedback.

 I think that it is still easier to use the SD card (maybe the docker image) than this, but for other boards/architectures it's a good option

sgiannulli, 2017-11-19:
installed on Debian 9.2.1 server machine....up to now everything up and working
great job....thank you very much!!

nachoparker, 2017-11-19:
@sgiannulli great! good to know, tell us how it goes

nachoparker, 2017-12-04:
Well, we cannot support more linux flavours, so if nobody is willing to take Ubuntu, we'll close this for now.

Please, reopen if anyone is interested in helping with this.

closed 2017-12-04 by nachoparker

CalumFreeman, 2018-10-17:
Hi, I thought I'd try to get nextcloud running on ubuntu(Desktop). Sorry this is a little sparse, I don't have much time at the moment, and this post is partly for me so I can try again in a couple of days.

## what I tried:
I installed a fresh(but old) version of Ubuntu 16.04.3 Desktop, then downloaded and ran the install script with these commands:
```
cd ~/Downloads
mkdir nextcloud
cd nextcloud
wget https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh
```
I removed OS check (as suggested [here](https://github.com/nextcloud/nextcloudpi/issues/283#issuecomment-342566044)) before running the next commands
```
chmod +x ./install.sh
sudo ./install.sh
```

This threw some errors...
```
WARNING: The following packages cannot be authenticated!
  debian-archive-keyring libc-l10n locales libc6-dbg libc-dev-bin libc6-dev
  libc6 libc-bin libzstd1 libapt-pkg5.0 libnettle6 libhogweed4 libunistring2
  libidn2-0 libtasn1-6 p11-kit-modules libp11-kit0 libgnutls-openssl27
  libgnutls30 apt-transport-https apt-utils apt cron
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```
As a quick fix, I added `--allow-unauthenticated` to the line which runs `apt install` in the install script, but this didn't clear up all the errors as the install script calls other scripts which I hadn't added this line to. So I manually installed all the packages listed with this type of error.

This is not an ideal solution since allowing unauthenticated packages isn't exactly secure, I'll come back to this when I have time and try to figure out what's going on.

I then re-ran `sudo ./install.sh` and got this error:
```
The following packages have unmet dependencies:
 apache2 : Depends: apache2-bin (= 2.4.34-1) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
so I tried `sudo apt install apache2`, this gave a similar error but with `apache2-bin` instead of `apache2`. `apache2-bin` had the same error but with `libcurl4`.
When I ran `sudo apt install libcurl4` it worked, then I could install `apache2-bin` and then I could install `apache2`.
 ## Finally installs, maybe?
At this point I ran `sudo ./install.sh` again and it ran. The final output was not comforting though. The whole output is too long to list so I'm showing the last 71 lines.

 I'm concerned but the various places where it says nextcloud is not installed. That could just mean I'm supposed to go to 127.0.0.1 on firefox to finish installing, but I'm not sure. In a few days I will re-try installing it again in a less chaotic manner and see where I get to.

```
E: Failed to fetch http://deb.debian.org/debian/pool/main/u/unattended-upgrades/unattended-upgrades_1.6_all.deb  Temporary failure resolving 'deb.debian.org'
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Activating unattended-upgrades
Unattended upgrades active: yes (autoreboot yes)

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Installing DDNS_spDYN
Installing nc-backup
Installing nc-restore
Nextcloud is not installed - only a limited number of commands are available

                                         
  Command "app:install" is not defined.  
                                         
  Did you mean one of these?             
      app:check-code                     
      maintenance:install                
                                         

Nextcloud is not installed - only a limited number of commands are available

                                        
  Command "app:enable" is not defined.  
                                        
  Did you mean this?                    
      app:check-code                    
                                        

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

Nextcloud is not installed - only a limited number of commands are available

                                                                   
  There are no commands defined in the "config:system" namespace.  
                                                                   

NextCloudPi updated to version v0.64.2
Activating nc-init
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value mysql.utf8mb4 set to boolean true
System config value trusted_domains => 1 set to string an.ip.address!
System config value trusted_domains => 5 set to string nextcloudpi.local
System config value trusted_domains => 7 set to string nextcloudpi
System config value trusted_domains => 8 set to string nextcloudpi.lan
System config value mail_smtpmode set to string sendmail
System config value mail_smtpauthtype set to string LOGIN
System config value mail_from_address set to string admin
System config value mail_domain set to string ownyourbits.com
Error: Could not download app calendar
```

ovpc, 2018-10-17:
Hi, thanks for taking time to share. 
Sorry to say, known issue, Ubuntu is not supported (Devs have to much on their plates as it is) but if you have time and skills and wish to take on the support you are welcome to do so ;-)  if not:
Try a clean install of Debian 9 (install ssh & curl) and let the installer do the rest. You can also try the docker image of course.

CalumFreeman, 2018-10-17:
I knew it wasn't officially supported (as you said, devs are busy), are there any particular known issues or is it just "we haven't tried it so it may or may not work"? My skills are extremely limited but I can give it a go, if there are no known issues I'll just try and get it working on my system then see what changed have to be made to get there.

bernd-wechner, 2018-10-17:
As side observation alone, given NextCloudPi provides a full SD card image to run cleanly on a Raspberry Pi from first boot, and given the price of a Raspberry Pi, I find myself puzzling what motivates any great effort to complicate matters by trying to install it on what I presume is a Pi running Ubuntu 16.04 (itself an interesting thing). I'm interested really ... I mean if we were talking a pricey server I'd understand, but that's not a Pi nor likely to be on ARM architecture.

CalumFreeman, 2018-10-17:
My use case is just that I have an old laptop (with proprietary wifi drivers, so easier on ubuntu) and the nextcloudpi seems to be built for security. I don't really have any security experience and the fact that the pi edition guards against things I wouldn't have thought of suggests it is more secure than anything I would set up.

If there are different names for required packages, that could be enough of a headache that it's simpler to sort out the laptop drivers for debian.

nachoparker, 2018-10-18:
@CalumFreeman you can always run it in a Virtual machine or docker x86 on top of Ubintu

CalumFreeman, 2018-10-18:
A VM would work, but, as it turns out the debian installer is better than the last time I tried... So I'm just switching to debian. Sorry to add to the list of people trying to install nextcloudpi on ubuntu then giving up and installing debian.