nextcloud/nextcloudpi #286: White Page and mysql down

sabrahummus, 2017-11-10:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )

v0.34.3

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

untagged

### Expected behavior

Today I noticed my NextcloudPi instance ran out of space. First I tried running apt-get autoremove and apt-get clean. It was still full. Then I noticed I had accidentally been backing up onto my SD Card instead of a USB stick. I deleted the folder. After I restarted my Raspberry I noticed my apps were giving me error 500. When I called up my URL all I saw is a whitepage. So I went to check up on the database. It tells me the service mariadb is down with this

mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2017-11-10 17:10:44 UTC; 5min ago
  Process: 8633 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 8630 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 8532 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set
  Process: 8528 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 8525 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 8633 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Nov 10 17:10:29 raspberrypi systemd[1]: Starting MariaDB database server...
Nov 10 17:10:41 raspberrypi mysqld[8633]: 2017-11-10 17:10:41 1989152768 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process
Nov 10 17:10:44 raspberrypi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Nov 10 17:10:44 raspberrypi systemd[1]: Failed to start MariaDB database server.
Nov 10 17:10:44 raspberrypi systemd[1]: mariadb.service: Unit entered failed state.
Nov 10 17:10:44 raspberrypi systemd[1]: mariadb.service: Failed with result 'exit-code'.

Then I thought I could just backup my nextcloudpi and restore it on a new instance. 

Launching nc-backup
Error reading data directory. Is NextCloud running and configured?
Done. Press any key...

I think I may have somehow corrupted my database or damaged mariadb.

What can I do to either restore nextcloud on this machine or move it intact to another one?

### Steps to reproduce, from a freshly installed image

No idea.

### Include logs

 apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2017-11-10 16:23:18 UTC; 1h 4min ago
  Process: 2237 ExecStop=/usr/sbin/apachectl stop (code=exited, status=0/SUCCESS)
  Process: 2244 ExecStart=/usr/sbin/apachectl start (code=exited, status=0/SUCCESS)
 Main PID: 2249 (apache2)
   CGroup: /system.slice/apache2.service
           ├─2249 /usr/sbin/apache2 -k start
           ├─2253 /usr/sbin/apache2 -k start
           └─2278 /usr/sbin/apache2 -k start

Nov 10 16:23:17 raspberrypi systemd[1]: Starting The Apache HTTP Server...
Nov 10 16:23:17 raspberrypi apachectl[2244]: AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1.
Nov 10 16:23:18 raspberrypi systemd[1]: Started The Apache HTTP Server.

Paste the output of `tail -n 100 /var/log/apache2/*.log`
```
==> /var/log/apache2/access.log <==

==> /var/log/apache2/error.log <==
[Fri Nov 10 06:25:03.001184 2017] [mpm_event:notice] [pid 747:tid 1992278016] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Fri Nov 10 06:25:03.001324 2017] [core:notice] [pid 747:tid 1992278016] AH00094: Command line: '/usr/sbin/apache2'
[Fri Nov 10 11:12:52.124205 2017] [mpm_event:notice] [pid 747:tid 1992278016] AH00491: caught SIGTERM, shutting down
[Fri Nov 10 11:13:08.005472 2017] [mpm_event:notice] [pid 841:tid 1992683520] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[... 251 lines cut ...]
2017-11-10 16:53:41 1988960256 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 199435823
2017-11-10 16:53:41 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-11-10 16:53:41 1988960256 [Note] Plugin 'FEEDBACK' is disabled.
2017-11-10 16:53:41 1988960256 [Note] Recovering after a crash using tc.log
2017-11-10 16:53:41 1988960256 [ERROR] Can't init tc log
2017-11-10 16:53:41 1988960256 [ERROR] Aborting

2017-11-10 17:10:41 1989152768 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-11-10 17:10:41 1989152768 [Note] InnoDB: The InnoDB memory heap is disabled
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-11-10 17:10:41 1989152768 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Using Linux native AIO
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Using generic crc32 instructions
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Completed initialization of buffer pool
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Highest supported file format is Barracuda.
2017-11-10 17:10:41 1989152768 [Note] InnoDB: 128 rollback segment(s) are active.
2017-11-10 17:10:41 1989152768 [Note] InnoDB: Waiting for purge to start
2017-11-10 17:10:41 1989152768 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 199435843
2017-11-10 17:10:41 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-11-10 17:10:41 1989152768 [Note] Plugin 'FEEDBACK' is disabled.
2017-11-10 17:10:41 1989152768 [Note] Recovering after a crash using tc.log
2017-11-10 17:10:41 1989152768 [ERROR] Can't init tc log
2017-11-10 17:10:41 1989152768 [ERROR] Aborting

nachoparker, 2017-11-10:
try to remove `tc.log` from the database directory and restart mysqld

sabrahummus, 2017-11-11:
Worked perfectly! Thank you so much. Your project and support is an inspiration to us all.

closed 2017-11-11 by nachoparker