nextcloud/nextcloudpi #287: Blank screen after update

Dolordo, 2017-11-11:
### PROBLEMS

If you are running into problems, please fill out the following information. For questions or feature requests you don't have to

Keep in mind that many problems come from faulty power sources and corrupted SD cards, so make sure this is not the case for you before reporting.
-->

### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.34.6

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
Old one, 'untagged'

<details>
<summary>Potential issue on Redis ?</summary>
When doing 'fix permission'
PHP Warning:  Redis::connect(): connect() failed: No such file or directory in /var/www/nextcloud/lib/private/RedisFactory.php on line 82
PHP Fatal error:  Uncaught RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(52): Redis->get('9307c17f1ed15a3...')
#1 /var/www/nextcloud/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/Log/File.php(101): spl_autoload_call('OC_User')
#4 [internal function]: OC\Log\File::write('PHP', 'Redis::connect(...', 3)
#5 /var/www/nextcloud/lib/private/Log.php(301): call_user_func(Array, 'PHP', 'Redis::connect(...', 3)
#6 /var/www/nextcloud/lib/private/Log.php(159): OC\Log->log(3, 'Redis::connect(...', Array)
#7 /var/www/nextcloud/lib/private/Log/ErrorHandler.php(68): OC\Log->critical('Redis::connect(...', Array)
#8 [internal function]: OC\Log\ErrorHandler::onShutdown()
#9 {main}
  thrown in /var/www/nextcloud/lib/private/Memcache/Redis.php on line 52

When launching sudo nextcloudpi-config
PHP Warning:  Redis::connect(): connect() failed: No such file or directory in /var/www/nextcloud/lib/private/RedisFactory.php on line 82
PHP Fatal error:  Uncaught RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(52): Redis->get('9307c17f1ed15a3...')
#1 /var/www/nextcloud/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/Log/File.php(101): spl_autoload_call('OC_User')
#4 [internal function]: OC\Log\File::write('PHP', 'Redis::connect(...', 3)
#5 /var/www/nextcloud/lib/private/Log.php(301): call_user_func(Array, 'PHP', 'Redis::connect(...', 3)
#6 /var/www/nextcloud/lib/private/Log.php(159): OC\Log->log(3, 'Redis::connect(...', Array)
#7 /var/www/nextcloud/lib/private/Log/ErrorHandler.php(68): OC\Log->critical('Redis::connect(...', Array)
#8 [internal function]: OC\Log\ErrorHandler::onShutdown()
#9 {main}
  thrown in /var/www/nextcloud/lib/private/Memcache/Redis.php on line 52

</details>

<details>
<summary>Media folder</summary>
I had some issues in the past. So I've ended creating the links per below

drwxr-xr-x  4 root root   4096 Nov 11 16:24 .
drwxr-xr-x 22 root root   4096 Sep 17 12:34 ..
drwxr-xr-x  5 root root   4096 Oct 13 15:42 myCloudDrive
drwxrwxr--  5 root nasgrp 4096 Nov  1 18:52 nas
lrwxrwxrwx  1 root root     12 Nov 11 16:24 USBdrive -> myCloudDrive
lrwxrwxrwx  1 root root     12 Nov 11 16:24 USBdrive1 -> myCloudDrive
lrwxrwxrwx  1 root root     12 Nov 11 16:24 USBdrive2 -> myCloudDrive

</details>
<details>
<summary>Nextcloud logs</summary>
Can't login

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
-> Not relevant, old log from August

</details>
[... 126 lines cut ...]
2017-11-11 17:17:15 1639236416 [Note] InnoDB: Starting shutdown...
2017-11-11 17:17:16 1639236416 [Note] InnoDB: Waiting for page_cleaner to finish flushing of buffer pool
2017-11-11 17:17:18 1639236416 [Note] InnoDB: Shutdown completed; log sequence number 563465178
2017-11-11 17:17:18 1639236416 [Note] /usr/sbin/mysqld: Shutdown complete

2017-11-11 17:17:43 1988739072 [Note] InnoDB: Using mutexes to ref count buffer pool pages
2017-11-11 17:17:43 1988739072 [Note] InnoDB: The InnoDB memory heap is disabled
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Mutexes and rw_locks use GCC atomic builtins
2017-11-11 17:17:43 1988739072 [Note] InnoDB: GCC builtin __atomic_thread_fence() is used for memory barrier
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Compressed tables use zlib 1.2.8
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Using Linux native AIO
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Using generic crc32 instructions
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Completed initialization of buffer pool
2017-11-11 17:17:43 1988739072 [Note] InnoDB: Highest supported file format is Barracuda.
2017-11-11 17:17:44 1988739072 [Note] InnoDB: 128 rollback segment(s) are active.
2017-11-11 17:17:44 1988739072 [Note] InnoDB: Waiting for purge to start
2017-11-11 17:17:44 1988739072 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 563465178
2017-11-11 17:17:44 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-11-11 17:17:44 1988739072 [Note] Plugin 'FEEDBACK' is disabled.
2017-11-11 17:17:44 1988739072 [Note] Server socket created on IP: '127.0.0.1'.
2017-11-11 17:17:44 1988739072 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

</details>

Ricardosgeral, 2017-11-11:
I'm facing the same problem after 0.34.6

Can't access cloud from anywhere.
in Firefox the page is blank.
any solution?

YarmoM, 2017-11-11:
Same here after update to 0.34.6.

When running `sudo ncp-update`, it says:

```
Downloading updates
Performing updates
vm.overcommit_memory = 1
sed: can't read /etc/redis/redis.conf: No such file or directory
grep: /etc/redis/redis.conf: No such file or directory
./update.sh: line 282: /etc/redis/redis.conf: No such file or directory
NextCloudPi updated to version v0.34.6
```

sandyCH, 2017-11-11:
same problem here. Update 0.34.6 caused failure to login from remote site.  Managed a restore to previous version but that too fails to login from remote site.

vlmd, 2017-11-11:
Same here after the update.
SSH seems to work though, but the Web GUI fails to load. I can set various options using the TUI (nextcloudpi-config).
However, in browser, it just shows a blank page.

Rufio87, 2017-11-11:
same problem after update. get an http error 500

Shadowstreik, 2017-11-11:
Access to NCP Web is ok. Looking at the version notes at https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md shows...

v0.34.6 (2017-11-11) redis: fix socket permissions

An SSL test shows a 500 error. Though, the test comes out ok.

Since the version is about SSL, the issues tell me it's a problem with the update, or SSL, or who knows?

I'm waiting for the next version release to see if a fix is deployed... and works.

nachoparker, 2017-11-12:
somehow it went fine in my testing. looking into this now

nachoparker, 2017-11-12:
Somehow this slipped through my testing. I think I tested the update on the wrong image >_<

It should be fixed now. Please update to v0.34.7 and confirm

Sorry for that

YarmoM, 2017-11-12:
Still not fixed, exact same issue remains. When updating, it says:

```
Downloading updates
Performing updates
vm.overcommit_memory = 1
sed: can't read /etc/redis/redis.conf: No such file or directory
grep: /etc/redis/redis.conf: No such file or directory
grep: /etc/redis/redis.conf: No such file or directory
./update.sh: line 289: /etc/redis/redis.conf: No such file or directory
NextCloudPi updated to version v0.34.7
```

None of the clients want to connect, web server returns `HTTP ERROR 503`.

nachoparker, 2017-11-12:
@YarmoM you don't seem to have redis installed? your issue is different from OP

what's the output of 

```
grep APCu /var/www/nextcloud/config/config.php
```

and

```
systemctl status redis
```

YarmoM, 2017-11-12:
Ow, my bad, I thought it was the same issue, since this error came up after the same update... I have never needed redis before 0.34.6, but if installing it solves the issue, I'll install it right away.

`grep APCu /var/www/nextcloud/config/config.php` returns nothing.
`systemctl status redis` returns `Unit redis.service could not be found.`

Sorry for the hijacking, my mistake...

nachoparker, 2017-11-12:
no problem!

weird, maybe the installation of redis failed for you. Yesterday I couldn't build the new image at all because the raspbian mirrors were down all day.

Let's make sure,

```
sudo -u www-data /var/www/nextcloud/occ config:list system
``

cfsancas, 2017-11-12:
I got another blank page with the version v0.31.25 when I tried to access the admin page. then I disabled  the nc-webui and I had a normal login page again. Maybe it is related with these interface.

nachoparker, 2017-11-12:
@cfsancas did you get problems updating to v0.34.6 too?

YarmoM, 2017-11-12:
Ok, I ran `sudo -u www-data /var/www/nextcloud/occ config:list system`, it gave me:

```
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudpi.local",
            "1": "192.168.2.2",
        },
        "datadirectory": "\/media\/xxx\/ncdata",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "12.0.3.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "mail_smtpmode": "php",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "admin",
        "mail_domain": "ownyourbits.com",
        "overwriteprotocol": "https",
        "maintenance": false,
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "\/var\/run\/redis\/redis.sock",
            "port": 0,
            "timeout": 0
        }
    }
}
```

Indeed, I'm struggling with installing redis right now, what's the preferred method? From source or apt?

YarmoM, 2017-11-12:
Redis is now working and active. However, the web client still gives a 503 error... Any logs I should check?

nachoparker, 2017-11-12:
> Indeed, I'm struggling with installing redis right now, what's the preferred method? From source or apt?

Paste this code ( just @YarmoM ), the rest of the people are in a different situation

```
sudo su
REDIS_CONF=/etc/redis/redis.conf
sysctl vm.overcommit_memory=1
apt-get update
apt-get install -y --no-install-recommends redis-server php7.0-redis

REDIS_MEM=3gb
sed -i "s|# unixsocket .*|unixsocket /var/run/redis/redis.sock|" $REDIS_CONF
sed -i "s|# unixsocketperm .*|unixsocketperm 770|"               $REDIS_CONF
sed -i "s|port.*|port 0|"                                        $REDIS_CONF
echo "maxmemory ${REDIS_MEM}" >> $REDIS_CONF
echo 'vm.overcommit_memory = 1' >> /etc/sysctl.conf
sudo usermod -a -G redis www-data
systemctl restart redis-server
systemctl enable redis-server

systemctl stop php7.0-fpm
systemctl stop mysqld
sleep 0.5
systemctl start php7.0-fpm
systemctl start mysqld

```

YarmoM, 2017-11-12:
Holy cow, that fixed it! Everything is up! Thanks so much for the support!

nachoparker, 2017-11-12:
Great!

nachoparker, 2017-11-12:
In the future, I will have to include the possibility of raspbian mirrordirector being down in my code >_<

Dolordo, 2017-11-12:
v0.34.8 fixes some issues. I'm now on a non-approved domain...
Error message on what is usually the nextcloud login asking to configure "trusted_domains" in config/config.php
I'm not sure that restoring the backup will slove the issue, may you comment @nachoparker ?

nachoparker, 2017-11-12:
@Dolordo reboot, or run

```
sudo systemctl start nextcloud-domain
```

nachoparker, 2017-11-12:
the main issue seems to be fixed

closed 2017-11-12 by nachoparker

Dolordo, 2017-11-12:
Did reboot after update as usual, and the issue was still here. Then finally I restore my backup, and everything went back. Thanks for your great support !

cfsancas, 2017-11-12:
@nachoparker yes I had problems with upgrade from v0.31.25 to v0.34.6. I got a blank page so I did a fresh install and didn't upgrade to v0.34.6 and when a opened the admin web page I get other blank page so I thought, maybe is due to  nc-webui so I disabled it and I got a normal login page. I did several restart between these steeps. Well this issue it's close so if I steel have problems I will open another one. thanks you.

nachoparker, 2017-11-12:
@cfsancas most likely your issue was that the upgrade to version had a bug (that I didn't notice because I tested with the wrong image), and/or the problems with the raspbian mirrors.

In any case, if this ever happens again (hope not), it's better to report it here and wait for the fix to come through ncp-update

> if I steel have problems I will open another one

yes, do

Ricardosgeral, 2017-11-12:
It's working now. Thanks