nextcloud/nextcloudpi #288: Blank page when upgrading to version v0.34.6

cfsancas, 2017-11-11:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
v0.34.6

### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_10-28-17.img

### Expected behavior
I just installed NextCloud pi so I expected a normal nexcloud login webpage

### Actual behaviour
I only get a blank webpage

### Steps to reproduce, from a freshly installed image
I copied the image on an sd card and when I run the nextcloudpi-config for the first time I get a message asking me if I want to upgrade to the version v0.34.6 and when I upgrade I get a blank page. Previously I had a normal login website. 
I've done this a couple of times and got the same results.

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
{"reqId":"WgdR9H8AAQEAAAM3310AAAAN","level":2,"time":"2017-11-11T19:39:32+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdSJ38AAQEAAANuKE0AAAAD","level":2,"time":"2017-11-11T19:40:23+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdSlH8AAQEAAANvnZYAAABA","level":2,"time":"2017-11-11T19:42:12+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdStH8AAQEAAANvnZcAAABB","level":2,"time":"2017-11-11T19:42:44+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdS1H8AAQEAAANvnc8AAABE","level":2,"time":"2017-11-11T19:43:16+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdS9H8AAQEAAANuKFMAAAAE","level":2,"time":"2017-11-11T19:43:48+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdTVH8AAQEAAANGUJsAAABA","level":2,"time":"2017-11-11T19:45:25+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdTdH8AAQEAAANGUJwAAABB","level":2,"time":"2017-11-11T19:45:56+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdTlH8AAQEAAANGUJ0AAABC","level":2,"time":"2017-11-11T19:46:28+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdTtH8AAQEAAANGUJ4AAABD","level":2,"time":"2017-11-11T19:47:00+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdT1X8AAQEAAANGULwAAABF","level":2,"time":"2017-11-11T19:47:33+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdT9H8AAQEAAANGUL0AAABG","level":2,"time":"2017-11-11T19:48:04+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdUFH8AAQEAAANGUL4AAABH","level":2,"time":"2017-11-11T19:48:36+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdUNH8AAQEAAANGUNsAAABJ","level":2,"time":"2017-11-11T19:49:08+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdUVH8AAQEAAANGUNwAAABK","level":2,"time":"2017-11-11T19:49:40+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdUdH8AAQEAAANGUN0AAABL","level":2,"time":"2017-11-11T19:50:12+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdUlH8AAQEAAANGUN4AAABM","level":2,"time":"2017-11-11T19:50:44+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdUtH8AAQEAAANFr9cAAAAN","level":2,"time":"2017-11-11T19:51:16+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdU9H8AAQEAAANIXmQAAAAA","level":2,"time":"2017-11-11T19:52:20+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdVFH8AAQEAAANIXmUAAAAB","level":2,"time":"2017-11-11T19:52:52+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdVNH8AAQEAAANIXmYAAAAC","level":2,"time":"2017-11-11T19:53:24+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as host.","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.3.2 (Nextcloud)","version":"12.0.3.3"}
{"reqId":"WgdVVX8AAQEAAANJxUwAAABA","level":2,"time":"2017-11-11T19:53:57+00:00","remoteAddr":"83.47.177.131","user":"--","app":"core","method":"GET","url":"\/status.php","message":"Trusted domain error. \"83.47.177.131\" tried to access using \"cscs.sytes.net\" as [... cut ...]
2017-11-11 21:19:47 1989087232 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 2911763
2017-11-11 21:19:47 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-11-11 21:19:47 1989087232 [Note] Plugin 'FEEDBACK' is disabled.
2017-11-11 21:19:47 1989087232 [Note] Server socket created on IP: '127.0.0.1'.
2017-11-11 21:19:47 1989087232 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0
```
</details>

kjmoon, 2017-11-12:
Same. 

It might be a Redis problem? I get this when I run ```sudo nextcloudpi-config```

```
pi@nextcloudpi:~ $ sudo nextcloudpi-config 
PHP Warning:  Redis::connect(): connect() failed: No such file or directory in /var/www/nextcloud/lib/private/RedisFactory.php on line 82
PHP Fatal error:  Uncaught RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(52): Redis->get('6d4f5e6ca7a8098...')
#1 /var/www/nextcloud/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/Log/File.php(101): spl_autoload_call('OC_User')
#4 [internal function]: OC\Log\File::write('PHP', 'Redis::connect(...', 3)
#5 /var/www/nextcloud/lib/private/Log.php(307): call_user_func(Array, 'PHP', 'Redis::connect(...', 3)
#6 /var/www/nextcloud/lib/private/Log.php(165): OC\Log->log(3, 'Redis::connect(...', Array)
#7 /var/www/nextcloud/lib/private/Log/ErrorHandler.php(68): OC\Log->critical('Redis::connect(...', Array)
#8 [internal function]: OC\Log\ErrorHandler::onShutdown()
#9 {main}
  thrown in /var/www/nextcloud/lib/private/Memcache/Redis.php on line 52
```

nachoparker, 2017-11-12:
Duplicate of #287

nachoparker, 2017-11-12:
please, see fix in #287 and report

closed 2017-11-12 by nachoparker