nextcloud/nextcloudpi #289: File not found when updating.

Iolaum, 2017-11-13:
### PROBLEMS

When updating nextcloudpi through ssh, file `/etc/apache2/sites-enabled/ncp.conf` was missing. Not sure if this is actually a problem (feel free to close if it's not), just putting it here in case it is.

### What version of NextCloudPi are you using? ( eg: v0.32.3 to v0.34.9 )

### What is the base image that you installed on the SD card? NextCloudPi_10-03-17

### Expected behavior

Normal update.

### Actual behaviour

Update proceeded with the following logs.

```
NextCloudPi v0.32.3 is outdated
update to v0.34.6 through 'nextcloudpi-config' or type 'sudo ncp-update'
pi@nextcloudpi:~ $ sudo nextcloud-config

Launching nc-update
Downloading updates
Performing updates
grep: /etc/apache2/sites-enabled/ncp.conf: No such file or directory
sed: can't read /etc/apache2/sites-enabled/ncp.conf: No such file or directory
vm.overcommit_memory = 1
installing redis...
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Get:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Fetched 15.0 kB in 1s (9,203 B/s)   
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
The following additional packages will be installed:
  php-igbinary redis-tools
Suggested packages:
  ruby-redis
The following NEW packages will be installed:
  php-igbinary php-redis redis-server redis-tools
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 840 kB of archives.
After this operation, 2,435 kB of additional disk space will be used.
Get:1 http://raspbian.mirror.uk.sargasso.net/raspbian stretch/main armhf php-igbinary armhf 2.0.1-1 [21.0 kB]
Get:2 http://raspbian.mirror.uk.sargasso.net/raspbian stretch/main armhf php-redis armhf 3.1.1-1 [97.9 kB]
Get:3 http://raspbian.mirror.uk.sargasso.net/raspbian stretch/main armhf redis-tools armhf 3:3.2.6-1 [381 kB]
Get:4 http://raspbian.mirror.uk.sargasso.net/raspbian stretch/main armhf redis-server armhf 3:3.2.6-1 [341 kB]
Fetched 840 kB in 1s (757 kB/s)       
Selecting previously unselected package php-igbinary.
(Reading database ... 41042 files and directories currently installed.)
Preparing to unpack .../php-igbinary_2.0.1-1_armhf.deb ...
Unpacking php-igbinary (2.0.1-1) ...
Selecting previously unselected package php-redis.
Preparing to unpack .../php-redis_3.1.1-1_armhf.deb ...
Unpacking php-redis (3.1.1-1) ...
Selecting previously unselected package redis-tools.
Preparing to unpack .../redis-tools_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-tools (3:3.2.6-1) ...
Selecting previously unselected package redis-server.
Preparing to unpack .../redis-server_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-server (3:3.2.6-1) ...
Setting up redis-tools (3:3.2.6-1) ...
Setting up php-igbinary (2.0.1-1) ...
Setting up php-redis (3.1.1-1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up redis-server (3:3.2.6-1) ...
Created symlink /etc/systemd/system/redis.service → /lib/systemd/system/redis-server.service.
Created symlink /etc/systemd/system/multi-user.target.wants/redis-server.service → /lib/systemd/system/redis-server.service.
Processing triggers for systemd (232-25+deb9u1) ...
Synchronizing state of redis-server.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable redis-server
Created symlink /etc/systemd/system/multi-user.target.wants/ramlogs.service → /usr/lib/systemd/system/ramlogs.service.
NextCloudPi updated to version v0.34.9
Done. Press any key...
l
pi@nextcloudpi:~ $ sudo apt update
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
pi@nextcloudpi:~ $ cd /etc/apache2/sites-enabled/
pi@nextcloudpi:/etc/apache2/sites-enabled $ ll
total 8
drwxr-xr-x 2 root root 4096 Oct 15 20:14 ./
drwxr-xr-x 8 root root 4096 Oct  2 16:45 ../
lrwxrwxrwx 1 root root   35 Oct  2 16:46 000-default.conf -> ../sites-available/000-default.conf
lrwxrwxrwx 1 root root   33 Oct  2 17:23 nextcloud.conf -> ../sites-available/nextcloud.conf
```

### Steps to reproduce, from a freshly installed image

N/A

I assume trying to update from the image v0.32.3 would reproduce the warnings:

```
grep: /etc/apache2/sites-enabled/ncp.conf: No such file or directory
sed: can't read /etc/apache2/sites-enabled/ncp.conf: No such file or directory
```

but as it is my "production" machine I haven't rolled back to test it.

nachoparker, 2017-11-13:
do you come from a quite old image? when ncp-web came out it was needed to manually install it. Seems like you didn't do that

nachoparker, 2017-11-13:
see here

https://ownyourbits.com/2017/07/24/nextcloupi-gets-a-web-interface/

closed 2017-11-13 by nachoparker

Iolaum, 2017-11-16:
I updated to a newer version and got this error again:

```
NextCloudPi v0.34.9 is outdated
update to v0.34.10 through 'nextcloudpi-config' or type 'sudo ncp-update'
pi@nextcloudpi:~ $ sudo ncp-update 
Downloading updates
Performing updates
grep: /etc/apache2/sites-enabled/ncp.conf: No such file or directory
sed: can't read /etc/apache2/sites-enabled/ncp.conf: No such file or directory
vm.overcommit_memory = 1
NextCloudPi updated to version v0.34.10
```

I believe this error happens because I have disabled the web admin interface. I 've done so because I administer ncp through ssh. I don't think it's because of an outdated image issue as my initial image is `NextCloudPi_10-03-17` which came 3 months after the post you mentioned that introduces ncp-web.

nachoparker, 2017-11-16:
It doesn't make sense... if you used the october image you should have that file. In any case, I will supress the warning for people with very old images that never installed it manually

Iolaum, 2017-11-16:
I looked at the repository and noticed that it's only symlinks there. But at the folder those symlinks point out there are more files. See below:

```
$ cd /etc/apache2/sites-enabled/
$ ls -laF
total 8
drwxr-xr-x 2 root root 4096 Oct 15 20:14 ./
drwxr-xr-x 8 root root 4096 Oct  2 16:45 ../
lrwxrwxrwx 1 root root   35 Oct  2 16:46 000-default.conf -> ../sites-available/000-default.conf
lrwxrwxrwx 1 root root   33 Oct  2 17:23 nextcloud.conf -> ../sites-available/nextcloud.conf
$ cd /etc/apache2/sites-available/
$ ls -laF
total 28
drwxr-xr-x 2 root root 4096 Oct  3 20:38 ./
drwxr-xr-x 8 root root 4096 Oct  2 16:45 ../
-rw-r--r-- 1 root root  224 Oct  3 20:38 000-default.conf
-rw-r--r-- 1 root root 6338 Sep 19 19:56 default-ssl.conf
-rw-r--r-- 1 root root  927 Nov 12 19:52 ncp.conf
-rw-r--r-- 1 root root  577 Oct  2 17:22 nextcloud.conf

```

So `ncp.conf` appears on the second directory. Maybe the symlink is disabled because I 've disabled webui and site is not enabled in apache? I tried to look at the code to verify this but couldn't. I did find however that the [install script](https://github.com/nextcloud/nextcloudpi/blob/868924c61fd3d08e63719cfa2e7d2bdf2636696c/nextcloudpi.sh#L42) creates the file at `/etc/apache2/sites-available/ncp.conf`. I am assuming that [those lines](https://github.com/nextcloud/nextcloudpi/blob/868924c61fd3d08e63719cfa2e7d2bdf2636696c/etc/nextcloudpi-config.d/nc-webui.sh#L22-L26) call commands that enable/disable those symlinks but I don't know enough about apache2 to be sure.

As a suggested fix, maybe edit the files at `/etc/apache2/sites-available/` instead of `/etc/apache2/sites-enabled/` when updating? I can try and create a PR if you think that this is a correct approach.

nachoparker, 2017-11-17:
It is not enabled, that is why it is not present in the sites-enabled folder. If you enable it through ncp-web or with `sudo a2ensite ncp` it will show up there.

The ones at sites-enabled are just symlinks to sites-available, so they are really the same file.

The issue is that this user comes from a very old image and never installed ncp-web as announced on ownyourbits for the upgrade.

Many thanks for the PR offer and looking into the code, we need a lot of that!

Cheers,