nextcloud/nextcloudpi #290: multi languages for ncp?

JimmyKater, 2017-11-13:
hey nacho & crew ;-)

ncp is still growing in features and reach the state of "devine", i guess ;)

and so it's difficult to improve it or coming up with random thoughts about improvement...
what about this one (and as usual i dunno how difficult it might be to implement it)...
- getting ncp translated into different languages using transifex (like nextcloud server and several apps do)?

i know of several users who don't feel well doing it all in english and so running and setting ncp up in their native tongues would be of an enormous help for them.

this is clearly an enhancement. but i only can mark this one as an issue

cheers
jimmy # #

nachoparker, 2017-11-13:
thanks for he suggestion

@Espina2 also keeps saying that this is important, if this keeps growing it will be something we'll have to do. 

I don't know how much work that would be either, but it seems like a lot :P

JimmyKater, 2017-11-13:
as i still don't really know how this works i could only guess...
you need to have all "texts" as variables to be stored in a file.
but i'm sure the guys of nc will be helping you out in case of questions... and i'll be helping you out in case of translations to german ;)

nachoparker, 2017-11-13:
> i'll be helping you out in case of translations to german ;)

thanks! that would be great

nachoparker, 2017-11-13:
we have spanish, portuguese, greek, english and german so far ;)

Daniel-Krebs, 2017-11-13:
If you need a german translation you can count on me too ;)

nachoparker, 2018-02-09:
finally some work is being contributed in this direction #372 

thanks to @theCalcaholic

chiyi4488, 2018-03-05:
or chinese ,haha

nachoparker, 2018-03-05:
> or chinese ,if you need.

great! the code for multilanguage is already out. We need to create some guide or mechanism to help people contribute with translations.

At this moment we have only German

nachoparker, 2018-03-18:
The code is out.

Thereis a small [guide](https://github.com/nextcloud/nextcloudpi/wiki/Add-a-new-language-to-ncp-web) in the wiki.

theCalcaholic, 2018-03-26:
Btw, I'm not sure if the character encoding is set up correctly for languages with non-latin characters yet... If any characters are displayed wrong for you, please let us know.

nachoparker, 2018-04-06:
This has been implemented for a while. I was thinking to show the language dropdown on the top bar, with a link/message to attract translators.

That will be more visible than this github issue

closed 2018-04-06 by nachoparker

cervorange, 2019-05-06:
Hello,

I'm interesting to help and try to translate in french but i don't know how to do this.
i look [here](https://docs.nextcloudpi.com/en/add-a-new-language-to-ncp-web/) but when i go to "development environment" i don't have access to update the wiki...

ovpc, 2019-05-06:
Hi @cervorange, thank you. Just create an account at docs.nextcloudpi.com and i will grant you translators priviliges.

ovpc, 2019-05-06:
@cervorange just realized I add not added French language to docs yet, just did, ;-)