nextcloud/nextcloudpi #291: Empty Nextcloud browser page after upgrade to NextcloudPi v0.34.9

datenbrei, 2017-11-13:
After updating to v0.34.9 the NextcloudPi webinterface on port 4443 is running, but not Nextcloud itself. The messagebox reported errors about redis directories not being existent. Also redis-server had not been installed. I did this by hand afterwards, but as it seems, it does not help.

I also was asked to run sudo dpkg --configure -a by hand. As it seems, there had been an issue with a changed file of samba and I was asked, if I'd like to keep the old one or install a fresh one. I kept the old one.

Also I did a reinstall (nc-update), but no luck.

I use RAM-Logs, so the messages below may be not complete ...

### What version of NextCloudPi are you using?
v0.34.9
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )
NextCloudPi_10-03-17
### Expected behavior
Normal operation
### Actual behaviour
No Nextcloud Webinterface
### Steps to reproduce, from a freshly installed image

### Include logs
<details>
<summary>Nextcloud logs</summary>

pi@nextcloudpi:/var $ sudo tail -n 100 /var/www/nextcloud/data/nextcloud.log
{"reqId":"Wd8Mmn8AAQEAAANhKJoAAEEY","level":2,"time":"2017-10-12T06:32:58+00:00","remoteAddr":"192.168.178.73","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"192.168.178.73\" tried to access using \"nextcloudpi\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:56.0) Gecko\/20100101 Firefox\/56.0","version":"12.0.3.3"}
{"reqId":"Wd8MnH8AAQEAAANhKKIAAEED","level":2,"time":"2017-10-12T06:33:00+00:00","remoteAddr":"192.168.178.73","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=5cfdca5cec3556faf4218ec6aeadf606-0","message":"Trusted domain error. \"192.168.178.73\" tried to access using \"nextcloudpi\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:56.0) Gecko\/20100101 Firefox\/56.0","version":"12.0.3.3"}
{"reqId":"Wd8MwX8AAQEAAANgtkAAAAAB","level":2,"time":"2017-10-12T06:33:37+00:00","remoteAddr":"192.168.178.73","user":"--","app":"core","method":"GET","url":"\/index.php\/settings\/admin?trustDomain=nextcloudpi","message":"Trusted domain error. \"192.168.178.73\" tried to access using \"nextcloudpi\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:56.0) Gecko\/20100101 Firefox\/56.0","version":"12.0.3.3"}
{"reqId":"Wd8MwX8AAQEAAANgtkMAAAAE","level":2,"time":"2017-10-12T06:33:38+00:00","remoteAddr":"192.168.178.73","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=5cfdca5cec3556faf4218ec6aeadf606-0","message":"Trusted domain error. \"192.168.178.73\" tried to access using \"nextcloudpi\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:56.0) Gecko\/20100101 Firefox\/56.0","version":"12.0.3.3"}
{"reqId":"Wd8OQ38AAQEAAANaJWQAAAAY","level":2,"time":"2017-10-12T06:40:03+00:00","remoteAddr":"192.168.178.73","user":"--","app":"core","method":"GET","url":"\/index.php\/settings\/admin?trustDomain=nextcloudpi","message":"Trusted domain error. \"192.168.178.73\" tried to access using \"nextcloudpi\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:56.0) Gecko\/20100101 Firefox\/56.0","version":"12.0.3.3"}
{"reqId":"Wd8OQ38AAQEAAANaJWUAAAAA","level":2,"time":"2017-10-12T06:40:03+00:00","remoteAddr":"192.168.178.73","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=5cfdca5cec3556faf4218ec6aeadf606-0","message":"Trusted domain error. \"192.168.178.73\" tried to access using \"nextcloudpi\" as host.","userAgent":"Mozilla\/5.0 (X11; Ubuntu; Linux x86_64; rv:56.0) Gecko\/20100101 Firefox\/56.0","version":"12.0.3.3"}

</details>
<details>
<summary>Apache logs</summary>
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-13 13:14:40 CET; 44min ago
 Main PID: 811 (apache2)
   CGroup: /system.slice/apache2.service
           ├─811 /usr/sbin/apache2 -k start
           ├─879 /usr/sbin/apache2 -k start
           └─880 /usr/sbin/apache2 -k start

Nov 13 13:14:38 nextcloudpi systemd[1]: Starting The Apache HTTP Server...
Nov 13 13:14:40 nextcloudpi systemd[1]: Started The Apache HTTP Server.

pi@nextcloudpi:/var/log/apache2 $ tail -n 100 /var/log/apache2/*.log
==> /var/log/apache2/error.log <==
[Mon Nov 13 13:14:39.641406 2017] [ssl:warn] [pid 598:tid 1992663040] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Nov 13 13:14:40.042413 2017] [ssl:warn] [pid 811:tid 1992663040] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Mon Nov 13 13:14:41.005155 2017] [mpm_event:notice] [pid 811:tid 1992663040] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l configured -- resuming normal operations
[Mon Nov 13 13:14:41.005318 2017] [core:notice] [pid 811:tid 1992663040] AH00094: Command line: '/usr/sbin/apache2'
[Mon Nov 13 13:25:07.642645 2017] [proxy_fcgi:error] [pid 879:tid 1960625200] [client 192.168.178.46:51510] AH01071: Got error 'PHP message: PHP Notice:  Undefined variable: params in /var/www/ncp-web/ncp-launcher.php on line 87\nPHP message: PHP Warning:  Invalid argument supplied for foreach() in /var/www/ncp-web/ncp-launcher.php on line 87\n'

==> /var/log/apache2/other_vhosts_access.log <==
localhost:4443 martin-notebook.fritz.box - - [13/Nov/2017:13:24:53 +0100] "POST /ncp-launcher.php HTTP/2.0" 200 102 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:56.0) Gecko/20100101 Firefox/56.0"
localhost:4443 martin-notebook.fritz.box - - [13/Nov/2017:13:24:55 +0100] "POST /ncp-launcher.php HTTP/2.0" 200 115 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:56.0) Gecko/20100101 Firefox/56.0"

</details>
<details>
<summary>mariaDB logs</summary>

pi@nextcloudpi:/var/log/apache2 $ systemctl status mysqld
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-13 13:14:53 CET; 46min ago
 Main PID: 1006 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─1006 /usr/sbin/mysqld

Nov 13 13:14:50 nextcloudpi mysqld[1006]: 2017-11-13 13:14:50 1988866048 [Note] InnoDB: Highest supported file format
Nov 13 13:14:51 nextcloudpi mysqld[1006]: 2017-11-13 13:14:51 1988866048 [Note] InnoDB: 128 rollback segment(s) are ac
Nov 13 13:14:51 nextcloudpi mysqld[1006]: 2017-11-13 13:14:51 1988866048 [Note] InnoDB: Waiting for purge to start
Nov 13 13:14:51 nextcloudpi mysqld[1006]: 2017-11-13 13:14:51 1988866048 [Note] InnoDB:  Percona XtraDB (http://www.pe
Nov 13 13:14:51 nextcloudpi mysqld[1006]: 2017-11-13 13:14:51 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet
Nov 13 13:14:52 nextcloudpi mysqld[1006]: 2017-11-13 13:14:52 1988866048 [Note] Plugin 'FEEDBACK' is disabled.
Nov 13 13:14:52 nextcloudpi mysqld[1006]: 2017-11-13 13:14:52 1988866048 [Note] Server socket created on IP: '127.0.0.
Nov 13 13:14:52 nextcloudpi mysqld[1006]: 2017-11-13 13:14:52 1988866048 [Note] /usr/sbin/mysqld: ready for connection
Nov 13 13:14:52 nextcloudpi mysqld[1006]: Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'
Nov 13 13:14:53 nextcloudpi systemd[1]: Started MariaDB database server.
pi@nextcloudpi:/var/log/apache2 $ tail -n 100 /var/log/mysql/*.log
tail: '/var/log/mysql/*.log' kann nicht zum Lesen geöffnet werden: Datei oder Verzeichnis nicht gefunden
(german. Means no file or directory)
</details>

nachoparker, 2017-11-13:
Duplicate of #287

nachoparker, 2017-11-13:
There were problems during the weekend with the Raspbian mirrors that disrupted the upgrade.

Reinstall using the code in this comment

https://github.com/nextcloud/nextcloudpi/issues/287#issuecomment-343726717

datenbrei, 2017-11-13:
That fixed the problem, many thanks! :-)

closed 2017-11-13 by datenbrei

nachoparker, 2017-11-13:
great