nextcloud/nextcloudpi #292: Problems with the new ncp user

Daniel-Krebs, 2017-11-13:
Using NextCloudPi_11-12-17

After we changed the default user from pi to ncp there are some problems using ssh:
1. ncp has no home directory: `Could not chdir to home directory /home/ncp: No such file or directory`

 2. ncp is not in the sudoers file: `ncp is not in the sudoers file.  This incident will be reported.`

2. The message `SSH is enabled and the default password for the 'pi' user has not been changed.
This is a security risk - please login as the 'pi' user and type 'passwd' to set a new password.` should be changed to `SSH is enabled and the default password for the 'ncp' user has not been changed.
This is a security risk - please login as the 'ncp' user and type 'passwd' to set a new password.`

nachoparker, 2017-11-14:
Hi,

In Raspbian, the user _pi_ still exists. The user _ncp_ is only used (internally) to authenticate to ncp-web, and it is not intended to be used in a login shell.

So everyone should keep using the user _pi_, except for loging in to the web interface

closed 2017-11-14 by nachoparker

Daniel-Krebs, 2017-11-14:
Ah ok, but "except for loging in to the web interface"? I need to use pi/raspberry in the webinterface also, ncp/ownyourbits doesn't work

nachoparker, 2017-11-14:
did you try the latest image? this is only from the latest onwards

Daniel-Krebs, 2017-11-14:
I used NextCloudPi_11-12-17.torrent

nachoparker, 2017-11-14:
mmm
 
that's weird. can you check in the file `/etc/apache2/sites-available/ncp.conf` wether it says 'Require user pi' or 'Require user ncp'?

Daniel-Krebs, 2017-11-14:
<RequireAny>

      Require env noauth

      Require user pi

   </RequireAny>

nachoparker, 2017-11-14:
man! thanks for noticing!

There was an old fix in the updates that was reverting it back to _pi_. Fixed now.

Everyone that updated it are using _pi_, but people that start from a fresh image are (and will be) on _ncp_

Daniel-Krebs, 2017-11-14:
Thanks for fixing ;)
So we will have a new image?

nachoparker, 2017-11-14:
no, it was a problem with `nc-update`. The image is fine, it is the update that changed the user name

Daniel-Krebs, 2017-11-14:
Got it ;)