nextcloud/nextcloudpi #293: Letsencrypt without Port 80 Setup

typxyr, 2017-11-14:
I have found a script that possibly can be ported to nextcloudpi for using letsencrypt and duckdns when port 80 is blocked by ISP. I am no coder, but I have looked through it and it looks plausible.

The script is found at https://github.com/lukas2511/dehydrated/blob/master/dehydrated

The author has a tutorial for the project/script found [here](https://www.splitbrain.org/blog/2017-08/10-homeassistant_duckdns_letsencrypt).

I understand this issue is closed but I didn't know if I should open an "issue" about this. Is there any way this is possible to integrate into the ncp config?

nachoparker, 2017-11-14:
thanks, that would be a big improvement for us, we could close HTTP port alltogether, and even use non conventional ports for HTTPS

somebody with time and energy to take on this?

ovpc, 2017-11-29:
Without dehydrated, but using certbot's dns option to verify, accomplished in 7 steps [https://github.com/nextcloud/nextcloudpi/wiki/How-to-get-certificate-with-Letsencrypt-using-DNS-to-verify-domain](url)
Comments and help to fine-tune and if possible automate very much appreciated.

nachoparker, 2017-11-29:
great!

let's see if we can start automating this process. If possible, it is better if we don't need to restart apache because it messes up when you fire from ncp-web. It's a minor issue but something to keep in mind.

thanks

nachoparker, 2017-12-04:
we now need someone to make these steps into a simple script, that would be eventually included in `letsencrypt.sh`

nachoparker, 2017-12-30:
we can add this script to the valuable resources list

https://github.com/Neilpang/acme.sh/blob/master/acme.sh

nachoparker, 2018-05-18:
acme.sh supports wildcard certificates and DNS challenge. Looking good!

nachoparker, 2018-06-18:
Also, seems like LE supports DNS challenge, worth investigating ofc

https://serverfault.com/questions/750902/how-to-use-lets-encrypt-dns-challenge-validation

crossr1, 2018-07-06:
When I return on Monday, I would like to try this project. Could you please assign it to me?

nachoparker, 2018-07-06:
that's great! will you try the letsencrypt DNS challenge? or acme.sh? or both?

crossr1, 2018-07-09:
@nachoparker - I will have to research and get back to you with the answer.

crossr1, 2018-07-18:
Dehydrated, like all of the other scripts for 'Letsencrypt’, has only two ways to perform the 'letsencrypt challenge’. This is determined by the ACME protocol standard.

1. http-01: uses HTTP only - if port 80 is blocked by an ISP, then there are two options:
    * redirect the HTTP request to HTTPS using the DNS records - because the ISP is blocking port 80 (HTTP request), it never gets to the web server. So, the redirect to HTTPS must be performed before trying to send the request thru the ISP. 
    * use DNS Challenge - option below
2. dns-01: uses DNS Challenge:
    * The DNS records for the web server (host) has to be updated with a ‘TXT’ record for the challenge. This requires the website admin to have the permissions to update the DNS records for web server, usually thru the DNS provider (ie: Namecheap, duckdns, freeDNS, no-ip, spDYN, etc)
Note: this does impact the DNS - ‘TXT’ is a normal DNS record for notes related to the host
    * Because the DNS record must be updated at the DNS Provider, it requires sufficient privileges to update the DNS records for the host

**Reference Notes**
ACME is a communication protocol standard set by http://ietf.org (Internet Engineering Task Force)
ACME’s standards determine the authenticity of the website thru two mechanisms:
- http (https is NOT allowed) - https://ietf-wg-acme.github.io/acme/draft-ietf-acme-acme.html#rfc.section.8.3
- dns - https://ietf-wg-acme.github.io/acme/draft-ietf-acme-acme.html#rfc.section.8.4
ACME Protocol Clients - https://letsencrypt.org/docs/client-options/

crossr1, 2018-07-18:
The preliminary research indicates that the best approach would be:

Lexicon (aka dns-lexicon)

with either:

dehydrated
acme.sh
other script???

nachoparker, 2018-07-18:
thanks Richard.

Well, lexicon seems an interesting find to modify those TXT records. There might even be some examples out there to do what we are trying to achieve.

If BASH based solutions work I am fine with that. LE is pretty slow on the pi. That being said, it is a well tested solution and I am also curious if we could make this work with LE. It seems like it supports DNS challenge

xelcho, 2018-09-10:
I too am interested in helping. 
Specifically on using the dns challenge (freedns). 

I am currently using freedns.afraid.org for my dns services. 

Please direct me to where I can assist.

nachoparker, 2018-09-11:
great @xelcho! that's great!

well, @crossr1 told me that he had made some progress but I still haven't made time for a conversation about it.

let's discuss it here, Richard so we can get more input/opinions/help!

cheers

nachoparker, 2018-12-30:
https://github.com/nextcloud/nextcloudpi/wiki/How-to-get-certificate-with-Letsencrypt-using-DNS-to-verify-domain

ovpc, 2019-01-31:
Is this [wiki page](https://github.com/nextcloud/nextcloudpi/wiki/How-to-get-certificate-with-Letsencrypt-using-DNS-to-verify-domain)  outdated with v1.5.1 and new letsencrypt?
It seems that letsencrypt-auto does not supports a **--preferred-challenges dns** option. ?
Which is the only way to get a certificate if NC is on an alternate port.