nextcloud/nextcloudpi #295: % used full

swr85260, 2017-11-14:
After new setu with fresh installation with the wizard, on a 180Go SSD I've got 91% full after only 400Mo
Enclosed 2 pictures with NCDATA location on the SSD 

[image]
[image]

sunjam, 2017-11-14:
SSH into the Pi and `df -h /path/to/ssd` to confirm it is actually full.  My up-to-date 12.0.3 instance currently displays 0% used in the Nextcloud interface, which is also wrong.  Fyi, my database and ncdata are stored on a USB drive, as setup through NCP.  Once I moved them the % became 0%.

nachoparker, 2017-11-15:
there was an issue with the available space, but I think it got fixed in 12.0.3

https://github.com/nextcloud/server/issues/6022

try what @sunjam suggests.

Maybe when you move it to the USB drive NC doesn't get it right? we would have to confirm this

nachoparker, 2017-11-15:
> Once I moved them the % became 0%.

I didn't read that right... I guess it is confirmed then? using nc-datadir breaks the available space indicator... that would have to be reported to nextcloud/server

swr85260, 2017-11-16:
**sujam : "df -h"**

pi@nextcloudpi:/media/USBdrive $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       2.7G  2.5G  152K 100% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   13M  452M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
tmpfs           464M     0  464M   0% /tmp
tmpfs           464M  184K  464M   1% /var/log
/dev/mmcblk0p1   42M   21M   21M  51% /boot
/dev/sda1       165G   61M  156G   1% /media/myCloudDrive
tmpfs            93M     0   93M   0% /run/user/1000

**Natchopaker : "moving to NCDATADIR"**
/media/USBdrive/ncdata is not empty

sunjam, 2017-11-16:
@swr85260 Looks like ` /dev/sda1 165G 61M 156G 1% /media/myCloudDrive` is your external storage device.
    
Easiest fix would be to change the mount point for `/dev/sda1` from `/media/myCloudDrive` to `/media/USBdrive/` so it will show up as NextcloudPi is currently expecting.

sunjam, 2017-11-16:
@nachoparker my NextcloudPi instance only sees data that I'm pushing to the device by dropping directly into the web interface via drag and drop -- example is the two screenshots, totalling 38kb, that I drag and dropped into the files app from a browser showing up in the % full.  However, external photos I'm constantly uploading to the default USB data location are displayed in the files app as taking 37.8gb, but are not displayed in % storage used.
[image]
[image]

nachoparker, 2017-11-17:
@swr85260 try moving it to `/media/myCloudDrive/myncdata

@sunjam probably `/media/USBdrive` and `/media/myCloudDrive` are the same folder. NCP creates symlinks for easier autoconfiguration.

```
ls -l /media
drwxrwxrwx 29 root root 4096 Nov 17 08:30 myCloudDrive
lrwxrwxrwx  1 root root   16 Nov 10 17:20 USBdrive -> /media/myCloudDrive
```

nachoparker, 2017-11-17:
@sunjam you are using the 'external storage' app right? this is unrelated to the way ncp does things, which is having the whole data folder in USBdrive

swr85260, 2017-11-17:
I understand the wizard mount the USB storage to the wrong place on "/media/USBdrive/ncdata" instead of "/media/myCloudDrive"

Having said that, I messed up the image by trying to move NCdata to a new place and I'm completetly stuck.

nachoparker, 2017-11-17:
`/media/USBdrive` and `/media/myCloudDrive` are the same place, one is a symlink to the other. See

```
ls -l /media
```

in ncp-web, try using `nc-datadir` to move it to `/media/myCloudDrive/myncdata`

still don't see anything to be fixed

swr85260, 2017-11-17:
I did a fresch new installation and moved Ncdata to "/media/myCloudDrive/myncdata"
[image]
And moved files 166Mb and already 25% of total space !
[image]

nachoparker, 2017-11-17:
Do you have automount selected?

output of 

```
ls -l /media
```

and

```
df -h
```
?

swr85260, 2017-11-18:
**Automount selected**
[image]

**pi@nextcloudpi:~ $ ls -l /media**
total 12
drwxr-xr-x 3 root root 4096 Nov 17 20:37 myCloudDrive
drwxr-xr-x 3 root root 4096 Nov 17 20:35 myCloudDrive1
drwxr-xr-x 3 root root 4096 Nov 17 20:36 USBdrive
lrwxrwxrwx 1 root root   20 Nov 17 20:50 USBdrive1 -> /media/myCloudDrive1

**pi@nextcloudpi:~ $ df -h**
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       2.7G  2.2G  304M  89% /
devtmpfs        460M     0  460M   0% /dev
tmpfs           464M     0  464M   0% /dev/shm
tmpfs           464M   13M  452M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           464M     0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1   42M   21M   21M  51% /boot
/dev/sda1       165G   61M  156G   1% /media/myCloudDrive1
tmpfs            93M     0   93M   0% /run/user/1000
pi@nextcloudpi:~ $

nachoparker, 2017-11-18:
mmm automount is doing something weird for you.

For a quick fix, use nc-datadir to move it to something like `/media/myCloudDrive1/ncdatadir1`

To understand what's going on, let's try

```
systemctl status automount-links
```

```
ls /media/USBdrive
```

```
ls /media/myCloudDrive
```

swr85260, 2017-11-18:
**pi@nextcloudpi:~ $ systemctl status nc-automount-links**
● nc-automount-links.service - Monitor /media for mountpoints and create USBdriv
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled;
   Active: active (running) since Sat 2017-11-18 08:13:11 UTC; 4h 15min ago
 Main PID: 373 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─373 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─398 inotifywait --monitor --event create --event delete --format %f
           ├─399 grep --line-buffered ISDIR
           └─400 /bin/bash /usr/local/etc/nc-automount-links-mon

Nov 18 08:13:11 nextcloudpi systemd[1]: Started Monitor /media for mountpoints a
Nov 18 08:13:12 nextcloudpi nc-automount-links-mon[373]: Setting up watches.
Nov 18 08:13:12 nextcloudpi nc-automount-links-mon[373]: Watches established.
Nov 18 08:13:15 nextcloudpi nc-automount-links-mon[373]: myCloudDrive1 DELETE,IS
Nov 18 08:13:15 nextcloudpi nc-automount-links-mon[373]: myCloudDrive1 CREATE,IS
lines 1-15/15 (END)

**pi@nextcloudpi:~ $ ls /media/USBdrive**
ncdata

**pi@nextcloudpi:~ $ ls /media/myCloudDrive**
myncdata

So...what's up doctor?

nachoparker, 2017-11-19:
I am glad that it worked.

It's hard to tell why those folders were created in the first place. In any case, now nc-datadir refuses to move to the SD card, so we shouldn't be seeing this again, and if something weird comes up we will be able to detect it better.

closed 2017-11-19 by nachoparker