nextcloud/nextcloudpi #296: NextCloudPi 12.0.3 Berryboot image download link broken

sunjam, 2017-11-14:
[`Or use this [link] to import and download directly from the internet` -- Click here to navigate to the github wiki page.](https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-an-external-drive-using-Berryboot.)

[fyi, no one has been seeding `NextCloudPi 12.0.3 Berryboot image ( 514 MB )` the last two days I've attempted to download it.](https://ownyourbits.com/2017/02/13/nextcloud-ready-raspberry-pi-image/)

nachoparker, 2017-11-15:
Me and @ovpc have been seeding it, you couldn't download it?

What do you mean broken? I just tested the link again and it works. 

It is a magnet link and firefox opens it with transmission in my computer.

sunjam, 2017-11-15:
The direct download link on the github page is missing for ncp Berryboot
due to missing formatting.  Search for the text I posted above.  Torrent
link is working, but I couldn’t locate any seeders for the Berry Boot
torrent.

nachoparker, 2017-11-15:
Sorry, do you mean in ownyourbits or the wiki? Anyhow, we are setting up a downloads folder in ownyourbits.com/download for future use.

In the wiki I don't see any link or torrent, only to ownyourbits :|

sunjam, 2017-11-16:
I wasn't clear enough.  Missing link is on the [github wiki page here.](https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-an-external-drive-using-Berryboot.)

`Or use this [link] to import and download directly from the internet into the Berryboot menu.`

Perhaps I just got confused, but I was looking for a link to plug into Berryboot.  I couldn't figure it out so I just ended up installing base Debian listed in Berryboot, then the curl script.

nachoparker, 2017-11-17:
I see it now. Let's ping @ovpc to correct it to what he originally intended

thanks

ovpc, 2017-11-17:
NCP berryboot Image now available from 
[[https://ownyourbits.com/downloads/NextCloudPi_11-12-17_for_berryboot.img]]
torrent from
[[https://ownyourbits.com/downloads/NextCloudPi_11-12-17_for_berryboot.img.torrent]]
Just added these two links to [wiki](https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-an-external-drive-using-Berryboot.) also. Thanks @sunjam for pointing out

closed 2017-11-17 by ovpc