nextcloud/nextcloudpi #298: config.php is empty

JVAQUEROM, 2017-11-15:
### What version of NextCloudPi are you using? ( eg: v0.17.2 )
 I am using  version v0.34.10 right now
### What is the base image that you installed on the SD card? ( eg: NextCloudPi_07-21-17 )

NextCloudPi_10-02-17

### Expected behavior
Function normally
### Actual behaviour
blank page
### Steps to reproduce, from a freshly installed image

It worked fine but apparently the installer didn't put the data in the USBdrive. The SD got full. Tried to move the data with `nextcloudpi-config`, but an error appeared, and the data got duplicated (in the SD and the USBdrive). In the middle of this I updated nextcloudpi. Then I realized that the config.php file is empty, and I get a blank page. To work in the SD, I moved the data to the USBdrive and created a symbolic link to USBdrive in /var/www/nextcloud/data. 

### Include logs
<details>
<summary>Nextcloud logs</summary>

```
Login as admin user into your Nextcloud and copy here the logs from
https://example.com/index.php/settings/admin/logging

If you don't have access to the web interface, open a terminal session and paste the last 100 lines of /var/www/nextcloud/data/nextcloud.log
{"reqId":"Wgt4cn8AAQEAAAPyb9cAAABL","level":2,"time":"2017-11-14T23:12:50+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt4kn8AAQEAAAPx0roAAAAF","level":2,"time":"2017-11-14T23:13:22+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt4sn8AAQEAAAPyb9kAAABN","level":2,"time":"2017-11-14T23:13:54+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt40n8AAQEAAAPyb9sAAABP","level":2,"time":"2017-11-14T23:14:26+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt48n8AAQEAAAPx0rwAAAAM","level":2,"time":"2017-11-14T23:14:58+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"uvOhvghoOQ9IO1dcYd4q","level":4,"time":"2017-11-14T23:15:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"Wgt5En8AAQEAAAPyb90AAABR","level":2,"time":"2017-11-14T23:15:30+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt5Mn8AAQEAAAPx0r4AAAAO","level":2,"time":"2017-11-14T23:16:02+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt5Un8AAQEAAAPyb98AAABT","level":2,"time":"2017-11-14T23:16:34+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt5cn8AAQEAAAPyb@EAAABV","level":2,"time":"2017-11-14T23:17:06+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt5kn8AAQEAAAPx0sAAAAAQ","level":2,"time":"2017-11-14T23:17:38+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt5sn8AAQEAAAPx0sIAAAAR","level":2,"time":"2017-11-14T23:18:10+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt50n8AAQEAAAPx0sQAAAAT","level":2,"time":"2017-11-14T23:18:42+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt58n8AAQEAAAPyb@MAAABX","level":2,"time":"2017-11-14T23:19:14+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt6En8AAQEAAAPyb@UAAABA","level":2,"time":"2017-11-14T23:19:46+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt6Mn8AAQEAAAPx0sYAAAAV","level":2,"time":"2017-11-14T23:20:18+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt6Un8AAQEAAAPyb@cAAABC","level":2,"time":"2017-11-14T23:20:50+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt6cn8AAQEAAAPyb@kAAABE","level":2,"time":"2017-11-14T23:21:22+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt6kn8AAQEAAAPyb@sAAABG","level":2,"time":"2017-11-14T23:21:54+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt6sn8AAQEAAAPx0sgAAAAX","level":2,"time":"2017-11-14T23:22:26+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
{"reqId":"Wgt60n8AAQEAAAPx0soAAAAD","level":2,"time":"2017-11-14T23:22:58+00:00","remoteAddr":"90.77.59.206","user":"--","app":"no app in context","method":"GET","url":"\/status.php","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"Mozilla\/5.0 (Linux) mirall\/2.2.4 (Nextcloud)","version":""}
[... cut ...]
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─1130 /usr/sbin/mysqld

Nov 14 22:43:51 nextcloudpi systemd[1]: Starting MariaDB database server...
Nov 14 22:44:05 nextcloudpi mysqld[1130]: 2017-11-14 22:44:05 1989136384 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 1130 ...
Nov 14 22:44:07 nextcloudpi systemd[1]: Started MariaDB database server.

Paste the output of `tail -n 100 /var/log/mysql/*.log`

Nothing here

```
</details>

nachoparker, 2017-11-15:
If the SD got full, you get corrupted data and empty files :^S

Do you have backups? How did you move the datadir? wizard? ncp-web? Do you only have data, or also calendars, contacts and so on?

JVAQUEROM, 2017-11-15:
too bad...
I do not have backups (if by backups you mean the data, I do, everything is on my laptop). First time, I moved with the wizard. Aparently this did not work because the SD card got full. When I realized the SD was full, I tried with the nextcloudpi-config. I have calendar and contacts too. 

I have set everything up in the weekend, so if you thing I should reinstall it would not be much work. I guess I could use the data that is in the RPI already to do a nc-scan and get a faster sync. 

I have had to reinstall several times, because I wanted to get a lot of data and quick and I have messed around with the data and database several times, but every time I reinstall everything works better and easier, and I get to learn some things too. So please keep up the good work.
Particularly, I found the upload of files in nextcloud faster in the last reinstall.

nachoparker, 2017-11-16:
It surprises me that this failed in the wizard for you, but I have been thinking for a while and I am going to have nc-datadir _refuse_ to move to the SD card, to avoid this situation. I wanted it to be flexible, but I guess that it could cause trouble.

I think that reinstalling is the easiest, and yes, you can use nc-scan for that ;)

Thanks for the feedback on the uploads, it was the biggest improvement of the last release, so it's good to know that people are noticing this ;)

nachoparker, 2017-11-16:
also, I recommend that you do periodic backups, even with no data for the calendars and so on

closed 2017-11-16 by nachoparker

JVAQUEROM, 2017-11-16:
Ok, thank you, I'll do as you suggest!!