nextcloud/nextcloudpi #301: NextcloudPi-config updates -- what is being updated?

sunjam, 2017-11-16:
Would it be possible to list what is actually being changed with each Nextcloudpi-config update..?  The dialog windows asks for update approval, but gives no description or recap of what is actually changing.
[image]

nachoparker, 2017-11-17:
in ncp-web you can click the version number, which will take you to the [changelog](https://github.com/nextcloud/nextcloudpi/blob/master/changelog.md)

in nextcloudpi-config... I guess we could add the URL as well...

nachoparker, 2017-11-17:
How about this

[image]

sunjam, 2017-11-17:
Do you think it would be possible to list the changes within ncp since the last upgrade?
[image]

nachoparker, 2017-11-17:
I guess it would be possible ;D

nachoparker, 2017-12-05:
implemented `v0.41.6`

[image]

closed 2017-12-05 by nachoparker