nextcloud/nextcloudpi #302: Blank page on Web GUI

eogut, 2017-11-17:
What version of NextCloudPi are you using?
0.34.15
What is the base image that you installed on the SD card?
untagged

I get the [same issue ](https://github.com/nextcloud/nextcloudpi/issues/287#issuecomment-343682867) as @YarmoM and not willing to screw things up, I didn't want to just go ahead and use [your script](https://github.com/nextcloud/nextcloudpi/issues/287#issuecomment-343726717).

Please tell me which logs you need or what steps I need to take. Thank you.

nachoparker, 2017-11-17:
when did you start seeing this? was it last saturday? please, use that script, it's safe.

Otherwise, fill in the info that you get in the template when you open a new issue on github.

eogut, 2017-11-18:
Hey nacho,

last Saturday sounds about right. Alright, I'll use the script and turn 
back to you if it didn't work. :) cheers!

nachoparker, 2017-11-18:
reopen in that case, good luck!

closed 2017-11-18 by nachoparker

eogut, 2017-11-18:
That didn't work unfortunately, I still get a blank page, here's what happened:
`sudo su
sed -i "s|# unixsocketperm .*|unixsocketperm 770|"               $REDIS_CONF
sed -i "s|port.*|port 0|"                                        $REDIS_CONF
echo "maxmemory ${REDIS_MEM}" >> $REDIS_CONF
echo 'vm.overcommit_memory = 1' >> /etc/sysctl.conf
sudo usermod -a -G redis www-data
systemctl restart redis-server
systemctl enable redis-server

systemctl stop php7.0-fpm
systemctl stop mysqld

systemctl start php7.0-fpm
systemctl start mysqld
root@Nextcloud:/home/pi# REDIS_CONF=/etc/redis/redis.conf
root@Nextcloud:/home/pi# sysctl vm.overcommit_memory=1
vm.overcommit_memory = 1
root@Nextcloud:/home/pi# apt-get update
Get:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Fetched 15.0 kB in 0s (15.5 kB/s)
Reading package lists... Done
root@Nextcloud:/home/pi# apt-get purge -y redis-server php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
Package 'redis-server' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.
root@Nextcloud:/home/pi# apt-get install -y --no-install-recommends redis-server php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
The following additional packages will be installed:
  php-igbinary redis-tools
Suggested packages:
  ruby-redis
The following NEW packages will be installed:
  php-igbinary php-redis redis-server redis-tools
0 upgraded, 4 newly installed, 0 to remove and 22 not upgraded.
Need to get 840 kB of archives.
After this operation, 2,435 kB of additional disk space will be used.
Get:1 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf php-igbinary armhf 2.0.1-1 [21.0 kB]
Get:2 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf php-redis armhf 3.1.1-1 [97.9 kB]
Get:3 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf redis-tools armhf 3:3.2.6-1 [381 kB]
Get:4 http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian stretch/main armhf redis-server armhf 3:3.2.6-1 [341 kB]
Fetched 840 kB in 3s (237 kB/s)
Selecting previously unselected package php-igbinary.
(Reading database ... 41028 files and directories currently installed.)
Preparing to unpack .../php-igbinary_2.0.1-1_armhf.deb ...
Unpacking php-igbinary (2.0.1-1) ...
Selecting previously unselected package php-redis.
Preparing to unpack .../php-redis_3.1.1-1_armhf.deb ...
Unpacking php-redis (3.1.1-1) ...
Selecting previously unselected package redis-tools.
Preparing to unpack .../redis-tools_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-tools (3:3.2.6-1) ...
Selecting previously unselected package redis-server.
Preparing to unpack .../redis-server_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-server (3:3.2.6-1) ...
Setting up redis-tools (3:3.2.6-1) ...
Setting up php-igbinary (2.0.1-1) ...
Setting up php-redis (3.1.1-1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up redis-server (3:3.2.6-1) ...
Created symlink /etc/systemd/system/redis.service → /lib/systemd/system/redis-server.service.
Created symlink /etc/systemd/system/multi-user.target.wants/redis-server.service → /lib/systemd/system/redis-server.service.
Processing triggers for systemd (232-25+deb9u1) ...
`

I can't remember which logs there were to post and creating a new issue just asks for `sudo ncp-report`, for which my Raspi responds with command not found. Which logs do you require? Cheers!

nachoparker, 2017-11-19:
Hi,

Ok, we can rule redis out of the ecuation then.

`ncp-report` is a new tool to help with support. First, update to the latest version with `sudo ncp-update`, and then `sudo ncp-report` will get you the logs.

Also, try a reboot first

eogut, 2017-11-19:
This is the log:

<details>
<summary>NextCloudPi diagnostics</summary>
```

PHP Warning:  Redis::connect(): connect() failed: No such file or directory in /var/www/nextcloud/lib/private/RedisFactory.php on line 82
PHP Fatal error:  Uncaught RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(52): Redis->get('b07840960848c54...')
#1 /var/www/nextcloud/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/Log/File.php(101): spl_autoload_call('OC_User')
#4 [internal function]: OC\Log\File::write('PHP', 'Redis::connect(...', 3)
#5 /var/www/nextcloud/lib/private/Log.php(301): call_user_func(Array, 'PHP', 'Redis::connect(...', 3)
#6 /var/www/nextcloud/lib/private/Log.php(159): OC\Log->log(3, 'Redis::connect(...', Array)
#7 /var/www/nextcloud/lib/private/Log/ErrorHandler.php(68): OC\Log->critical('Redis::connect(...', Array)
#8 [internal function]: OC\Log\ErrorHandler::onShutdown()
#9 {main}
  thrown in /var/www/nextcloud/lib/private/Memcache/Redis.php on line 52
NextCloudPi version  v0.37.0
NextCloudPi image    untagged
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          sda
datadir              /media/Festplatte/Nextcloud/Data
data in SD           no
data filesystem      ext2/ext3
data disk usage      61G/874G
rootfs usage         2.6G/15G
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.178.39
gateway              192.168.178.1
interface            enxb827eb09ee11
certificates         kushwraboowymgkl.myfritz.net
certs due            69 days
NAT loopback         no
uptime               13min

```
</details>

<details>
<summary>Nextcloud configuration</summary>
```

PHP Warning:  Redis::connect(): connect() failed: No such file or directory in /var/www/nextcloud/lib/private/RedisFactory.php on line 82
An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(52): Redis->get('b07840960848c54...')
#1 /var/www/nextcloud/lib/autoloader.php(146): OC\Memcache\Redis->get('OC_User')
#2 [internal function]: OC\Autoloader->load('OC_User')
#3 /var/www/nextcloud/lib/private/User/Session.php(211): spl_autoload_call('OC_User')
#4 /var/www/nextcloud/lib/private/legacy/app.php(319): OC\User\Session->getUser()
#5 /var/www/nextcloud/lib/private/legacy/app.php(110): OC_App::getEnabledApps()
#6 /var/www/nextcloud/lib/base.php(673): OC_App::loadApps(Array)
#7 /var/www/nextcloud/lib/base.php(1086): OC::init()
#8 /var/www/nextcloud/console.php(56): require_once('/var/www/nextcl...')
#9 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#10 {main}PHP Fatal error:  Uncaught RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:52
Stack trace:
[... 61 lines cut ...]
```

{"reqId":"WhDBDH8AAQEAAAsLUTMAAABD","level":4,"time":"2017-11-18T23:23:56+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
{"reqId":"WhDBIX8AAQEAAAsLUTQAAABE","level":4,"time":"2017-11-18T23:24:17+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
{"reqId":"WhDBQX8AAQEAAAsLUTUAAABH","level":4,"time":"2017-11-18T23:24:49+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
{"reqId":"WhDBYX8AAQEAAAsKvFoAAAAP","level":4,"time":"2017-11-18T23:25:21+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
{"reqId":"WhDBgX8AAQEAAAsKvFsAAAAQ","level":4,"time":"2017-11-18T23:25:53+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
{"reqId":"WhDBpX8AAQEAAAsKvFwAAAAW","level":4,"time":"2017-11-18T23:26:29+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}
{"reqId":"WhDBwX8AAQEAAAsLUTYAAABI","level":4,"time":"2017-11-18T23:26:57+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not availabl [... cut ...] /Memcache\/Factory.php#96","userAgent":"Mozilla\/5.0 (Windows NT 6.1; Win64; x64; rv:57.0) Gecko\/20100101 Firefox\/57.0","version":"12.0.2.0"}
{"reqId":"WhDDIX8AAQEAAAsKvGAAAAAT","level":4,"time":"2017-11-18T23:32:49+00:00","remoteAddr":"89.249.64.163","user":"--","app":"remote","method":"GET","url":"\/status.php","message":"Memcache \\OC\\Memcache\\Redis not available for local cache","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.2.0"}

```
</details>

nachoparker, 2017-11-19:
Ok, it is a redis problem

```
sudo cat /var/log/redis/redis-server.log
```

and

```
sudo grep unixsocket /etc/redis/redis.conf
sudo grep "^port" /etc/redis/redis.conf
```

eogut, 2017-11-22:
<details>
<summary>redis-server.log</summary>

```

7535:M 19 Nov 00:29:34.030 # Warning: 32 bit instance detected but no memory limit set. Setting 3 GB maxmemory limit with 'noeviction' policy now.
                _._
           _.-``__ ''-._
      _.-``    `.  `_.  ''-._           Redis 3.2.6 (00000000/0) 32 bit
  .-`` .-```.  ```\/    _.,_ ''-._
 (    '      ,       .-`  | `,    )     Running in standalone mode
 |`-._`-...-` __...-.``-._|'` _.-'|     Port: 6379
 |    `-._   `._    /     _.-'    |     PID: 7535
  `-._    `-._  `-./  _.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |           http://redis.io
  `-._    `-._`-.__.-'_.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |
  `-._    `-._`-.__.-'_.-'    _.-'
      `-._    `-.__.-'    _.-'
          `-._        _.-'
              `-.__.-'

7535:M 19 Nov 00:29:34.033 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
7535:M 19 Nov 00:29:34.033 # Server started, Redis version 3.2.6
7535:M 19 Nov 00:29:34.033 * The server is now ready to accept connections on port 6379
7535:signal-handler (1511047973) Received SIGTERM scheduling shutdown...
7535:signal-handler (1511047973) Received SIGTERM scheduling shutdown...
7535:M 19 Nov 00:32:53.805 # User requested shutdown...
7535:M 19 Nov 00:32:53.806 * Saving the final RDB snapshot before exiting.
7535:M 19 Nov 00:32:53.837 * DB saved on disk
7535:M 19 Nov 00:32:53.838 * Removing the pid file.
7535:M 19 Nov 00:32:53.838 # Redis is now ready to exit, bye bye...
601:M 19 Nov 00:33:09.262 # Warning: 32 bit instance detected but no memory limit set. Setting 3 GB maxmemory limit with 'noeviction' policy now.
                _._
           _.-``__ ''-._
      _.-``    `.  `_.  ''-._           Redis 3.2.6 (00000000/0) 32 bit
  .-`` .-```.  ```\/    _.,_ ''-._
 (    '      ,       .-`  | `,    )     Running in standalone mode
 |`-._`-...-` __...-.``-._|'` _.-'|     Port: 6379
 |    `-._   `._    /     _.-'    |     PID: 601
  `-._    `-._  `-./  _.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |           http://redis.io
  `-._    `-._`-.__.-'_.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |
  `-._    `-._`-.__.-'_.-'    _.-'
      `-._    `-.__.-'    _.-'
          `-._        _.-'
              `-.__.-'

601:M 19 Nov 00:33:09.305 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
601:M 19 Nov 00:33:09.305 # Server started, Redis version 3.2.6
601:M 19 Nov 00:33:09.324 * DB loaded from disk: 0.019 seconds
601:M 19 Nov 00:33:09.324 * The server is now ready to accept connections on port 6379
601:signal-handler (1511117974) Received SIGTERM scheduling shutdown...
601:M 19 Nov 19:59:34.510 # User requested shutdown...
601:M 19 Nov 19:59:34.510 * Saving the final RDB snapshot before exiting.
601:M 19 Nov 19:59:34.525 * DB saved on disk
601:M 19 Nov 19:59:34.525 * Removing the pid file.
601:M 19 Nov 19:59:34.525 # Redis is now ready to exit, bye bye...
675:M 19 Nov 20:00:00.194 # Warning: 32 bit instance detected but no memory limit set. Setting 3 GB maxmemory limit with 'noeviction' policy now.
                _._
           _.-``__ ''-._
      _.-``    `.  `_.  ''-._           Redis 3.2.6 (00000000/0) 32 bit
  .-`` .-```.  ```\/    _.,_ ''-._
 (    '      ,       .-`  | `,    )     Running in standalone mode
 |`-._`-...-` __...-.``-._|'` _.-'|     Port: 6379
[... 37 lines cut ...]
      `-._    `-.__.-'    _.-'
          `-._        _.-'
              `-.__.-'

595:M 22 Nov 01:28:05.433 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
595:M 22 Nov 01:28:05.433 # Server started, Redis version 3.2.6
595:M 22 Nov 01:28:05.442 * DB loaded from disk: 0.009 seconds
595:M 22 Nov 01:28:05.442 * The server is now ready to accept connections on port 6379

```
</details>

<details>
<summary>The other thingy</summary>

```

# unixsocket /var/run/redis/redis.sock
# unixsocketperm 700
port 6379

```
</details>

There's still a blank page.

eogut, 2017-11-22:
I used your script but still I get a blank page. Maybe because I first copied your script paragraph by paragraph and then noticed that I had to press enter to apply the last line, too, before inserting the next paragraph...

Here's what I did:

<details>

```

sudo su
root@Nextcloud:/home/pi# REDIS_CONF=/etc/redis/redis.conf
root@Nextcloud:/home/pi# sysctl vm.overcommit_memory=1
vm.overcommit_memory = 1
root@Nextcloud:/home/pi# apt-get update
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done
root@Nextcloud:/home/pi# apt-get purge -y redis-server php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
The following package was automatically installed and is no longer required:
  redis-tools
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  redis-server*
0 upgraded, 0 newly installed, 1 to remove and 22 not upgraded.
After this operation, 877 kB disk space will be freed.
(Reading database ... 41082 files and directories currently installed.)
Removing redis-server (3:3.2.6-1) ...
Processing triggers for man-db (2.7.6.1-2) ...
(Reading database ... 41073 files and directories currently installed.)
Purging configuration files for redis-server (3:3.2.6-1) ...
Processing triggers for systemd (232-25+deb9u1) ...
root@Nextcloud:/home/pi# REDIS_MEM=3gb
root@Nextcloud:/home/pi# sed -i "s|# unixsocket .*|unixsocket /var/run/redis/redis.sock|" $REDIS_CONF
sed: can't read /etc/redis/redis.conf: No such file or directory
root@Nextcloud:/home/pi# sed -i "s|# unixsocketperm .*|unixsocketperm 770|"               $REDIS_CONF
sed: can't read /etc/redis/redis.conf: No such file or directory
root@Nextcloud:/home/pi# sed -i "s|port.*|port 0|"                                        $REDIS_CONF
sed: can't read /etc/redis/redis.conf: No such file or directory
root@Nextcloud:/home/pi# echo "maxmemory ${REDIS_MEM}" >> $REDIS_CONF
bash: /etc/redis/redis.conf: No such file or directory
root@Nextcloud:/home/pi# echo 'vm.overcommit_memory = 1' >> /etc/sysctl.conf
root@Nextcloud:/home/pi# sudo usermod -a -G redis www-data
usermod: group 'redis' does not exist
root@Nextcloud:/home/pi# systemctl restart redis-server
Failed to restart redis-server.service: Unit redis-server.service not found.
root@Nextcloud:/home/pi# systemctl enable redis-server
Failed to enable unit: File redis-server.service: No such file or directory
root@Nextcloud:/home/pi# systemctl stop php7.0-fpm
root@Nextcloud:/home/pi# systemctl stop mysqld
root@Nextcloud:/home/pi#
root@Nextcloud:/home/pi# systemctl start php7.0-fpm
root@Nextcloud:/home/pi# systemctl start mysqld
root@Nextcloud:/home/pi# REDIS_CONF=/etc/redis/redis.conf
echo 'vm.overcommit_memory = 1' >> /etc/sysctl.conf
sudo usermod -a -G redis www-data
systemctl restart redis-server
systemctl enable redis-server

systemctl stop php7.0-fpm
systemctl stop mysqld

systemctl start php7.0-fpm
systemctl start mysqldroot@Nextcloud:/home/pi# sysctl vm.overcommit_memory=1
vm.overcommit_memory = 1
root@Nextcloud:/home/pi# apt-get update
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists... Done
root@Nextcloud:/home/pi# apt-get purge -y redis-server php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
Package 'redis-server' is not installed, so not removed
The following package was automatically installed and is no longer required:
  redis-tools
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 22 not upgraded.
root@Nextcloud:/home/pi# apt-get install -y --no-install-recommends redis-server php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
php-redis is already the newest version (3.1.1-1).
The following NEW packages will be installed:
  redis-server
0 upgraded, 1 newly installed, 0 to remove and 22 not upgraded.
Need to get 0 B/341 kB of archives.
After this operation, 877 kB of additional disk space will be used.
Selecting previously unselected package redis-server.
(Reading database ... 41061 files and directories currently installed.)
Preparing to unpack .../redis-server_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-server (3:3.2.6-1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up redis-server (3:3.2.6-1) ...
Created symlink /etc/systemd/system/redis.service → /lib/systemd/system/redis-server.service.
Created symlink /etc/systemd/system/multi-user.target.wants/redis-server.service → /lib/systemd/system/redis-server.service.
Processing triggers for systemd (232-25+deb9u1) ...
root@Nextcloud:/home/pi# reboot

```

</details>

Thanks for your help mate.

nachoparker, 2017-11-22:
you forgot some lines, don't do it line by line, you can copy the whole thing and paste it in the terminal

eogut, 2017-11-22:
Still blank...

<details>

```

pi@Nextcloud:~ $ sudo su
REDIS_CONF=/etc/redis/redis.conf
sysctl vm.overcommit_memory=1
apt-get update
apt-get purge -y redis-server php7.0-redis
apt-get install -y --no-install-recommends redis-server php7.0-redis

REDIS_MEM=3gb
sed -i "s|# unixsocket .*|unixsocket /var/run/redis/redis.sock|" $REDIS_CONF
sed -i "s|# unixsocketperm .*|unixsocketperm 770|"               $REDIS_CONF
sed -i "s|port.*|port 0|"                                        $REDIS_CONF
echo "maxmemory ${REDIS_MEM}" >> $REDIS_CONF
echo 'vm.overcommit_memory = 1' >> /etc/sysctl.conf
sudo usermod -a -G redis www-data
systemctl restart redis-server
systemctl enable redis-server

systemctl stop php7.0-fpm
systemctl stop mysqld

systemctl start php7.0-fpm
systemctl start mysqldroot@Nextcloud:/home/pi# REDIS_CONF=/etc/redis/redis.conf
root@Nextcloud:/home/pi# sysctl vm.overcommit_memory=1
vm.overcommit_memory = 1
root@Nextcloud:/home/pi# apt-get update
Get:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [123 kB]
Fetched 11.8 MB in 13s (857 kB/s)
Reading package lists... Done
root@Nextcloud:/home/pi# apt-get purge -y redis-server php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
The following package was automatically installed and is no longer required:
  redis-tools
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  redis-server*
0 upgraded, 0 newly installed, 1 to remove and 22 not upgraded.
After this operation, 877 kB disk space will be freed.
(Reading database ... 41082 files and directories currently installed.)
Removing redis-server (3:3.2.6-1) ...
Processing triggers for man-db (2.7.6.1-2) ...
(Reading database ... 41073 files and directories currently installed.)
Purging configuration files for redis-server (3:3.2.6-1) ...
Processing triggers for systemd (232-25+deb9u1) ...
root@Nextcloud:/home/pi# reboot

```

</details>

nachoparker, 2017-11-23:
for whatever reasons, only the first few lines are being executed. Try line by line then, but be careful not to miss any!

You only executed

```
sudo su
REDIS_CONF=/etc/redis/redis.conf
sysctl vm.overcommit_memory=1
apt-get update
apt-get purge -y redis-server php7.0-redis
```

so you can start from

```
apt-get install -y --no-install-recommends redis-server php7.0-redis
```

eogut, 2017-11-23:
It appears not to have worked unfortunately...

<details>

```

pi@Nextcloud:~ $ sudo su
root@Nextcloud:/home/pi# apt-get install -y --no-install-recommends redis-serv php7.0-redis
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'php-redis' instead of 'php7.0-redis'
php-redis is already the newest version (3.1.1-1).
Suggested packages:
  ruby-redis
The following NEW packages will be installed:
  redis-server redis-tools
0 upgraded, 2 newly installed, 0 to remove and 22 not upgraded.
Need to get 0 B/722 kB of archives.
After this operation, 1,948 kB of additional disk space will be used.
Selecting previously unselected package redis-tools.
(Reading database ... 41045 files and directories currently installed.)
Preparing to unpack .../redis-tools_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-tools (3:3.2.6-1) ...
Selecting previously unselected package redis-server.
Preparing to unpack .../redis-server_3%3a3.2.6-1_armhf.deb ...
Unpacking redis-server (3:3.2.6-1) ...
Setting up redis-tools (3:3.2.6-1) ...
Processing triggers for systemd (232-25+deb9u1) ...
Processing triggers for man-db (2.7.6.1-2) ...
Setting up redis-server (3:3.2.6-1) ...
Created symlink /etc/systemd/system/redis.service → /lib/systemd/system/redis-rver.service.
Created symlink /etc/systemd/system/multi-user.target.wants/redis-server.servi → /lib/systemd/system/redis-server.service.
Processing triggers for systemd (232-25+deb9u1) ...
root@Nextcloud:/home/pi# REDIS_MEM=3gb
root@Nextcloud:/home/pi# sed -i "s|# unixsocket .*|unixsocket /var/run/redis/ris.sock|" $REDIS_CONF
sed: no input files
root@Nextcloud:/home/pi# sed -i "s|# unixsocketperm .*|unixsocketperm 770|"             $REDIS_CONF
sed: no input files
root@Nextcloud:/home/pi# sed -i "s|port.*|port 0|"                                      $REDIS_CONF
sed: no input files
root@Nextcloud:/home/pi# echo "maxmemory ${REDIS_MEM}" >> $REDIS_CONF
bash: $REDIS_CONF: ambiguous redirect
root@Nextcloud:/home/pi# echo 'vm.overcommit_memory = 1' >> /etc/sysctl.conf
root@Nextcloud:/home/pi# sudo usermod -a -G redis www-data
root@Nextcloud:/home/pi# systemctl restart redis-server
root@Nextcloud:/home/pi# systemctl enable redis-server
Synchronizing state of redis-server.service with SysV service script with /libystemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable redis-server
root@Nextcloud:/home/pi# systemctl stop php7.0-fpm
root@Nextcloud:/home/pi# systemctl stop mysqld
root@Nextcloud:/home/pi# systemctl start php7.0-fpm
root@Nextcloud:/home/pi# systemctl start mysqld
root@Nextcloud:/home/pi# reboot

```

</details>

nachoparker, 2017-11-23:
finally!!

Well, there is a donations button in ownyourbits.com ;)

Thanks!