nextcloud/nextcloudpi #303: Add the option for Hybrid installation in the Wiki

el-Salmon, 2017-11-17:
Please update the How-to-install-NextCloudPi-on-a-Raspberry-Pi wiki by adding a third option:

### Option 3: Hybrid (/boot partition on SD card, / partition on USB)

Follow the Option 1 steps and then [move the root partition to USB disk](https://elinux.org/Transfer_system_disk_from_SD_card_to_hard_disk). This will limit the filesystem corruptions that happen usually when root partition is in a SD card.

testbird, 2017-11-24:
Maybe also mention the armbian-config tool feature available for other boards.

nachoparker, 2017-11-24:
we should also mention that spinning disk will be spinning constantly if we do this

testbird, 2017-11-24:
Linux can spin down system disks. A nice tool that (also) allows this:
http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html

Strange they call it wiki, but I can't see an edit button.

nachoparker, 2017-11-25:
> Strange they call it wiki, but I can't see an edit button.

our wiki?

testbird, 2017-11-25:
Yes, github.

el-Salmon, 2017-11-25:
> Linux can spin down system disks. A nice tool that (also) allows this:
> http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html
Are you sure tlp works fine in Raspberry Pi? I have no found references about this tool under Raspbian.

testbird, 2017-11-25:
You would need to try, if shell skipts of the package work with the rpi kernel. (They should I guess, as the linux io handling is hardware independent?).

testbird, 2017-11-25:
Kernel feature: laptop_mode

nachoparker, 2017-11-25:
For controlling that kind of thing I've always used hdparm

testbird, 2017-11-25:
Hdparm is just one part of it, it lets the disk spin down when idle, but only laptop_mode enables buffering writes for say 10 minutes of max lost data in case of system crash. And this buffer might be increased to 1 day or one week if is just the daily `<mark>` that's added to the logs.

nachoparker, 2017-11-28:
> Yes, github.

@testbird I think the permissions for the wiki were misconfigured. If you want to write some stuff down over there, please try again and tell me if it works now.

testbird, 2017-11-28:
Thantks, editing worked now, may sometimes even be quicker than posting an issue.
https://github.com/nextcloud/nextcloudpi/wiki/How-to-install-NextCloudPi-on-a-Raspberry-Pi

closed 2017-11-28 by el-Salmon

nachoparker, 2017-11-28:
great, thank you very much!