nextcloud/nextcloudpi #304: Wrong default user name and password for the Nextcloud web on the Wiki

el-Salmon, 2017-11-17:
The steps described to install and initial config says [the default web user is pi](https://github.com/nextcloud/nextcloudpi/wiki/How-to-access-NextCloudPi), but it's admin. Check the [comment](https://github.com/nextcloud/nextcloudpi/issues/30#issuecomment-309199739)

nachoparker, 2017-11-17:
hi,

thanks for reporting.

the user for Nextcloud it's _admin_, and for NextCloudPi web panel (or ncp-web) on port 4443 used to be _pi_, and we are changing it on new images to _ncp_

closed 2017-11-17 by nachoparker

Iolaum, 2017-12-09:
@nachoparker 

The information on the wiki [here](https://github.com/nextcloud/nextcloudpi/wiki/How-to-configure-NextCloudPi#webui) need to be updated with the new defaults (ncp, ownyourbits).

Daniel-Krebs, 2017-12-09:
@Iolaum Thanks for noticing, I added Info to the wiki. AFAIK old installations still use pi/raspberry and only new installations use ncp/ownyourbits. I'm not sure at which point we changed it, though. @nachoparker can you remember this? Then I would change the wiki to "if you installed ncp before ... use pi/raspberry, if you installed after ... use ncp/ownyourbits"

nachoparker, 2018-09-19:
You should either set up auto-updates, or update at least once a month. Otherwise, expect missconfigurations. If you can access through `ncp-config`, then you can change the ncp-web password through `nc-passwd`

Dutchgent, 2018-09-20:
Hi Nacho, thanks for your response and effort to help, appreciate that.
If I open the ncp-config why don't I see nc-passwd? In my list I have nc-notify-updates followed immediately by nc-ramlogs. The option nc-passwd is missing in my list. Again in advance many thanks for your support.

nachoparker, 2018-09-20:
Glad to help

No, if you have an old instance, you won't have an `ncp` user in NC. It is extremely weird that you don't see `nc-passwd`, but if you have such an old version and you didn't upgrade frequently... anything could happen.

It should be under `/usr/local/etc/ncp-config.d`. Also check `/usr/local/etc/nextcloudpi-config.d`.

I suggest that you backup through `ncp-config`, and then install a new version, follow the wizard (this will enable autoupdates), and then restore. 

If you want to restore your ncp-web password through terminal, type

```
sudo passwd ncp
```

Dutchgent, 2018-09-21:
Again thank you for you help nachoparker!
We are obviously in different time zone, I'm in Asia, you are in US I assume :-)
Apologies for not having much programming knowledge (wish I could learn more how all really is structured and works), but I love to figure things out with software and experimenting to build my own server on my raspberry pi and i love your nextcloud pi. I updated my Pi every weekend I assumed that whatever was installed on it was then updated, obviously nextcloud was not updated with it, lesson learned. However all is working perfectly, I can use NC without issues, see all folders and files in my NAS which I linked. All users are there. Only is that I am not able to use 4443.
Now based on your suggestions using WinSCP I had a look in both /usr/local/etc/ncp-config.d as well as /usr/local/etc/nextcloudpi-config.d and could see nc-passwd in /usr/local/etc/ncp-config.d but not in /usr/local/etc/nextcloudpi-config.d, so I experimented bit further. I always used 'sudo nextcloudpi-config' but now instead I tried 'sudo ncp-config'. This starts the same configuration tool but see much many more options including the nc-passw. I also went in nc-admin and I can see the user is ncp and password is ownyourbits. So there it all seems as it should be. But when I go to https:localaddress:4443 it will prompt me for username and password and does not let me in =(
I notice one difference between starting ncp-config and nextcloudpi-config, both v0.59.18. When starting ncp-config I do not get a message but with nextcloudpi-config I receive "Cannot load Zend OPcache - it was already loaded".
So unless you have some suggestion for me based on given details here I am ok to install a new version, but have some questions about this.
I am correct to conclude that there are 2 nextclouds are installed namely nextcloudpi and ncp?
If I do nc-backup I will keep the configuration if i understand correctly. Will all setup users and the paths to my NAS and the passwords etc be captured in that backup? I have no data on sd in the pi, all is on Nas so no need to backup data, so i specify that.
Then if I do nc-init it will install a new nextcloud with clean configuration and if my assumption of 2 installed nextclouds is correct will this result in just 1? Or do I have to completely install from scratch on a clean sd card? That would be bit painful as I set up few other things as mumble and noip and cron etc. Then I have to figure all that out again. It is not something I do daily and my rusty brains do not all remember =)
Apologies for this long story but again much thanks for you effort and support. Have a nice weekend ahead nachoparker!

Iolaum, 2018-09-21:
@Dutchgent  

- "Cannot load Zend OPcache - it was already loaded" is related to #566 so it should be fixed - maybe you skipped the updates that fixed it? You can maybe try the fixes mentioned there manually if you want.
- The are 'two' nextclouds, assuming I understand your question. NC and NCP. Nextcloud (NC) is the file syncing and storing solution that is also accessible through $WEBSITE:443. NextCloudPi (NCP) is the software that administrates the device/server that NC is installed on. It is accessible though $WEBSITE:4443, assuming you have ncp-web enabled. 

Hope that helps.

nachoparker, 2018-09-22:
The easiest is to try to [backup and restore](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-ncp-config).

If you don't have moved your datadir. I recommend that you just use a different SD card (or you can back it up too). This way you will save headaches in case something goes wrong, because your system is a bit messed up.

Dutchgent, 2018-09-22:
Many thanks both

p6002, 2020-11-13:
Welcome in open source community, when "something" works, but not always.
Remember to not store important data on "cloud" services like nextcloud.