nextcloud/nextcloudpi #305: Unable to use 1TB hard drive as data location

aramsichord, 2017-11-18:
I am trying to use a 1TB hard drive (spinning disk) as the data location with my raspberry pi 2. Every time I try to do this with the NextCloudPi Wizard, I get the message 

[ nc-datadir ]
Error reading data directory. Is NextCloud running and configured?

I have also recieved other error messages but I can't recreate them at the moment. I can't log into the web interface because it says there is no data directory. I have ssh access currently. Please help! I am planning on using it to sync multiple files across computers, I have gotten it working in the past with just the sd card's small storage, but I can't upgrade to the hard drive.

Thanks!

aramsichord, 2017-11-18:
I just remembered, when I do manage to set up the external drive as the storage source, the data doesn't appear in the drive (/media/7blah blah blah). For some reason it appears in /media/USBdrive. As far as I can tell, the drive is not mounted to /media/USBdrive, and adding files to nextcloud does not put them on the external drive.

Thanks for your help!

nachoparker, 2017-11-18:
Hi,

what's the output of

```
sudo grep datadirectory /var/www/nextcloud/config/config.php
```

```
ls -l /media
```

nachoparker, 2017-11-18:
Hi,

I just finished a tool to help with support. Please, update

```
sudo ncp-update
```

and then post the output of

```
sudo ncp-report
```

aramsichord, 2017-11-18:
[sudo grep datadirectory] 
 'datadirectory' => '/media/USBdrive/ncdata',

[ls -l /media]
total 16
drwxr-x--- 7 www-data www-data 4096 Nov 18 00:15 7*
drwxr-xr-x 3 root     root     4096 Nov 18 01:15 a08d85ae-015d-49c8-98f1-604c416f0a6e
drwxr-xr-x 2 root     root     4096 Nov 18 01:17 ncdata
drwxr-xr-x 2 root     root     4096 Nov 18 01:17 USBdrive

[sudo ncp-report]
<details>
<summary>NextCloudPi diagnostics</summary>
```

ping: none: Name or service not known
NextCloudPi version  v0.34.15
NextCloudPi image    NextCloudPi_11-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata (doesn't exist)
rootfs usage         1.9G/2.7G
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.133
gateway              192.168.1.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               4min

```
</details>

<details>
<summary>Nextcloud configuration</summary>
```

Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Cannot create "data" directory
This can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions

An unhandled exception has been thrown:
Exception: Environment not properly prepared. in /var/www/nextcloud/lib/private/Console/Application.php:145
Stack trace:
#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#2 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>
```

[Sat Nov 18 00:35:11.460805 2017] [proxy_fcgi:error] [pid 829:tid 1977328688] (104)Connection reset by peer: [client 192.168.1.192:57676] AH01075: Error dispatching request to :4443: , referer: https://192.168.1.133:4443/wizard/
[Sat Nov 18 00:35:11.527600 2017] [proxy_fcgi:error] [pid 829:tid 1893442608] [client 192.168.1.192:57676] AH01067: Failed to read FastCGI header, referer: https://192.168.1.133:4443/wizard/
[Sat Nov 18 00:35:11.527873 2017] [proxy_fcgi:error] [pid 829:tid 1893442608] (104)Connection reset by peer: [client 192.168.1.192:57676] AH01075: Error dispatching request to :4443: , referer: https://192.168.1.133:4443/wizard/
[Sat Nov 18 00:35:13.313510 2017] [mpm_event:notice] [pid 728:tid 1992589312] AH00491: caught SIGTERM, shutting down
[... 31 lines cut ...]
2017-11-18  1:24:34 1988771840 [Note] InnoDB: The log sequence numbers 3192362 and 3192362 in ibdata files do not match the log sequence number 3781673 in the ib_logfiles!
2017-11-18  1:24:34 1988771840 [Note] InnoDB: Restoring possible half-written data pages from the doublewrite buffer...
2017-11-18  1:24:34 1988771840 [Note] InnoDB: 128 rollback segment(s) are active.
2017-11-18  1:24:34 1988771840 [Note] InnoDB: Waiting for purge to start
2017-11-18  1:24:34 1988771840 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3781673
2017-11-18  1:24:34 1442837312 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-11-18  1:24:34 1988771840 [Note] Plugin 'FEEDBACK' is disabled.
2017-11-18  1:24:34 1988771840 [Note] Recovering after a crash using tc.log
2017-11-18  1:24:34 1988771840 [Note] Starting crash recovery...
2017-11-18  1:24:34 1988771840 [Note] Crash recovery finished.
2017-11-18  1:24:35 1988771840 [Note] Server socket created on IP: '127.0.0.1'.
2017-11-18  1:24:35 1988771840 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>
```

tail: cannot open '/media/USBdrive/ncdata/nextcloud.log' for reading: No such file or directory

```
</details>

aramsichord, 2017-11-18:
Thanks for the help!

aramsichord, 2017-11-18:
Also, nextcloudpi never seems to update. When I boot it up and login over ssh, I get this message "NextCloudPi v0.34.15 is up to date". Then when I open raspi-config--> nextcloudpi config it says "Update to v0.36.0?" I do this, it says 
"Downloading updates
Performing updates
vm.overcommit_memory = 1"
Is it supposed to say that bit at the end? When I reboot it still says v0.34.15" Could this be related?

nachoparker, 2017-11-19:
Your data is not at `/media/USBdrive/ncdata`. Have you been moving files around? Maybe it is that folder in `/media/ncdata`?

Try this 
```
sudo mv /media/ncdata /media/USBdrive
```

And if it doesn't work, output of

```
mount
```

aramsichord, 2017-11-19:
Here is the output of mount. Will doing a fresh install have the same effect as mv the files? If I were trying a fresh install, should I try using the wizard to setup the external drive? Does the nextcloud external drive setup use symlinks? What directory would the drive be linked to, /media/USBdrive?

/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=470184k,nr_inodes=117546,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=31,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
/dev/sda on /media/a08d85ae-015d-49c8-98f1-604c416f0a6e type ext4 (rw,nosuid,nodev,relatime,stripe=8191,data=ordered,uhelper=udisks2)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94956k,mode=700,uid=1000,gid=1000)

aramsichord, 2017-11-19:
UPDATE: I started fresh and wrote the nextcloud image to the sd card again, and updated with nc-update (which worked fine). I loaded up the nextcloudpi wizard in the browser, formatted the usb drive, then tried to move the data to the usb drive. I got this message:
[ nc-format-USB ]
/dev/sda: 2 bytes were erased at offset 0x00000438 (ext4): 53 ef
Drive sda formatted successfuly and labeled myCloudDrive
[ nc-datadir ]
Refusing to move to the SD card. Abort

All I see in the /media directory is USBdrive, I don't see any myclouddrive. It kinda looks like the usb drive isn't being mounted properly, despite being formatted correctly.

nachoparker, 2017-11-19:
Hi,

Thanks for reporting again. What you mention surprises me... is automount failing?

Can you try the wizard again, skipping the formatting step?

If it fails, give m the output of

```
sudo ncp-report
```

aramsichord, 2017-11-19:
I reran the wizard, sudo ncp-report says:

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>
```

NextCloudPi version  v0.37.0
NextCloudPi image    NextCloudPi_11-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.9G/2.7G
rootfs usage         1.9G/2.7G
Nextcloud check      ok
Nextcloud version    12.0.3.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.133
gateway              192.168.1.1
interface            eth0
certificates         none
certs due            none
NAT loopback         no
uptime               27min

```
</details>

<details>
<summary>Nextcloud configuration</summary>
```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.133",
            "5": "nextcloudpi.local"
        },
        "datadirectory": "\/var\/www\/nextcloud\/data",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "12.0.3.3",
        "dbname": "nextcloud",
        "dbhost": "localhost",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "oclqkkiqcr6y",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "\/var\/run\/redis\/redis.sock",
            "port": 0,
[... 59 lines cut ...]
2017-11-19 21:46:07 1989201920 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 3043123
2017-11-19 21:46:07 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2017-11-19 21:46:07 1989201920 [Note] Plugin 'FEEDBACK' is disabled.
2017-11-19 21:46:07 1989201920 [Note] Server socket created on IP: '127.0.0.1'.
2017-11-19 21:46:07 1989201920 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>
```

{"reqId":"WTSgmYsaiwa6dP0UWloB","level":4,"time":"2017-11-12T09:15:17+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"mOepx4OvoJnvEPG4ceyq","level":4,"time":"2017-11-12T09:30:14+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"iT0KOL06IWMgzQ45bQER","level":4,"time":"2017-11-12T09:45:12+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"usdnVNQgPTuyPts4i3xC","level":4,"time":"2017-11-12T10:00:18+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"C8f4UMPnsqYoygWQCZzW","level":4,"time":"2017-11-12T10:15:11+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"kV6EJNKVTueHrwFWYM6T","level":4,"time":"2017-11-12T10:30:12+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"i4ZUxeGKCem0ALWEojEo","level":4,"time":"2017-11-12T10:45:12+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"WhIAQ38AAQEAAANo5j8AAABH","level":2,"time":"2017-11-19T22:05:56+00:00","remoteAddr":"192.168.1.192","user":"--","app":"core","method":"PROPFIND","url":"\/remote.php\/webdav\/","message":"Login failed: 'admin' (Remote IP: '192.168.1.192')","userAgent":"Mozilla\/5.0 (Windows) mirall\/2.3.2 (build 1) (Nextcloud)","version":"12.0.3.3"}

```
</details>

nachoparker, 2017-11-19:
the wizard fails again in the same place?

`ls -l /media` doesn't show the drive?

aramsichord, 2017-11-19:
Well, it shows it here:

total 8
drwxr-xr-x 3 root root 4096 Nov 19 21:34 myCloudDrive
drwxr-xr-x 2 root root 4096 Nov 19 21:40 USBdrive

aramsichord, 2017-11-19:
Now `ls /media` shows myCloudDrive  USBdrive

aramsichord, 2017-11-20:
When I specify /media/USBdrive/ncdata for `nc-datadir` from the nextcloudpi config website, I get `Refusing to move to the SD card. Abort`.

nachoparker, 2017-11-20:
I think I know what's going in your case. Let's try this

 - start from scratch: copy SD image
 - update to `v.37.2` ( from web-ui or `sudo ncp-update` )
 - try wizard again

or

- update to `v.37.2` ( from web-ui or `sudo ncp-update` )
- clean all mounts
```
sudo umount /media/*
sudo rmdir /media/*
sudo rm /media/*
sudo reboot
```
- after reboot, try the wizard again

aramsichord, 2017-11-20:
Most of the commands failed. I think the drive wasn't automounted. I tried to open the wizard, but it didn't work, so I opened the nextcloud web interface first and then tried the wizard and it worked. In the wizard, I automounted the usb drive then tried to format it, but of course it said it was still mounted. I unmounted it via the terminal, and tried formatting it again, which took quite a while. The formatting was succesfull and it said `Drive sda formatted successfuly and labeled myCloudDrive`, but then when I tried to move the data directory it said
 `[ nc-datadir ]
/media/USBdrive does not exist`

So I did `sudo mkdir /media/USBdrive` and then `sudo mount /dev/sda /media/USBdrive` and that command didn't work. I tried some other commands and then tried nc-datadir again and it said
`[ nc-datadir ]
Refusing to move to the SD card. Abort`

aramsichord, 2017-11-20:
What directory is nc-datadir looking for? Can I just mount the drive manually in the right spot?

nachoparker, 2017-11-20:
ok, that's what I thought, so let's see why you are not getting the link...

- remove the `/media/USBdrive` folder
- reboot
- output of `systemctl status nc-automount-links`

nachoparker, 2017-11-20:
> What directory is nc-datadir looking for? Can I just mount the drive manually in the right spot?

Yes, we can do that. I am trying to investigate why it is not working for you, so I can fix it for others. It is strange because in all this time this hasn't happened before

aramsichord, 2017-11-20:
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-20 16:35:06 UTC; 23s ago
 Main PID: 308 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─308 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─316 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─317 grep --line-buffered ISDIR
           └─318 /bin/bash /usr/local/etc/nc-automount-links-mon

Nov 20 16:35:06 nextcloudpi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Nov 20 16:35:06 nextcloudpi nc-automount-links-mon[308]: Setting up watches.
Nov 20 16:35:06 nextcloudpi nc-automount-links-mon[308]: Watches established.
Nov 20 16:35:11 nextcloudpi nc-automount-links-mon[308]: myCloudDrive CREATE,ISDIR

nachoparker, 2017-11-20:
output of 
```
ls -l /media
```

aramsichord, 2017-11-20:
Do you think the reason I am having the problem is due to the hard drive's power strain on the pi? Everything seems to be fine though.

nachoparker, 2017-11-20:
> Do you think the reason I am having the problem is due to the hard drive's power strain on the pi? Everything seems to be fine though.

shouldn't be related

aramsichord, 2017-11-20:
total 4
drwxr-xr-x 3 root root 4096 Nov 20 16:06 myCloudDrive

nachoparker, 2017-11-20:
Ok, now run this command, 
```
sudo /usr/local/etc/nc-automount-links
```

and let's see

```
ls -l /media
```

aramsichord, 2017-11-20:
total 4
drwxr-xr-x 3 root root 4096 Nov 20 16:06 myCloudDrive
lrwxrwxrwx 1 root root   19 Nov 20 16:38 USBdrive -> /media/myCloudDrive

nachoparker, 2017-11-20:
ok, now it worked... the problem is why it doesn't work by itself.

- remove `/media/USBdrive`
- unplug USB
- `systemctl stop nc-automount-links`
- `systemctl start nc-automount-links`
- plug USB
- ls -l /media
- `systemctl status nc-automount-links`

aramsichord, 2017-11-20:
This is what I got for the first commands
```
Failed to stop nc-automount-links.service: The name org.freedesktop.PolicyKit1 was not provided by any .service files
See system logs and 'systemctl status nc-automount-links.service' for details.
Failed to start nc-automount-links.service: The name org.freedesktop.PolicyKit1 was not provided by any .service files
```

aramsichord, 2017-11-20:
```
● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-20 16:35:06 UTC; 39min ago
 Main PID: 308 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─308 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─316 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─317 grep --line-buffered ISDIR
           └─318 /bin/bash /usr/local/etc/nc-automount-links-mon

Nov 20 16:35:06 nextcloudpi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Nov 20 16:35:06 nextcloudpi nc-automount-links-mon[308]: Setting up watches.
Nov 20 16:35:06 nextcloudpi nc-automount-links-mon[308]: Watches established.
Nov 20 16:35:11 nextcloudpi nc-automount-links-mon[308]: myCloudDrive CREATE,ISDIR
Nov 20 16:44:21 nextcloudpi nc-automount-links-mon[308]: myCloudDrive DELETE,ISDIR

```

nachoparker, 2017-11-20:
sorry, I forgot that everything goes with sudo

- remove `/media/USBdrive`
- unplug USB
- `sudo systemctl stop nc-automount-links`
- `sudo systemctl start nc-automount-links`
- plug USB
- `ls -l /media`
- `systemctl status nc-automount-links`

aramsichord, 2017-11-20:
`total 0`

`● nc-automount-links.service - Monitor /media for mountpoints and create USBdrive* symlinks
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-20 17:20:32 UTC; 2min 3s ago
 Main PID: 1972 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─1972 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─1974 inotifywait --monitor --event create --event delete --format %f %e /media/
           ├─1976 grep --line-buffered ISDIR
           └─1977 /bin/bash /usr/local/etc/nc-automount-links-mon

Nov 20 17:20:32 nextcloudpi systemd[1]: Started Monitor /media for mountpoints and create USBdrive* symlinks.
Nov 20 17:20:32 nextcloudpi nc-automount-links-mon[1972]: Setting up watches.
Nov 20 17:20:32 nextcloudpi nc-automount-links-mon[1972]: Watches established.`

nachoparker, 2017-11-20:
did you plug the USB in after `systemctl start` and before `systemctl status`?

aramsichord, 2017-11-20:
Hmm, I'll have to try it again.

nachoparker, 2017-11-20:
Ok, just do the whole sequence step by step in order

sorry, I forgot that everything goes with sudo

- remove `/media/USBdrive`
- unplug USB
- `sudo systemctl stop nc-automount-links`
- `sudo systemctl start nc-automount-links`
- plug USB in
- `ls -l /media`
- `systemctl status nc-automount-links`

aramsichord, 2017-11-20:
`total 0`

● nc-automount-links.service - Monitor /media for mountpoints and create USBdriv
   Loaded: loaded (/usr/lib/systemd/system/nc-automount-links.service; enabled;
   Active: active (running) since Mon 2017-11-20 19:14:43 UTC; 21min ago
 Main PID: 4048 (nc-automount-li)
   CGroup: /system.slice/nc-automount-links.service
           ├─4048 /bin/bash /usr/local/etc/nc-automount-links-mon
           ├─4049 inotifywait --monitor --event create --event delete --format %
           ├─4050 grep --line-buffered ISDIR
           └─4051 /bin/bash /usr/local/etc/nc-automount-links-mon

Nov 20 19:14:43 nextcloudpi systemd[1]: Started Monitor /media for mountpoints a
Nov 20 19:14:43 nextcloudpi nc-automount-links-mon[4048]: Setting up watches.
Nov 20 19:14:43 nextcloudpi nc-automount-links-mon[4048]: Watches established.

aramsichord, 2017-11-20:
Currently if I do the wizard it says
```
[ nc-datadir ]
Refusing to move to the SD card. Abort
```
As if it doesn't even recognize the drive.

nachoparker, 2017-11-20:
This doesn't make sense. Automount was working for you before. Is it activated?

```
systemctl status nc-automount
```

aramsichord, 2017-11-20:
Yes it is.

nachoparker, 2017-11-20:
systemctl status nc-automount

aramsichord, 2017-11-20:
● nc-automount.service - Automount USB drives
   Loaded: loaded (/usr/lib/systemd/system/nc-automount.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2017-11-20 16:35:06 UTC; 3h 43min ago
 Main PID: 325 (udiskie)
   CGroup: /system.slice/nc-automount.service
           └─325 /usr/bin/python3 /usr/bin/udiskie -NTF

Nov 20 17:10:47 nextcloudpi udiskie[325]:        In some cases useful info is found in syslog - try
Nov 20 17:10:47 nextcloudpi udiskie[325]:        dmesg | tail or so.
Nov 20 17:22:04 nextcloudpi udiskie[325]: failed to mount /org/freedesktop/UDisks2/block_devices/sda1: GDBus.Error:org.freedesktop.UDisks2.Error.Failed: Error mounting system-mana
Nov 20 17:22:04 nextcloudpi udiskie[325]:        missing codepage or helper program, or other error
Nov 20 17:22:04 nextcloudpi udiskie[325]:        In some cases useful info is found in syslog - try
Nov 20 17:22:04 nextcloudpi udiskie[325]:        dmesg | tail or so.
Nov 20 19:35:54 nextcloudpi udiskie[325]: failed to mount /org/freedesktop/UDisks2/block_devices/sda1: GDBus.Error:org.freedesktop.UDisks2.Error.Failed: Error mounting system-mana
Nov 20 19:35:54 nextcloudpi udiskie[325]:        missing codepage or helper program, or other error
Nov 20 19:35:54 nextcloudpi udiskie[325]:        In some cases useful info is found in syslog - try
Nov 20 19:35:54 nextcloudpi udiskie[325]:        dmesg | tail or so.

aramsichord, 2017-11-20:
I ran dmesg | tail:

[10844.589652] sd 0:0:0:0: Attached scsi generic sg0 type 0
[10845.601250] sd 0:0:0:0: [sda] 1953525167 512-byte logical blocks: (1.00 TB/932 GiB)
[10845.601758] sd 0:0:0:0: [sda] Write Protect is off
[10845.601775] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[10845.602261] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[10845.714065]  sda: sda1
[10845.718467] sd 0:0:0:0: [sda] Attached SCSI disk
[10846.326360] FAT-fs (sda1): bogus number of reserved sectors
[10846.326373] FAT-fs (sda1): Can't find a valid FAT filesystem
[10985.465760] systemd[1]: apt-daily.timer: Adding 55min 54.860604s random time.

nachoparker, 2017-11-20:
aha, you have some problem with udiskie

can you mount manually?

```
sudo mount /dev/sda1 /mnt
mount
```

aramsichord, 2017-11-20:
/dev/mmcblk0p2 on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=470184k,nr_inodes=117546,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/net_cls type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=28,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
mqueue on /dev/mqueue type mqueue (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/sda1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=94956k,mode=700,uid=1000,gid=1000)
/dev/sda1 on /mnt type ext4 (rw,relatime,data=ordered)

aramsichord, 2017-11-20:
ls -l still says `total 0`

nachoparker, 2017-11-20:
ok, udiskie can't mount it but you can mount it manually... smells bad. Did you format it with the wizard ?

can you try doing that again? careful, if you have data it will be lost.

aramsichord, 2017-11-20:
I could format it again, it takes a long time.

nachoparker, 2017-11-20:
let's try that

aramsichord, 2017-11-20:
It's formating right now.

aramsichord, 2017-11-20:
[ nc-format-USB ]
/dev/sda: 8 bytes were erased at offset 0x00000200 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 8 bytes were erased at offset 0xe8e0db5c00 (gpt): 45 46 49 20 50 41 52 54
/dev/sda: 2 bytes were erased at offset 0x000001fe (PMBR): 55 aa
Error: Partition(s) 1 on /dev/sda have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.

aramsichord, 2017-11-20:
Just got that message.

nachoparker, 2017-11-20:
sorry forgot, do this first and try again

```
sudo umount /mnt
```

aramsichord, 2017-11-20:
I rebooted, seems to be working now. Do you know why formatting the drive takes so long? There isn't any data on it.

nachoparker, 2017-11-20:
I don't know, it's pretty fast on my drives... that's why I am starting to suspect there might be something wrong with your drive, because it's acting up.

Do you have the possibility of using a standalone power supply for it?

aramsichord, 2017-11-20:
Not really

aramsichord, 2017-11-20:
Perhaps I'll go back and try it with a normal flash drive

nachoparker, 2017-11-20:
if this fails again, it's worth a try.

If it fails again I would also try running

```
sudo e2fsck /dev/sda1
```

aramsichord, 2017-11-20:
What does that do?

aramsichord, 2017-11-20:
With a normal flash drive, it says 
[ nc-datadir ]
/media/USBdrive does not exist
Even after I run `sudo mkdir /media/USBdrive`

nachoparker, 2017-11-20:
> Even after I run sudo mkdir /media/USBdrive

please, don't run that, it won't work

> What does that do?

checks for problems

Did you format the flash drive using the wizard?

[25 more replies]

closed 2017-11-20 by nachoparker