nextcloud/nextcloudpi #306: nc-update doesn't work anymore

user-jan, 2017-11-18:
Hi Nacho,

When I try to use "nc-update" I get the following message:

[image]

I'm not sure why this is happening, perhaps you have an idea :-)

Best,
Jan

<summary>NextCloudPi diagnostics</summary>
```

NextCloudPi version  v0.34.15
NextCloudPi image    NextCloudPi_11-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      3.7G/15G
rootfs usage         3.7G/15G
Nextcloud check      ok
Nextcloud version    12.0.3.3
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.2
gateway              192.168.1.1
interface            eth0
certificates         27jas06.ddns.net
certs due            83 days
NAT loopback         no
uptime               6:38

vlmd, 2017-11-18:
Same here. Tried to update from NextCloudPi v0.34.15 to v0.36. 
It stuck on the "vm.overcommit_memory = 1" message. 
After that, the update process is done without updating.

nachoparker, 2017-11-19:
fixed

thanks for reporting

closed 2017-11-19 by nachoparker