nextcloud/nextcloudpi #310: can't connect to nextcloudpi -

Ricardosgeral, 2017-11-21:
I'm getting this 
https://screenshots.firefox.com/c0BrCf8hjbWiaCmf/ricarcloud.duckdns.org

what is happening? I have automatic updates!

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>
```
sudo: /var/www/nextcloud/occ: command not found

regards

nachoparker, 2017-11-21:
did you run `nc-nextcloud` by any chance?

Ricardosgeral, 2017-11-21:
no, I'm sure i didn't!

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

nachoparker, 2017-11-21:
automatic updates will never remove your config.. and you don't have the occ command?

```
sudo ls /var/www/nextcloud/occ
```

Ricardosgeral, 2017-11-21:
The only thiing i've done was to install pico csm app .

do you recomend a new instalation of nextcloudpi? or a simple nc-restore
will fix it? I've a backup with 2 days old

nachoparker, 2017-11-21:
> The only thiing i've done was to install pico csm app .

is this as nextcloud app?

you can run `nc-init` which will initialize config, and after that `nc-restore`

I would like to know what happened though

nachoparker, 2017-11-21:
> but i don't think this would mess the config files of nextcloud

no, no way

also ncp-update doesn't touch anything

```
sudo ls /var/www/nextcloud/occ
sudo ls /var/www/nextcloud/config/config.php
```

Ricardosgeral, 2017-11-22:
doing nc-init and nc-restore didnt solve it.

error in mySQL something about socket.

it seems that ive lost database configuration

i think i have to do a new instalation

nachoparker, 2017-11-22:
what in the world happened? did you run out of free space?

```
df -h
```

Maybe you can try

```
sudo ncp-update
sudo ncp-report
```

, or just start fresh and restore your backup

Ricardosgeral, 2017-11-22:
well i've started from start

just some comments to improve nextcloudpi:
- maybe the image could have already SSH enabled by default. This would
avoid to connect the pi to a screen.
- when doing 192.168.1.x:4443 i'm asked to put a user and password. Tried
seveal user and passwords but nothing seens to work. admin/ownyourbits
pi/raspberry don't seem to work.anyway i've done the instalation the
traditional way using SSH.
-  By the way, maybe its time to also change the logo that apperar in the
terminal ;) to the new one

nachoparker, 2017-11-22:
thanks for your suggestions

> maybe the image could have already SSH enabled by default. This would avoid to connect the pi to a screen.

I don't think it is a good idea securitywise. Raspbian developers seem to think the same way. An option could be this

https://github.com/nextcloud/nextcloudpi/issues/130

> By the way, maybe its time to also change the logo that apperar in the terminal ;) to the new one

I know, work is piling up! ;D 

do you like the new logo better?

> when doing 192.168.1.x:4443 i'm asked to put a user and password. Tried seveal user and passwords but nothing seens to work. admin/ownyourbits pi/raspberry don't seem to work.anyway i've done the instalation the traditional way using SSH.

We are in a weird spot, because not all the images have changed yet.. it is either pi/raspberry or ncp/ownyourbits. Sorry about that

closed 2017-11-22 by nachoparker

Ricardosgeral, 2017-11-22:
yes. The new logo cooler and self-explanatory. I like it.